lagadic/visp #613: C++11 or C++14 as a minimal requirement for the next release

s-trinh, 2019-08-11:
Should we consider moving to C++11 or C++14 as a minimal requirement for the next release?

OpenCV 4 requires C++11, PCL [requires C++14](https://github.com/PointCloudLibrary/pcl/projects/1), [ROS 2 is C++14](http://design.ros2.org/articles/changes.html) also (but ROS 2 adoption is still the beginning).

This would allow removing all the checks for C++11 compiler support for things like move constructor, move assignment operator, ... code and use the new features or modern library.

For compiler support:
  - [C++ compiler support](https://en.cppreference.com/w/cpp/compiler_support#C.2B.2B14_library_features)
  - [Support For C++11/14/17 Features](https://docs.microsoft.com/en-us/previous-versions/hh567368(v=vs.140)#featurelist)

---

On a related topic, we should add a command line parser library to avoid using custom code or `vpParseArgv`. For instance with [Lyra](https://github.com/bfgroup/Lyra) (C++11), the code is as simple as:

```
int width = 0;
std::string name;
bool doIt = false;
std::string command;
auto cli
    = lyra::opt( width, "width" )
        ["-w"]["--width"]
        ("How wide should it be?")
    | lyra::opt( name, "name" )
        ["-n"]["--name"]
        ("By what name should I be known")
    | lyra::opt( doIt )
        ["-d"]["--doit"]
        ("Do the thing" )
    | lyra::arg( command, "command" )
        ("which command to run");
```

fspindle, 2019-08-21:
Sorry, but we cannot make C++11 as minimal. 
We have an old Ultrasonix machine running WinXP and Visual Studio 2010 that doesn't have C++11. We cannot upgrade the machine.

---
Yes I agree, it could be nice to introduce Lyra. Lyra license is compatible with GPL and sounds compatible with the Professional Edition license. We could copy lyra-1.0 release in `3rdparty` folder. What do you think ?

s-trinh, 2019-08-21:
I see.

Maybe in the future if there is no big development targeting the medical platform a solution would be to use 2 versions / 2 branches similar to OpenCV. They manage to [backport developments](https://github.com/opencv/opencv/pull/15323) made in the 3.4 branch into the master branch.

---

[Lyra](https://bfgroup.github.io/Lyra/) is licensed under the Boost Software License so there should be no issue.
Introducing Lyra which is a C++11 header only will necessite to duplicate code if we want to use it in ViSP and will add more clutter than the current state.

For personnal project, this is one file to download and it is simple to use. For now, there is no need to integrate it into ViSP in my opinion.

closed 2019-08-21 by s-trinh