lagadic/visp #620: Can not run on Realsense D435 with sample-realsense.cpp LNK2019 problem on windows 10

AaronZhu2018, 2019-09-02:
I try to implement VISP on realsense D435. It has LNK2019 problem. I have try for all the day. Really need help! Thanks you! 
[image]
To reproduce my probelm
1. First I copy sample-realsense.cpp from 
https://visp-doc.inria.fr/doxygen/visp-daily/classvpRealSense2.html
2. Then I add #define VISP_HAVE_REALSENSE2 True in sample-realsense.cpp
3. I write cmakelist in order to link to realsense2.lib.  I follow the rule which this website taught 
https://github.com/zivsha/librealsense/blob/getting_started_example/examples/getting-started/CMakeLists.txt

I deduce the problem. I run all the following example in the link on windows using cmakelist
1. I had successfully run this tutorial https://visp-doc.inria.fr/doxygen/visp-2.9.0/tutorial-tracking-mb.html, so I think the setting of Visp library should be fine. I think the problem is mainly on realsense2.lib. 
2. So I try this https://github.com/zivsha/librealsense/tree/getting_started_example/examples/getting-started in order to check my realsense2.lib is working. I successfully run it. So the problem must be vpRealSense2.h. 
3. But the Cmakelist in sample-realsense.cpp success and didn't have any problem.
[image]

I really don't know what's the problem  
The following is the sample-realsense.cpp and cmakelist file.
```
#include <visp3/gui/vpDisplayGDI.h>
#include <visp3/gui/vpDisplayX.h>
#define VISP_HAVE_REALSENSE2 True
#include <visp3/sensor/vpRealSense2.h>
int main()
{
	vpRealSense2 rs;
	rs.open();
	vpImage<unsigned char> I(rs.getIntrinsics(RS2_STREAM_COLOR).height, rs.getIntrinsics(RS2_STREAM_COLOR).width);
#ifdef VISP_HAVE_X11
	vpDisplayX d(I);
#elif defined(VISP_HAVE_GDI)
	vpDisplayGDI d(I);
#endif
	while (true) {
		rs.acquire(I);
		vpDisplay::display(I);
		vpDisplay::flush(I);
		if (vpDisplay::getClick(I, false))
			break;
	}
	return 0;
}
```

cmakelist
```
#  minimum required cmake version: 3.1.0
cmake_minimum_required(VERSION 3.1.0)

project(sample-realsense)

# Save the command line compile commands in the build output
set(CMAKE_EXPORT_COMPILE_COMMANDS 1)

# Make project require C++11
include(CheckCXXCompilerFlag)
CHECK_CXX_COMPILER_FLAG("-std=c++11" COMPILER_SUPPORTS_CXX11)
CHECK_CXX_COMPILER_FLAG("-std=c++0x" COMPILER_SUPPORTS_CXX0X)
if(COMPILER_SUPPORTS_CXX11)
    set(CMAKE_C_FLAGS "${CMAKE_C_FLAGS} -std=c11")
    set(CMAKE_CXX_FLAGS "${CMAKE_CXX_FLAGS} -std=c++11")
elseif(COMPILER_SUPPORTS_CXX0X)
    set(CMAKE_CXX_FLAGS "${CMAKE_CXX_FLAGS} -std=c++0x")
endif()

# Simple non robust way to find the librealsense library
if(WIN32)
    if("${CMAKE_SIZEOF_VOID_P}" STREQUAL "8")
        set(LIBRARY_DIR "C:\\Program Files (x86)\\Intel RealSense SDK 2.0\\lib\\x64")          # TODO: Update this variable to correct path - folder where realsense2.lib is found
        set(DLL_DIR "C:\\Program Files (x86)\\Intel RealSense SDK 2.0\\bin\\x64")              # TODO: Update this variable to correct path - folder where realsense2.dll is found
    else()
        set(LIBRARY_DIR "C:\\Program Files (x86)\\Intel RealSense SDK 2.0\\lib\\x86")          # TODO: Update this variable to correct path - folder where realsense2.lib is found
        set(DLL_DIR "C:\\Program Files (x86)\\Intel RealSense SDK 2.0\\bin\\x86")              # TODO: Update this variable to correct path - folder where realsense2.dll is found
    endif()
    set(PROJECT_BINARY_OUTPUT_PATH "${PROJECT_BINARY_DIR}/\$\(Configuration\)")                # TODO: Update this variable to correct path - folder to which your project will compile
    set(ADDITIONAL_INCLUDE_DIRS "C:\\Program Files (x86)\\Intel RealSense SDK 2.0\\include")   # TODO: Update this variable to correct path - folder where librealsense2 folder is found
endif()

find_library(REALSENSE2_FOUND realsense2 HINTS ${LIBRARY_DIR} REQUIRED)
find_package(PCL 1.9.1 REQUIRED COMPONENTS common io)
if(NOT REALSENSE2_FOUND)
    SET(REALSENSE2_FOUND "realsense2")
    message(WARN "Failed to find_library(realsense2)")
endif()
# getting-started
find_package(VISP REQUIRED)
add_executable(sample-realsense sample-realsense.cpp)
include_directories(sample-realsense ${ADDITIONAL_INCLUDE_DIRS} ${VISP_INCLUDE_DIRS})
target_link_libraries(sample-realsense ${REALSENSE2_FOUND} ${VISP_LIBRARIES})
set_target_properties (sample-realsense PROPERTIES FOLDER Examples)

# Post Build script to copy realsense2.dll
if(WIN32)
message(STATUS "Adding Post build script to copy realsense2.dll to project's binary folder")
message(STATUS "Will try to copy from ${DLL_DIR} to ${PROJECT_BINARY_OUTPUT_PATH}")
add_custom_command(TARGET sample-realsense POST_BUILD        # Adds a post-build event to sample-realsense
    COMMAND ${CMAKE_COMMAND} -E copy_if_different              # which executes "cmake - E copy_if_different..."
        "${DLL_DIR}/realsense2.dll"                            # <--this is in-file
        ${PROJECT_BINARY_OUTPUT_PATH})                 # <--this is out-file path
endif()

install(
    TARGETS

    sample-realsense

    RUNTIME DESTINATION
    ${CMAKE_INSTALL_PREFIX}/bin
)
```

fspindle, 2019-09-24:
The question is how you install librealsense and which is the version you are using ?
Did you follow [this tutorial](https://visp-doc.inria.fr/doxygen/visp-daily/tutorial-install-win10-msvc15.html#install_win10_msvc15_3rdparty_realsense) ?

fspindle, 2019-11-10:
Any updates, otherwise I will close...

closed 2020-01-17 by fspindle