lagadic/visp #629: cannot run ./tutorial-grabber-realsense

sebymv, 2019-09-30:
I have already installed realsense package for ViSP. I tried to use intel realsense d435  for image capturing.
seby@seby-Lenovo-G50-80:~/visp-ws/visp-build/tutorial/grabber$ ./tutorial-grabber-realsense

However it is not working and gives an error:

Install librealsense, configure and build ViSP again to use this example

fspindle, 2019-09-30:
Your librealsense installation is not detected by ViSP.

Check your installation:
```
$ sudo updatedb
$ locate librealsense2.so
/usr/lib/x86_64-linux-gnu/librealsense2.so
/usr/lib/x86_64-linux-gnu/librealsense2.so.2.26
/usr/lib/x86_64-linux-gnu/librealsense2.so.2.26.0
```
If you have the same, it means that you forgot to run `cmake ../visp` after librealsense installation

sebymv, 2019-10-01:
@fspindle Thank you for the timely response. The issue is solved.

closed 2019-10-01 by sebymv