lagadic/visp #630: visp_vs cannot find pcl-1.8

dmkr001, 2019-09-30:
The error is:
Imported target "visp_vs" includes non-existent path

    "/usr/include/pcl-1.7"

  in its INTERFACE_INCLUDE_DIRECTORIES.  

However, my pcl is version 1.8.

dmkr001, 2019-09-30:
my cmakelists:
project(sbccr)
cmake_minimum_required(VERSION 3.0)

set(CMAKE_CXX_STANDARD 11)

find_package(VISP REQUIRED)

## Specify additional locations of header files
## Your package locations should be listed before other locations
include_directories(
# include
  ${VISP_INCLUDE_DIRS}
)

add_executable(visp_sim src/visp_simulation.cpp)
target_link_libraries(visp_sim ${VISP_LIBRARIES})

dmkr001, 2019-09-30:
project(sbccr)
cmake_minimum_required(VERSION 3.0)

set(CMAKE_CXX_STANDARD 11)

find_package(VISP REQUIRED)

## Specify additional locations of header files
## Your package locations should be listed before other locations
include_directories(
# include
  ${VISP_INCLUDE_DIRS}
)

add_executable(visp_sim src/visp_simulation.cpp)
target_link_libraries(visp_sim ${VISP_LIBRARIES})

dmkr001, 2019-09-30:
project(sbccr)
cmake_minimum_required(VERSION 3.0)

set(CMAKE_CXX_STANDARD 11)

find_package(VISP REQUIRED)

include_directories(
  ${VISP_INCLUDE_DIRS}
)

add_executable(visp_sim src/visp_simulation.cpp)
target_link_libraries(visp_sim ${VISP_LIBRARIES})

my cmakelist

fspindle, 2019-09-30:
This issue is already tracked on the [forum](https://gforge.inria.fr/forum/message.php?msg_id=153400&group_id=397).

closed 2019-10-31 by fspindle