lagadic/visp #649: Build error

pureYioYi, 2019-10-30:
Hi, I met some error when built visp in windows 10, any help are appreciated!
Build Environment：windows 10, visual studio 2017

Error when Cmake:
CMake Error at cmake/VISPUtils.cmake:185 (if):
  if given arguments:

    "D:/SDKs/C++/Visp/source/3rdparty/clipper" "MATCHES" "^D:/SDKs/C++/Visp/source"

  Regular expression "^D:/SDKs/C++/Visp/source" cannot compile
Call Stack (most recent call first):
  3rdparty/clipper/CMakeLists.txt:7 (vp_include_directories)

fspindle, 2019-10-30:
Very strange !

Could you precise:
- what ViSP version (release number, github) ?
- what cmake version ?

pureYioYi, 2019-10-30:
Thanks!  I tried on :
Visp version： 3.1.0, 3.2.0
Cmake version: 3.12, 3.15
All occurs the same error.

Meanwhile, it seems that cmake cannot found  path of the 3rdparties: 
When I try to uncheck "with CLIPPER", the error exchanged:

CMake Error at cmake/VISPUtils.cmake:203 (if):
  if given arguments:

    "D:/SDKs/C++/Visp/source/3rdparty/clapack/PRIVATE" "MATCHES" "^D:/SDKs/C++/Visp/source"

  Regular expression "^D:/SDKs/C++/Visp/source" cannot compile
Call Stack (most recent call first):
  3rdparty/clapack/CMakeLists.txt:21 (vp_target_include_directories)

Thanks again!!!

pureYioYi, 2019-10-30:
I found where the problem is! 
It is my fault that Path of the source codes could not include "++"!
I just wonder how could i success with the same path before ...