lagadic/visp #651: visp_ros  installation error

sebymv, 2019-10-31:
I tried to install visp_ros , but I am not able to build the package 
the following error occurs
``Error 2
Makefile:138: recipe for target 'all' failed
make: *** [all] Error 2
Invoking "make -j4 -l4" failed

fspindle, 2019-10-31:
`visp_ros` needs ViSP > 3.2.0.
I'm pretty sure that you are using `ros-<distro>-visp` package that corresponds to ViSP 3.2.0.

If this is the case, uninstall `ros-<distro>-visp`, and build ViSP from source coming from github/master followin [Build visp_ros from source](http://wiki.ros.org/visp_ros/Tutorials/How%20to%20do%20visual%20servoing%20with%20Parrot%20Bebop%202%20drone%20and%20visp_ros#Build_visp_ros_from_source) tutorial.

fspindle, 2019-11-04:
My last commit in `visp_ros` keep compat with ViSP 3.2.0

closed 2019-11-04 by fspindle