lagadic/visp #652: visp_ros intalltation error

makcoco, 2019-10-31:
why i had already install the pkg,but still don't find it
when i
catkin_make
it return follows:

 <CMake Error at /home/coco/catkin_ws/devel/share/visp_ros/cmake/visp_rosConfig.cmake:150 (message):
  Project 'ros2visp' tried to find library
  'libvisp_ros_bebop2_visual_servo_nodelet'.  The library is neither a target
  nor built/installed properly.  Did you compile project 'visp_ros'? Did you
  find_package() it before the subdirectory containing its code is included?
Call Stack (most recent call first):
  /opt/ros/kinetic/share/catkin/cmake/catkinConfig.cmake:76 (find_package)
  ros2visp/CMakeLists.txt:10 (find_package)
>

makcoco, 2019-10-31:
and this error
Platform: Ubuntu16.04 
ros version: kinectic 
error follows:

'
/home/coco/catkin_ws/src/visp_ros/tutorial/bebop2/visual_servo_nodelet.cpp: In member function ‘void bebopVSNodelet::imageCallback(const ImageConstPtr&)’:
/home/coco/catkin_ws/src/visp_ros/tutorial/bebop2/visual_servo_nodelet.cpp:307:34: error: ‘selectAtanOneOverRho’ is not a member of ‘vpFeatureVanishingPoint’
                                  vpFeatureVanishingPoint::selectAtanOneOverRho());
                                  ^
/home/coco/catkin_ws/src/visp_ros/tutorial/bebop2/visual_servo_nodelet.cpp:369:37: error: no matching function for call to ‘vpDisplay::displayLine(vpImage<unsigned char>&, __gnu_cxx::__alloc_traits<std::allocator<vpImagePoint> >::value_type&, __gnu_cxx::__alloc_traits<std::allocator<vpImagePoint> >::value_type&, const vpColor&, int, bool)’
                                false);
'

I has already install ros-kinecitc-visp (visp3).  Why It occur this problem.  Please HELP me . Thank you!

fspindle, 2019-10-31:
@makcoco Same as #651. Follow my recommandation, remove `ros_kinetic_visp` and build ViSP from source

closed 2019-10-31 by fspindle