lagadic/visp #656: If I use another branch robot, how I can I use package about visual-servoing?

CaffeMrDe, 2019-11-04:
I change the industry robot. how can I adapter the robot motion interface interface in visp?

Thank you!

CaffeMrDe, 2019-11-04:
> I change the industry robot. How can I adapter the robot motion interface in visp?
> 
> Thank you!

fspindle, 2019-11-09:
You need to create a new class that provide an interface with your robot in order to control your robot mainly in velocity if you are interested in visual servoing.

There is a similar work in progress for the Kinova Jaco 2 robot with PR #625

CaffeMrDe, 2019-11-11:
I have already adapt ros industriy robot package for my own robot sucessfully.  Is  fessible that I want to user ros industriy robot package to adapt visual-servoing?

closed 2021-02-17 by fspindle