lagadic/visp #658: Difficulty to link with visp_ros

gchabert, 2019-11-05:
I've installed visp_ros without any problem following [this tutorial](http://wiki.ros.org/visp_ros/Tutorials/Howto_install_visp_ros), but when I am trying to compile ``tutorial-ros-grabber.cpp`` as explained in [here](http://docs.ros.org/hydro/api/visp_ros/html/tutorial-ros-grabber.html#ros-grabber), I get the following error:

```
CMake Error at /home/gilles/catkin_ws/install/share/visp_ros/cmake/visp_rosConfig.cmake:150 (message):
  Project 'tutorial-ros-grabber' tried to find library
  'libvisp_ros_bebop2_visual_servo_nodelet'.  The library is neither a target
  nor built/installed properly. 
```

Can you help me? I'm using ros melodic under Ubuntu 18.

Sorry, I don't know much about cmake.

fspindle, 2019-11-05:
Try upgrading visp_ros

```
cd /home/gilles/catkin_ws/src/visp_ros
git pull
cd ..
catkin_make
```

gchabert, 2019-11-05:
It says 'Already up to date'

fspindle, 2019-11-06:
Which version of ViSP are you using ? and how do you install it ?

gchabert, 2019-11-06:
I first installed visp 3.2 on my system using ``apt-get install ros-melodic-visp`` (I read the version number in ``package.xml``).
Then I installed visp_ros 1.0.0 in my catkin workspace as explained [here](http://wiki.ros.org/visp_ros/Tutorials/Howto_install_visp_ros).

fspindle, 2019-11-06:
Fixed. Do a `git pull` in visp_ros folder.

closed 2019-11-06 by fspindle

gchabert, 2019-11-06:
Thanks Fabien