lagadic/visp #659: Extrinsic calibration

sebymv, 2019-11-07:
I would like to perform extrinsic calibration of an Aubo i5  6-dof robot using Visp. What are  the major modifications required  ** tutorial-franka-acquire-calib-data.cpp**  to do the same?

fspindle, 2019-11-09:
You should adapt
- image acquisition of the OpenCV calibration grid with your camera
- save the corresponding geometric model transformation; the pose of the Aubo i5 6-dof robot end-effector wrt the robot base frame

sebymv, 2019-11-09:
Thank you  Fabien

engr-zubair, 2020-01-08:
@sebymv, hi, I am also working on visual servo control of another type of arm, i want to ask , are you satisfied with the visp extrinsic calibration output eMc matrix result for your robot, or did you find an error in obtained and real position of camera w.r.t robot endeffector. ?

sebymv, 2020-01-10:
I have  used visp calibrator ros package. The results  were satisfactory as the  robot  end effector  reached  the specified  pose goal after the calibration

mbajlo, 2020-03-18:
Hello,
I am trying to make the extrinsic calibration of the robot end effector in the camera frame. Camera is mounted on the robot's end effector. Robot is simulated in the gazebo, so I cannot modify the 

> tutorial-franka-acquire-calib-data.cpp

Instead of this, I have decided to use the `tf` to get the transform between the tip of the end effector in the fixed frame of the robot (base). The pose files are 1 column 6 row matrix (translation and rotation) saved in `pose_fPe_<pair_number>.yaml` as it is done in the  `tutorial-franka-acquire-calib-data.cpp`, since the transform is between base frame of the robot and the end effector frame, the pose is also the position of the end effector frame.
To get the image which corresponds to the certain pose I used the`image_viewer` and I am saved the images as `image-<pair_number>.png`.
Image/pose pairs for each position of the robot are saved and used in the next step.

After this I have followed the procedure to compute the chessboard pose from the images and get the `pose_cPo_<pair_number>.yaml`  with this one:

> ./tutorial-chessboard-pose --square_size 0.025 --input image-%d.png

This part went just fine. The last step after acquiring end effector poses (`pose_fPe_<pair_number>.yaml`) and chessboard poses (`pose_cPo_<pair_number>.yaml`) would be to estimate end effector frame to camera frame transform. This is done by running:

> ./tutorial-hand-eye-calibration --ndata  number of images or robot poses

after this in the directory I got the eMc.yaml and it is filled with numbers, but I am worrying because  but in the terminal I got this response (there is a `Problem in solving Hand-Eye Calibration by VVS`):

> Use fPe=pose_fPe_1.yaml, cPo=pose_cPo_1.yaml
> Use fPe=pose_fPe_2.yaml, cPo=pose_cPo_2.yaml
> Use fPe=pose_fPe_3.yaml, cPo=pose_cPo_3.yaml
> Use fPe=pose_fPe_4.yaml, cPo=pose_cPo_4.yaml
> Use fPe=pose_fPe_5.yaml, cPo=pose_cPo_5.yaml
> Use fPe=pose_fPe_6.yaml, cPo=pose_cPo_6.yaml
> Use fPe=pose_fPe_7.yaml, cPo=pose_cPo_7.yaml
> Use fPe=pose_fPe_8.yaml, cPo=pose_cPo_8.yaml
> Use fPe=pose_fPe_9.yaml, cPo=pose_cPo_9.yaml
> 
>  Problem in solving Hand-Eye Calibration by VVS 
> 
> Save eMc.yaml
> 
> Output: Hand-eye calibration result: eMc estimated 
> -0.008544601404  -0.7744191954  0.6326151275  -0.2829281744
> 0.3911410716  -0.5848209708  -0.7106286613  -0.02218137314
> 0.9202910691  0.2413697202  0.3079042161  -0.3025602478
> 0  0  0  1
> theta U (deg): 80.77078334 -24.40740615 98.89010069

I am kind of stuck here because I have no idea what could have gone wrong here.

Do you know what `Problem in solving Hand-Eye Calibration by VVS` means?
Does the `pose_fPe_<pair_number>.yaml` represents the pose of the end effector and `image-<pair_number>.png` represents the what camera sees in that position?
and based on these pair (8-10 of them) I should be able to estimate the transform?

Thanks

engr-zubair, 2020-03-19:
hi , @mbajlo , It seems you are using ROS, try using easy hand eye calibration tools for camera extrinsic calibration , it is more convenient , and which robot are you using ? and how its pose is recorded by the controller also affects the procedure , do let me know .? 

 As for your answers ..
1. Does the pose_fPe_<pair_number>.yaml represents the pose of the end effector and image-<pair_number>.png represents the what camera sees in that position?

A : yes you got it right .

2. Do you know what Problem in solving Hand-Eye Calibration by VVS means?

A: I also occasionally got this error or warning , using visp calibration tools, it is apparently due to following reason ,
1. the pose and image data is not correctly paired or readable , or it is difficult to extract poses from it, try using another set of pairs , repeat experiment with different set of poses and images.
2. also you can try different combination of ndata ,,,try using ndata 6 or 7 [ although 10 or more are generally recommended for accuracy ]  , and run the same again , you may get the eMc correctly...

mbajlo, 2020-03-19:
Hi @engr-zubair ,first, thank you for the answer.
Yes I am using ROS with the Fanuc robot in the Gazebo simulation. But I wanted to follow the procedure because I often switch robots and combine variety of grippers, which is simulation all about. I have read about the easy hand eye package for extrinsic camera calibration. But I was confused with the part of the `calibration.launch`  in which I should somehow provide `tracking_base_frame` and `tracking_marker_frame`
>     
>     <arg name="tracking_base_frame" value="/optical_origin"/>
>     <arg name="tracking_marker_frame" value="/optical_target"/>
But I do not know where this frames should come from?
It should be published by some node?

Since after this I wanted to test the PBVS and IBVS in the simulation with different robots I went with the VISP instead of with the ROS package.

In my opinion when I run the:
> ./tutorial-hand-eye-calibration --ndata number of images or robot poses

the binary `tutorial-hand-eye-calibration` doesn't know where the `pose_fPe_<pair_number>.yaml` came from, only important is it consist of the relevant data populated in the yaml form. I made the small node which is gives me the `pose_fPe_<pair_number>.yaml` populated this way:

```
rows: 6
cols: 1
data:
  - [0.891184]
  - [-0.000048]
  - [0.288398]
  - [0.995847]
  - [0.000139]
  - [0.091044]
```
therefore together with the  `pose_cPo_<pair_number>.yaml` from relevant images I should get the right eMc.yaml file?
Do you think I should do more than 9 pose/image pairs and try again?
Thanks

closed 2021-02-17 by fspindle