lagadic/visp #661: vpMomentCInvariant: calculation of invariant I11

francofusco, 2019-11-16:
I believe there is a mistake in the calculation of the invariant `I11` (used to evaluate the invariant `C7` inside `vpMomentCInvariant`). The [wrong line](https://github.com/lagadic/visp/blob/4efcc0b64b9911d676ed3c021d713e21dcfee3a0/modules/core/src/tracking/moments/vpMomentCInvariant.cpp#L137) is:
```c++
I_val[11] = 3 * mu13_2 + 2 * mu31 * mu13 + (-3 * mu40 - 3 * mu04) * mu22 - 2 * mu40 * mu04 + 3 * mu31_2;
```
which, if I am not mistaken, should instead be:
 ```c++
I_val[11] = 3 * mu13_2 + 2 * mu31 * mu13 + (3 * mu40 - 3 * mu04) * mu22 - 2 * mu40 * mu04 + 3 * mu31_2;
```
(the `mu40` inside the brackets should be positive). I checked it against Omar Tahri's PhD thesis and the paper on Image Moments by Omar Thari and François Chaumette and it *should* be as I said - however, please double check it before! :sweat_smile:

If you validate this, can you please back-port the fix to v 3.1?

fspindle, 2019-11-18:
Thanks for your feedback.
I don't know which is the right equation, since the one in the code corresponds to eq 2.70 in [Manikandan Bakthavatchalam's Phd thesis](http://rainbow-doc.irisa.fr/pdf/2015_these_mani.pdf) page 60.

I will investigate with François Chaumette and let you know.

francofusco, 2019-11-18:
> I don't know which is the right equation, since the one in the code corresponds to eq 2.70 in Manikandan Bakthavatchalam's Phd thesis page 60.

Thanks for the reply, I will have a look at it!

> I will investigate with François Chaumette and let you know.

Amazing! Meanwhile, I found few other potential issues, I will list them below hoping that they are useful. I apologize in advance for possible mistakes I made below...

---
#### `vpFeatureMomentArea`: possible inconsistency in `get_s` and `interaction`

From what I can understand, the value returned by `vpFeatureMomentArea::get_s` in the case of *discrete* objects is the [sum of the centered moments](https://github.com/lagadic/visp/blob/4efcc0b64b9911d676ed3c021d713e21dcfee3a0/modules/core/src/tracking/moments/vpMomentArea.cpp#L48)

<a href="https://www.codecogs.com/eqnedit.php?latex=\mu_{2,0}&space;&plus;&space;\mu_{0,2}" target="_blank"><img src="https://latex.codecogs.com/gif.latex?\mu_{2,0}&space;&plus;&space;\mu_{0,2}" title="\mu_{2,0} + \mu_{0,2}" /></a>

since the actual "area" moment is just the number of points and would be of no use in a Servoing scheme. This absolutely makes sense, however the interaction matrix [is set as a zero-matrix](https://github.com/lagadic/visp/blob/4efcc0b64b9911d676ed3c021d713e21dcfee3a0/modules/visual_features/src/visual-feature/vpFeatureMomentArea.cpp#L65) (as if the considered moment was instead the number of points):
```c++
if (mobj.getType() == vpMomentObject::DISCRETE) {
    /*
     *  The interaction matrix for the discrete case is zero
     *  since the feature m00 is constant.
     *  Refer thesis of Omar Tahri 2005 [Section 3.4.22]
     */
    interaction_matrices[0][0][0] = 0.;
    interaction_matrices[0][0][1] = 0.;
    interaction_matrices[0][0][2] = 0.;
    interaction_matrices[0][0][3] = 0.;
    interaction_matrices[0][0][4] = 0.;
    interaction_matrices[0][0][5] = 0.;
  }
```
I think it would be worth updating either `get_s` or `compute_interaction` so that the feature and the interaction matrices are consistent. In addition, I believe that the documentation of neither `vpMomentArea` nor `vpFeatureMomentArea` mentions the fact that for discrete objects the area is not `m00` but `mu20+mu02`.

---
#### Evaluation of invariant `I5`

This issue is similar to the original one. In the paper from Omar and François the definition of the invariant is:

<a href="https://www.codecogs.com/eqnedit.php?latex=I_5&space;=&space;-&space;\mu_{30}^2&space;\mu_{03}^2&space;&plus;&space;\underline{6&space;\mu_{30}&space;\mu_{21}^3}&space;-&space;4&space;\mu_{30}&space;\mu_{12}^3&space;-&space;4&space;\mu_{21}^3&space;\mu_{03}&space;&plus;&space;3&space;\mu_{21}^2&space;\mu_{12}^2" target="_blank"><img src="https://latex.codecogs.com/gif.latex?I_5&space;=&space;-&space;\mu_{30}^2&space;\mu_{03}^2&space;&plus;&space;\underline{6&space;\mu_{30}&space;\mu_{21}^3}&space;-&space;4&space;\mu_{30}&space;\mu_{12}^3&space;-&space;4&space;\mu_{21}^3&space;\mu_{03}&space;&plus;&space;3&space;\mu_{21}^2&space;\mu_{12}^2" title="I_5 = - \mu_{30}^2 \mu_{03}^2 + \underline{6 \mu_{30} \mu_{21}^3} - 4 \mu_{30} \mu_{12}^3 - 4 \mu_{21}^3 \mu_{03} + 3 \mu_{21}^2 \mu_{12}^2" /></a>

This differs from the definition used in ViSP - like in my original issue, it is using the formula reported in the reference you provided:

<a href="https://www.codecogs.com/eqnedit.php?latex=I_5&space;=&space;-&space;\mu_{30}^2&space;\mu_{03}^2&space;&plus;&space;\underline{6&space;\mu_{30}\mu_{21}\mu_{03}\mu_{12}}&space;-&space;4&space;\mu_{30}&space;\mu_{12}^3&space;-&space;4&space;\mu_{21}^3&space;\mu_{03}&space;&plus;&space;3&space;\mu_{21}^2&space;\mu_{12}^2" target="_blank"><img src="https://latex.codecogs.com/gif.latex?I_5&space;=&space;-&space;\mu_{30}^2&space;\mu_{03}^2&space;&plus;&space;\underline{6&space;\mu_{30}\mu_{21}\mu_{03}\mu_{12}}&space;-&space;4&space;\mu_{30}&space;\mu_{12}^3&space;-&space;4&space;\mu_{21}^3&space;\mu_{03}&space;&plus;&space;3&space;\mu_{21}^2&space;\mu_{12}^2" title="I_5 = - \mu_{30}^2 \mu_{03}^2 + \underline{6 \mu_{30}\mu_{21}\mu_{03}\mu_{12}} - 4 \mu_{30} \mu_{12}^3 - 4 \mu_{21}^3 \mu_{03} + 3 \mu_{21}^2 \mu_{12}^2" /></a>

If you can confirm with François which version is supposed to be correct it would help me :+1: 

---
#### Interaction Matrix of Alpha - DISCRETE objects

This is the last potential issue I found. The angle of the main orientation axis is defined as: 

<a href="https://www.codecogs.com/eqnedit.php?latex=\alpha&space;=&space;\frac{1}{2}\mathrm{atan}\left(&space;\frac{2\mu_{11}}{\mu_{20}-\mu_{02}}&space;\right&space;)" target="_blank"><img src="https://latex.codecogs.com/gif.latex?\alpha&space;=&space;\frac{1}{2}\mathrm{atan}\left(&space;\frac{2\mu_{11}}{\mu_{20}-\mu_{02}}&space;\right&space;)" title="\alpha = \frac{1}{2}\mathrm{atan}\left( \frac{2\mu_{11}}{\mu_{20}-\mu_{02}} \right )" /></a>

whose time derivative should be

<a href="https://www.codecogs.com/eqnedit.php?latex=\dot{\alpha}&space;=&space;\frac{&space;\left(\mu_{20}-\mu_{02}\right)&space;\dot{\mu}_{11}&space;&plus;&space;\mu_{11}\left(\dot{\mu}_{02}-\dot{\mu}_{20}\right)&space;}{&space;4\mu_{11}^2&space;&plus;&space;\left(&space;\mu_{20}-\mu_{02}&space;\right&space;)^2&space;}" target="_blank"><img src="https://latex.codecogs.com/gif.latex?\dot{\alpha}&space;=&space;\frac{&space;\left(\mu_{20}-\mu_{02}\right)&space;\dot{\mu}_{11}&space;&plus;&space;\mu_{11}\left(\dot{\mu}_{02}-\dot{\mu}_{20}\right)&space;}{&space;4\mu_{11}^2&spac [... cut ...]
I checked the implementation and there are two things I can say:
- It seems to match the formula from the paper by Omar and François 
- The "mismatch" happens only for those elements that use the parameters `beta` and `gamma`.

Since the ratio is constant, my wild guess is that the parameters `beta` and `gamma` have been evaluated wrongly - perhaps they should be both halved? The reason why I believe this could be the error is that, when using *dense* objects, the matrices `La` and `Lb` are perfectly equal.

fspindle, 2019-11-19:
@francofusco Thanks again for your feedback
@chaumett I let you complete my answer if necessary

If it's ok for you to sign the [copyright transfert form](http://visp-doc.inria.fr/right-transfert/visp-right-transfer.pdf), I would appreciate a Pull Request with the following changes. It would be a nice way to contribute to ViSP.

---

**`I11` and `I5` computation**

We cannot ensure that the equations in the papers and Omar Tahri's PhD thesis are correct. The equation of `I11` and `I5` was considered as an error and modified by Manikandan Bakthavatchalam during his Phd thesis. Again, we don't know if this modification is the right one.

To know which is the right version we can only suggest that you make a simulation. You can initialize the camera plane parallel to your object and just introduce a pure rotation around optical axis. The  values of `I11` and `I5` should be always the same. Check also the interaction matrix where there should be 0 on some places.

---
**`vpFeatureMomentArea` possible inconsistency in `get_s()` and `interaction()`**

Fot `get_s()` we recommand to use the sum of the centered moments

<a href="https://www.codecogs.com/eqnedit.php?latex=\mu_{2,0}&space;&plus;&space;\mu_{0,2}" target="_blank"><img src="https://latex.codecogs.com/gif.latex?\mu_{2,0}&space;&plus;&space;\mu_{0,2}" title="\mu_{2,0} + \mu_{0,2}" /></a>
@francofusco Thanks again for your feedback

As you mention, the interaction matrix is wrong and should be  

<a href="https://www.codecogs.com/eqnedit.php?latex={\bf&space;L}_{\mu_{2,0}}&space;&plus;&space;{\bf&space;L}_{\mu_{0,2}}" target="_blank"><img src="https://latex.codecogs.com/gif.latex?{\bf&space;L}_{\mu_{2,0}}&space;&plus;&space;{\bf&space;L}_{\mu_{0,2}}" title="{\bf L}_{\mu_{2,0}} + {\bf L}_{\mu_{0,2}}" /></a>

---
**Interaction Matrix of Alpha - DISCRETE objects**

What you propose is right. We should definitively modify [vpFeatureMomentAlpha::compute_interaction()](https://github.com/lagadic/visp/blob/master/modules/visual_features/src/visual-feature/vpFeatureMomentAlpha.cpp#L94) with your code that is more general without making a difference between discrete and dense.

francofusco, 2019-11-20:
Alright, I will work on a PR to integrate the updates (and try to investigate about the invariants before). Thank you for your time!

francofusco, 2019-11-20:
I checked the problem with the C-invariants. Manikandan's version ensures invariancy to translation and planar scaling, so nothing needs to be changed inside ViSP! :+1:

chaumett, 2019-11-20:
Dear Franco,

It is more the invariance to rotation around the optical axis that is
important to check, if you have time of course.

Thanks a lot for your efforts and improvements.

François.

francofusco, 2019-11-20:
> It is more the invariance to rotation around the optical axis that is
important to check

Yes, I wrote it wrong, sorry. I am actually checking invariance to 3D translation and rotation about the optical axis of the camera. What I have done is to build a `vpHomogeneousMatrix` with time-varying elements in `tx`, `ty`, `tz` and `uz` (while `ux` and `uy` are kept null) and use it as object's pose in the camera frame. I can confirm that `C3` and `C7` remain constant during the simulation using the formulas from Manikandan.

fspindle, 2019-11-20:
Sounds ok to me

fspindle, 2019-11-28:
Fixed in PR #665

closed 2019-11-28 by fspindle