lagadic/visp #667: tutorial-apriltag-detector seg fault with

jokla, 2019-11-28:
Hi, 

I am trying the tutorial-apriltag-detector. I get a segmentation fault when I set `BEST_RESIDUAL_VIRTUAL_VS` or `HOMOGRAPHY_ORTHOGONAL_ITERATION` as pose estimation method.

I run the application with:
`./tutorial-apriltag-detector --pose_method 5`

I am under Ubuntu 18.04.3 LTS and I just cloned and built the branch master of ViSP from Github. 

This is the bt from gdb:

```
#0  0x00007ffff7bc7a3a in matd_create_scalar(double) () at /home/jokla/repo/visp/build/lib/libvisp_detection.so.3.2
#1  0x00007ffff7bc9bf7 in matd_op_recurse(char const*, int*, matd_t*, matd_t**, int*, matd_t**, int*, int) [clone .constprop.0] ()
    at /home/jokla/repo/visp/build/lib/libvisp_detection.so.3.2
#2  0x00007ffff7bca361 in matd_op(char const*, ...) () at /home/jokla/repo/visp/build/lib/libvisp_detection.so.3.2
#3  0x00007ffff7789b7a in fix_pose_ambiguities(matd_t**, matd_t**, matd_t*, matd_t*, int) () at /home/jokla/repo/visp/build/lib/libvisp_detection.so.3.2
#4  0x00007ffff778a70c in get_second_solution(matd_t**, matd_t**, apriltag_pose_t*, apriltag_pose_t*, int, double*) ()
    at /home/jokla/repo/visp/build/lib/libvisp_detection.so.3.2
#5  0x00007ffff778a953 in estimate_tag_pose_orthogonal_iteration(apriltag_detection_info_t*, double*, apriltag_pose_t*, double*, apriltag_pose_t*, int) ()
    at /home/jokla/repo/visp/build/lib/libvisp_detection.so.3.2
#6  0x00007ffff7780bf1 in vpDetectorAprilTag::Impl::getPose(unsigned long, double, vpCameraParameters const&, vpHomogeneousMatrix&, vpHomogeneousMatrix*, double*, double*) ()
    at /home/jokla/repo/visp/build/lib/libvisp_detection.so.3.2
#7  0x00007ffff778230e in vpDetectorAprilTag::Impl::detect(vpImage<unsigned char> const&, std::vector<std::vector<vpImagePoint, std::allocator<vpImagePoint> >, std::allocator<std::vector<vpImagePoint, std::allocator<vpImagePoint> > > >&, std::vector<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::allocator<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > > >&, bool, vpColor, unsigned int, std::vector<vpHomogeneousMatrix, std::allocator<vpHomogeneousMatrix> >*, std::vector<vpHomogeneousMatrix, std::allocator<vpHomogeneousMatrix> >*, std::vector<double, std::allocator<double> >*, std::vector<double, std::allocator<double> >*) ()
    at /home/jokla/repo/visp/build/lib/libvisp_detection.so.3.2
#8  0x00007ffff777e43f in vpDetectorAprilTag::detect(vpImage<unsigned char> const&, double, vpCameraParameters const&, std::vector<vpHomogeneousMatrix, std::allocator<vpHomogeneousMatrix> >&, std::vector<vpHomogeneousMatrix, std::allocator<vpHomogeneousMatrix> >*, std::vector<double, std::allocator<double> >*, std::vector<double, std::allocator<double> >*) () at /home/jokla/repo/visp/build/lib/libvisp_detection.so.3.2
#9  0x0000555555558445 in main ()

```

fspindle, 2019-11-28:
Hi @jokla, nice to hear from you 👍 

I'm able to reproduce the issue. I will investigate.

jokla, 2019-11-28:
Hi @fspindle, 
Thank you very much for the quick answer and for the fix. I tested it and now it is working!

PR #669 (2019-11-28): Fix segfault and memory leak when BEST_RESIDUAL_VIRTUAL_VS or HOMOGRA…
…PHY_ORTHOGONAL_ITERATION pose estimation method are used. Closes: #667
commit 7631540c (2019-11-28): Fix segfault and memory leak when BEST_RESIDUAL_VIRTUAL_VS or HOMOGRAPHY_ORTHOGONAL_ITERATION pose estimation method are used. Closes: #667
closed 2019-11-28 by fspindle