lagadic/visp #677: Format of image name not recognized in calibration calibration

gchabert, 2019-12-20:
I'm trying to use the camera_calibration.cpp example (which apparently was renamed calibrate_camera.cpp): https://github.com/lagadic/visp/blob/master/example/calibration/calibrate-camera.cpp

I have a folder with PNG images and set the pattern in the `Input` option of the configuration file using the '%' wildcard as said in the documentation:
``Input: [some-path]/cam1-chess%.png``
But the program fails with the message *Format of image name wasn't recognized*.
I also tried with:
``Input: [some-path]/cam1-chess*.png``
and
``Input: [some-path]/cam1-chess1.png``  (to look for a single file)
: same issue.

fspindle, 2019-12-20:
It should be `Input: [some-path]/cam1-chess%d.png` or `Input: [some-path]/cam1-chess%02d.png`

gchabert, 2019-12-20:
Oops, silly me.
Thanks for the quick answer Fabien.

closed 2019-12-20 by gchabert