lagadic/visp #697: Compile errors when using with OpenCV 4

lazydroid, 2020-02-03:
Seems like `IplImage` is deprecated in OpenCV 4 or something alike that gives me compile errors when trying to use VISP in my project:

```
/usr/include/x86_64-linux-gnu/visp3/core/vpImageConvert.h:128:29: error: ‘IplImage’ does not name a type; did you mean ‘vpImage’?
```

fspindle, 2020-02-03:
I'm unable to reproduce your issue with OpenCV and VISP build from source using current version from github for both.

OpenCV 4.2.0 doesn't deprecate IplImage.

Which version of ViSP are you using ?
Do you install ViSP from source or from a package ?

lazydroid, 2020-02-04:
I have installed VISP from the package manage on Ubuntu 18, had the error. Then I have cloned the latest from github and compiled myself -- the compilation of the VISP went well, but when I'm trying to use it in another project -- I get the before mentioned error message coming from vpImageConvert.h

fspindle, 2020-02-04:
It seems that you don't use the version build from source.

This could be seen using:
```
$ sudo updatedb
$ locate VISPConfig.cmake
```
You should see 2 times this file.

If I'm right remove installation from package

lazydroid, 2020-02-04:
ok, seems like my fault. somehow the order of the include files makes a difference between the clean compilation and a bunch of obscure errors that are not actually errors, but just a fluke.

I'm closing this issue, sorry for disturbance.

closed 2020-02-04 by lazydroid