lagadic/visp #713: 3.3.0 build fails on non-linux systems: fatal error: 'linux/serial.h' file not found

yurivict, 2020-02-17:
```
--- 3rdparty/qbdevice/CMakeFiles/visp_qbdevice.dir/all ---
/wrkdirs/usr/ports/misc/visp/work/visp-3.3.0/3rdparty/qbdevice/qbmove_communications.cpp:68:14: fatal error: 'linux/serial.h' file not found
    #include <linux/serial.h>
             ^~~~~~~~~~~~~~~~
1 error generated.
```

FreeBSD 12.

fspindle, 2020-02-18:
Can you update the locate database to see where the `linux.h` header is:

```
# /usr/libexec/locate.updatedb
# locate serial.h
```

yurivict, 2020-02-18:
FreeBSD doesn't have the ```linux.h``` header.

fspindle, 2020-02-18:
Does ViSP build if you comment the line in `qbmove_communication.cpp`?

yurivict, 2020-02-18:
> Does ViSP build if you comment the line in qbmove_communication.cpp?

No, because there are several constants that are defined in linux.h.

fspindle, 2020-02-18:
OK, I need to install a FreeBSD VM to investigate on my side.

In the mean time, a quick fix is to configure ViSP with -DWITH_QBDEVICE=OFF cmake option

yurivict, 2020-02-18:
I used ```-DWITH_QBDEVICE=OFF```, thanks.

commit 70b5e86b (2020-02-18): Fix FreeBSD build. Closes #713.
closed 2020-02-18 by fspindle