lagadic/visp #748: INTERFACE_INCLUDE_DIRECTORIES property contains path ...

TFisichella, 2020-04-28:
Hi,

I have followed this tutorial to install Visp with OpenCV-4.3.0 and Eigen-3.3.7 on Windows 10 : https://visp-doc.inria.fr/doxygen/visp-daily/tutorial-install-win10-msvc16.html#install_win10_msvc16_ws
I installed the 3rd party librairies without problem but I don't manage to build visp with those libraries.
This is the message I get during the Cmake generate : 

CMake Error in modules/core/CMakeLists.txt:
  Target "visp_core" INTERFACE_INCLUDE_DIRECTORIES property contains path:

    "E:/SUGAR/plugins/visp/opencv-4.3.0/build-vc16"

  which is prefixed in the source directory.

CMake Error in modules/core/CMakeLists.txt:
  Target "visp_core" INTERFACE_INCLUDE_DIRECTORIES property contains path:

    "E:/SUGAR/plugins/visp/opencv-4.3.0/include"

  which is prefixed in the source directory.

CMake Error in modules/core/CMakeLists.txt:
  Target "visp_core" INTERFACE_INCLUDE_DIRECTORIES property contains path:

    "E:/SUGAR/plugins/visp/eigen-3.3.7/build-vc16/install/include/eigen3"

  which is prefixed in the source directory.

Can anyone help me to fix this issue pls ?

TFisichella, 2020-04-30:
Hi, 
I'm using Visp 3.3.1

TFisichella, 2020-05-05:
Hi,
Any idea of what could be the problem ?

fspindle, 2020-05-05:
After a quick googling, it sound that this cmake issue comes from an in-source build of OpenCV and Eigen.

For OpenCV if you followed the [tutorial](https://visp-doc.inria.fr/doxygen/visp-daily/tutorial-install-win10-msvc16.html#install_win10_msvc16_3rdparty_opencv):
- build folder should be `E:/SUGAR/plugins/visp/opencv-4.3.0/build-vc16` (sounds the case)
- installation folder should be `E:/SUGAR/plugins/visp/opencv-4.3.0/build` (sounds not the case)
- includes should be found in `E:/SUGAR/plugins/visp/opencv-4.3.0/build/include`
- `OpenCV_DIR` should be set to `E:/SUGAR/plugins/visp/opencv-4.3.0/build`

If you followed the tutorial you should find `E:/SUGAR/plugins/visp/opencv-4.3.0/build/x64/vc16/bin/opencv_world430.dll`. Is this the case ?
Check also your `OpenCV_DIR` value

Same for Eigen.

TFisichella, 2020-05-05:
Hi, 
Thx, I finally managed to install it properly by installing openCV and Eigen outside of the Visp directory.

closed 2020-05-06 by fspindle