lagadic/visp #756: bad allocation exception thrown when trying to instantiate a new vpDisplayOpenCV object

DimitarDevP, 2020-05-07:
I built VISP with OpenCV and I am trying to test out the simple KLT tracker example given in the tutorials folder. 

When i compiled the solution, I got an error on the following line:
`vpDisplayOpenCV d(I, frame.rows, frame.cols, "Klt tracking");`
which lead me to a `bad_alloc` exception, so I wrapped it around a try/catch block and printed the exception:
```
try {
    vpDisplayOpenCV d(img, frame.rows, frame.cols, "Klt tracking");
}
catch (std::bad_alloc& exception) {
	std::cout << exception.what();
}
```
which as one would expect printed out `bad allocation`.

I have dynamically linked all the libraries, added all the relative include paths to the linker and got no connection issues. 
I dont have any issues with the OpenCV build or the camera. I tested just using `imshow` to render `frame `from which `img `is generated.

DimitarDevP, 2020-05-08:
UPDATE: When i try to instantiate a `vpDisplayOpenCV` object without overloading any parameters, it doesnt give me an error but it doesnt work as expected. 

I tried allocating the vpDisplayOpenCV instance on the heap to see if that will solve the stack fill-up but it did not resolve the issue since the object itself isn't the problem, the problem is what happens inside the constructor body.

My guess is that there is something in the overloaded constructor where the vpImage is passed that loads up the stack and causes bad_alloc exception to be thrown.

s-trinh, 2020-05-08:
Can you try the following code?

```
#include <visp3/gui/vpDisplayOpenCV.h>

int main()
{
  vpImage<unsigned char> I(480, 640);

  vpDisplayOpenCV d(I, 0, 0, "Klt tracking");

  vpDisplay::display(I);
  vpDisplay::flush(I);
  vpDisplay::getClick(I);

  return 0;
}
```

If you still have an issue, more information are needed:

- paste the content in the `ViSP-third-party.txt` file (located in the ViSP build directory)
- OS version, platform, ...
- image size: `frame.rows`, `frame.cols`
- if you are on Linux, maybe paste the output of `ldd tutorial-klt-tracker`

DimitarDevP, 2020-05-11:
I tried out that code snippet, i still got the same exception. 
I am running the following:

OS: Windows 10 LTSC 64 bit
Platform: Visual Studio Community Edition 2019

The image size is 480x640 - the camera resolution. 

Here is the VISP-third-party.txt output:

```
==========================================================
General configuration information for ViSP 3.3.1

  Version control:               3.2.0-967-g3353a7c0b

  Platform:
    Timestamp:                   2020-05-06T09:34:17Z
    Host:                        Windows 10.0.17763 AMD64
    CMake:                       3.17.2
    CMake generator:             Visual Studio 16 2019
    CMake build tool:            C:/Program Files (x86)/Microsoft Visual Studio/2019/Community/MSBuild/Current/Bin/MSBuild.exe
    MSVC:                        1925

  C/C++:
    Built as dynamic libs?:      yes
    C++ Compiler:                C:/Program Files (x86)/Microsoft Visual Studio/2019/Community/VC/Tools/MSVC/14.25.28610/bin/Hostx64/x64/cl.exe  (ver 19.25.28614.0)
    C++ flags (Release):         /DWIN32 /D_WINDOWS /W3 /GR /EHsc /MP8   /EHa -openmp /Gy /bigobj /MD /O2 /Ob2 /DNDEBUG
    C++ flags (Debug):           /DWIN32 /D_WINDOWS /W3 /GR /EHsc /MP8   /EHa -openmp /Gy /bigobj /MDd /Zi /Ob0 /Od /RTC1
    C Compiler:                  C:/Program Files (x86)/Microsoft Visual Studio/2019/Community/VC/Tools/MSVC/14.25.28610/bin/Hostx64/x64/cl.exe
    C flags (Release):           /DWIN32 /D_WINDOWS /W3   /MP8   /EHa -openmp /MD /O2 /Ob2 /DNDEBUG
    C flags (Debug):             /DWIN32 /D_WINDOWS /W3   /MP8   /EHa -openmp /MDd /Zi /Ob0 /Od /RTC1
    Linker flags (Release):      /machine:x64 /INCREMENTAL:NO
    Linker flags (Debug):        /machine:x64 /debug /INCREMENTAL /ignore:4099

  ViSP modules:
    To be built:                 core gui imgproc io java_bindings_generator klt me sensor ar blob robot visual_features vs vision detection mbt tt tt_mi
    Disabled:                    -
    Disabled by dependency:      -
    Unavailable:                 java

  Windows RT support:            no

  Python (for build):            C:/Users/Dimitar/AppData/Local/Programs/Python/Python38/python.exe

  Java:                          
    ant:                         NO
    JNI:                         NO

  Build options: 
    Build deprecated:            yes
    Build with moment combine:   no

  Mathematics: 
    Blas/Lapack:                 yes
    \- Use MKL:                  no
    \- Use OpenBLAS:             no
    \- Use Atlas:                no
    \- Use Netlib:               no
    \- Use GSL:                  no
    \- Use Lapack (built-in):    yes (ver 3.2.1)
    Use Eigen3:                  yes (ver 3.3.7)
    Use OpenCV:                  yes (ver 4.2.0)

  Simulator: 
    Ogre simulator: 
    \- Use Ogre3D:               no
    \- Use OIS:                  no
    Coin simulator: 
    \- Use Coin3D:               no
    \- Use SoWin:                no
    \- Use SoXt:                 no
    \- Use SoQt:                 no
    \- Use Qt5:                  no
    \- Use Qt4:                  no
    \- Use Qt3:                  no

  Media I/O: 
    Use JPEG:                    no
    Use PNG:                     no
    \- Use ZLIB:                 no
    Use OpenCV:                  yes (ver 4.2.0)
    Use stb_image (built-in):    no

  Real robots: 
    Use Afma4:                   no
    Use Afma6:                   no
    Use Franka:                  no
    Use Viper650:                no
    Use Viper850:                no
    Use aria (Pioneer):          no
    Use PTU46:                   no
    Use Biclops PTU:             no
    Use Flir PTU SDK:            no
    Use Parrot ARSDK:            no
    \-Use ffmpeg:                no
    Use Virtuose:                no
    Use qbdevice (built-in):     yes (ver 2.6.0)
    Use takktile2 (built-in):    no

  GUI: 
    Use X11:                     no
    Use GTK:                     no
    Use OpenCV:                  yes (ver 4.2.0)
    Use GDI:                     yes
    Use Direct3D:                no

  Cameras: 
    Use DC1394-2.x:              no
    Use CMU 1394:                no
    Use V4L2:                    no
    Use directshow:              no
    Use OpenCV:                  yes (ver 4.2.0)
    Use Flycapture:              yes
    Use Pylon:                   yes (ver 5.0.12)

  RGB-D sensors: 
    Use Realsense:               no
    Use Realsense2:              yes (ver 2.33.1)
    Use Kinect:                  no
    \- Use libfreenect:          no
    \- Use libusb-1:             no
    \- Use pthread:              no
    Use PCL:                     no
    \- Use VTK:                  no

  F/T sensors: 
    Use atidaq (built-in):       no
    Use comedi:                  no
    Use IIT SDK:                 no

  Detection: 
    Use zbar:                    no
    Use dmtx:                    no
    Use AprilTag (built-in):     yes (ver 3.1.1)
    \- Use AprilTag big family:  no

  Misc: 
    Use Clipper (built-in):      yes (ver 6.4.2)
    Use pugixml (built-in):      yes (ver 1.9.0)
    Use libxml2:                 no

  Optimization: 
    Use OpenMP:                  yes
    Use pthread:                 no
    Use pthread (built-in):      yes (ver 3.0.1)
    Use cxx standard:            11

  Documentation: 
    Use doxygen:                 no

  Tests and samples:
    Use catch2 (built-in):       yes (ver 2.9.2)
    Tests:                       yes
    Demos:                       yes
    Examples:                    yes
    Tutorials:                   yes

  Install path:                  C:/visp-ws/visp/visp-build-vc16/install

==========================================================

```

fspindle, 2020-05-11:
Very strange.

Can you try using GDI instead of OpenCV:

```
#include <visp3/gui/vpDisplayGDI.h>

int main()
{
  vpImage<unsigned char> I(480, 640);

  vpDisplayGDI d(I, 0, 0, "Klt tracking");

  vpDisplay::display(I);
  vpDisplay::flush(I);
  vpDisplay::getClick(I);

  return 0;
}
```

DimitarDevP, 2020-05-11:
With that i get an `std::length_error`

Last time i got this error, I was compiling a shader which had string length of more then `std::string.max_length()` characters. 

As expected, i printed the `exception.what()` and got `string too long` message.

fspindle, 2020-05-11:
I have never seen that. 

Some ideas:
1. Run GUI test:
    ```
    visp-build-vc16/modules/gui/Release/testDisplays -l
    visp-build-vc16/modules/gui/Release/testDisplays
    ```

2. If one of the previous test is not working, copy visp dll (`visp-build-vc16/install/x64/vc16/bin/*.dll`) in `visp-build-vc16/modules/gui/Release` folder and run the previous tests again

3. If you continue to catch the exception, check your PATH environment var, especially if you have multiple VISP or OpenCV build folders. My guess is that you build ViSP with a given version of (ViSP and/or OpenCV), but when you run the binaries you are using an other version that is found thanks to the PATH env var

4. Finally, if it is not better, I would suggest to do a fresh build turning all 3rd parties off appart GDI and testing the code snippet with GDI.

    - delete `visp_build-vc16` folder content
    - configure without OpenCV, Eigen, Qbdevice, pylon and realsense:
        ```
        cd visp_build-vc16
        cmake ../visp -DUSE_OPENCV=OFF -DUSE_EIGEN3=OFF -DWITH_QBDEVICE=OFF \
              -DUSE_PYLON=OFF -DUSE_REALSENSE2=OFF
        ```
    - build ViSP
    - do also a fresh snippet config and build

DimitarDevP, 2020-05-11:
I built VISP without any 3rd party libraries and the example using the GDI display works perfectly.

fspindle, 2020-05-11:
This is a good news.
Try now to introduce each 3rd party one after the other, starting with OpenCV. 
Let me know your results

DimitarDevP, 2020-05-11:
I built it with Eigen and OpenCV on second run, it works just fine. 

I honestly dont know what went wrong with previous build. I didnt see any errors in cmake or visual studio build.

closed 2020-05-11 by DimitarDevP