lagadic/visp #759: VISP built from source with the openCV 4.3.0-dev

mbajlo, 2020-05-11:
Hello guys,

Recently I have installed the Ubuntu 18.04. on my new PC and I started installing the software I had on my old PC. I have cloned, built and installed the newest `openCV` with the `openCV contrib`:

```
Python 3.6.9 (default, Apr 18 2020, 01:56:04) 
[GCC 8.4.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import cv2
>>> cv2.__version__
'4.3.0-dev'
>>> 
```

After successful installation I have cloned the VISP and tried to build it from the source(followed instruction on the from the [documentation),](https://visp-doc.inria.fr/doxygen/visp-daily/tutorial-install-ubuntu.html) but I got stuck here:

> sudo make -j8
> [  0%] Built target man
> [  0%] Built target visp_pugixml
> [  0%] Built target visp_clipper
> [  0%] Built target developer_scripts
> [  0%] Built target visp_reflex_takktile2
> [  0%] Built target visp_qbdevice
> [  3%] Built target visp_apriltag
> [ 11%] Built target visp_core
> [ 12%] Built target visp_me
> [ 12%] Built target visp_io
> [ 12%] Built target visp_klt
> [ 13%] Built target visp_imgproc
> [ 13%] Built target visp_blob
> [ 14%] Built target visp_gui
> [ 15%] Built target visp_sensor
> [ 15%] Built target visp_ar
> [ 15%] Built target testImgproc
> [ 15%] Built target testConnectedComponents
> [ 15%] Built target testFloodFill
> [ 16%] Built target testContours
> [ 16%] Built target testAutoThreshold
> [ 16%] Built target testUndistortImage
> [ 16%] Built target testMath
> [ 16%] Built target testThread2
> [ 16%] Built target testArray2D
> [ 16%] Built target testEigenConversion
> [ 16%] Built target testKalmanAcceleration
> [ 16%] Built target testColVector
> [ 19%] Built target visp_visual_features
> [ 19%] Built target testRotation
> [ 20%] Built target testCPUFeatures
> [ 20%] Built target perfMatrixMultiplication
> [ 20%] Built target testImageOwnership
> [ 20%] Built target testCameraParametersConversion
> [ 21%] Built target perfImageWarp
> [ 21%] Built target testTime
> [ 21%] Built target testMatrixConditionNumber
> [ 21%] Built target perfColorConversion
> [ 21%] Built target testImageWarp
> [ 21%] Built target testIoPPM
> [ 21%] Built target testMatrixPseudoInverse
> [ 21%] Built target testImagePrint
> [ 22%] Built target testPerformanceLUT
> [ 22%] Built target testImageMorphology
> [ 22%] Built target testImageResize
> [ 23%] Built target testIoPGM
> [ 23%] Built target testXmlParserCamera
> [ 23%] Built target testCrop
> [ 23%] Built target testEndian
> [ 23%] Built target testImageBinarise
> [ 24%] Built target testMatrix
> [ 24%] Built target testCropAdvanced
> [ 24%] Built target testImageAddSub
> [ 24%] Built target testImageGetValue
> [ 24%] Built target testImageComparison
> [ 25%] Built target testImageDifference
> [ 25%] Built target testKalmanVelocity
> [ 25%] Built target testConversion
> [ 26%] Built target testReadImage
> [ 26%] Built target testImageDraw
> [ 26%] Built target testPoseVector
> [ 26%] Built target testImageNormalizedCorrelation
> [ 26%] Built target testImageFilter
> [ 26%] Built target testMatrixDeterminant
> [ 26%] Built target testMatrixConvolution
> [ 26%] Built target testMatrixException
> [ 26%] Built target testRect
> [ 26%] Built target testSerialRead
> [ 26%] Built target testQuaternion
> [ 26%] Built target testConvert
> [ 27%] Built target testMatrixInverse
> [ 27%] Built target testRobust
> [ 27%] Built target testImageTemplateMatching
> [ 27%] Built target testTranslationVector
> [ 27%] Built target testMomentAlpha
> [ 27%] Built target testXmlParserHomogeneousMatrix
> [ 28%] Built target testRowVector
> [ 28%] Built target testRand
> [ 28%] Built target testSvd
> [ 28%] Built target testMatrixInitialization
> [ 29%] Built target testTwistMatrix
> [ 29%] Built target testSerialWrite
> [ 29%] Built target testClient
> [ 29%] Built target testServer
> [ 29%] Built target testUDPServer
> [ 29%] Built target perfMatrixTranspose
> [ 29%] Built target testIoTools
> [ 29%] Built target testPolygon
> [ 30%] Built target testImagePoint
> [ 30%] Built target testBuildInformation
> [ 30%] Built target testHistogram
> [ 30%] Built target testXmlParserRectOriented
> [ 30%] Built target testMutex
> [ 30%] Built target testUDPClient
> [ 30%] Built target testVideoDeviceDual
> [ 31%] Built target testThread
> [ 31%] Built target testXmlParser
> [ 32%] Built target testDisplayPolygonLines
> [ 32%] Built target testVideoDevice
> [ 32%] Built target testDisplayRoi
> [ 32%] Built target testDisplays
> [ 32%] Built target testClick
> [ 32%] Built target testMouseEvent
> [ 32%] Built target testNurbs
> [ 32%] Built target test1394TwoGrabber
> [ 32%] Built target testDisplayScaled
> [ 32%] Built target testRealSense2_D435
> [ 32%] Built target testForceTorqueAti
> [ 32%] Built target testForceTorqueAtiNetFTSensor
> [ 32%] Built target testRealSense2_D435_opencv
> [ 33%] Built target testRealSense_SR300
> [ 34%] Built target testForceTorqueIitSensor
> [ 34%] Built target testPylonGrabber
> [ 34%] Built target testRealSense2_D435_pcl
> [ 34%] Built target testRealSense2_SR300
> [ 34%] Built target testComedi
> [ 34%] Built target testRealSense_R200
> [ 34%] Built target testRealSense2_D435_align
> [ 34%] Built target test1394TwoResetBus
> [ 34%] Built target testTrackDot
> [ 34%] Built target testPoint
> [ 34%] Built target visp_vs
> [ 34%] Building CXX object modules/vision/CMakeFiles/visp_vision.dir/src/key-point/vpKeyPoint.cpp.o
> [ 39%] Built target visp_robot
> [ 39%] Built target testFeatureSegment
> [ 40%] Built target testFeatureMoment
> [ 40%] Built target testFeature
> /home/ros-industrial/Desktop/VISP_github/visp/modules/vision/src/key-point/vpKeyPoint.cpp: In member function ‘void vpKeyPoint::initDetector(const string&)’:
> /home/ros-industrial/Desktop/VISP_github/visp/modules/vision/src/key-point/vpKeyPoint.cpp:2272:66: **error: ‘cv::xfeatures2d::SIFT’ has not been declared
>      cv::Ptr<cv::FeatureDetector> siftDetector = cv::xfeatures2d::SIFT::create();**
>                                                                   ^~~~
> /home/ros-industrial/Desktop/VISP_github/visp/modules/vision/src/key-point/vpKeyPoint.cpp: In member function ‘void vpKeyPoint::initExtractor(const string&)’:
> /home/ros-industrial/Desktop/VISP_github/visp/modules/vision/src/key-point/vpKeyPoint.cpp:2450:52: error: **‘cv::xfeatures2d::SIFT’ has not been declared
>      m_extractors[extractorName] = cv::xfeatures2d::SIFT::create();**
>                                                     ^~~~
> **modules/vision/CMakeFiles/visp_vision.dir/build.make:326: recipe for target 'modules/vision/CMakeFiles/visp_vision.dir/src/key-point/vpKeyPoint.cpp.o' failed
> make[2]: *** [modules/vision/CMakeFiles/visp_vision.dir/src/key-point/vpKeyPoint.cpp.o] Error 1
> CMakeFiles/Makefile2:9784: recipe for target 'modules/vision/CMakeFiles/visp_vision.dir/all' failed
> make[1]: *** [modules/vision/CMakeFiles/visp_vision.dir/all] Error 2
> Makefile:140: recipe for target 'all' failed
> make: *** [all] Error 2**
> 

I seems I got stuck on the `visp_vision` module and it can't find the openCV's `xfeatures2d` but then I ran the `locate` function to check `xfeatures2d` is present and it found it:

> /usr/local/include/opencv4/opencv2/xfeatures2d
> /usr/local/include/opencv4/opencv2/xfeatures2d.hpp
> /usr/local/include/opencv4/opencv2/xfeatures2d/cuda.hpp
> /usr/local/include/opencv4/opencv2/xfeatures2d/nonfree.hpp
> /usr/local/lib/libopencv_xfeatures2d.so
> /usr/local/lib/libopencv_xfeatures2d.so.4.3
> /usr/local/lib/libopencv_xfeatures2d.so.4.3.0

Do you know what am I doing wrong here?
What could be the problem because I am kind of stuck here.

Thanks

s-trinh, 2020-05-11:
Since the patent for SIFT expired recently, SIFT features moved from `opencv_contrib` to the main `opencv` repository.

To fix the issue, I think that updating
```
cv::xfeatures2d::SIFT
```
with
```
cv::SIFT
```
should solve the issue.

mbajlo, 2020-05-12:
Hello @s-trinh,
Thanks for the reply, this got me through the build, but I had to configure it again and build it few times. 
Anyway, after I tried to run some binaries I got the errror, for example in the 
`tutorial-chessboard-pose`:

`./tutorial-chessboard-pose: error while loading shared libraries: libopencv_core.so.4.3: cannot open shared object file: No such file or directory
`

but the library definitely exists:
```
locate libopencv_core.so.4.3
/home/ros-industrial/Desktop/openCV/build/lib/libopencv_core.so.4.3
/home/ros-industrial/Desktop/openCV/build/lib/libopencv_core.so.4.3.0
/usr/local/lib/libopencv_core.so.4.3
/usr/local/lib/libopencv_core.so.4.3.0

```

maybe this is not a VISP related issue, but on the another hand, maybe someone has had this before.

Thanks

fspindle, 2020-05-12:
Use `LD_LIBRARY_PATH` to help finding these libraries:
```
$export LD_LIBRARY_PATH=$LD_LIBRARY_PATH:/usr/local/lib
```

mbajlo, 2020-05-12:
Hello @fspindle  thanks for the reply,

I have found the way to fix it through the `/etc/ld.so.conf.d/opencv.conf`.
But now when I run the `./tutorial-chessboard-pose --square_size 0.025 --input image-%d.png` I get these, even though the images of the chessboard are in the same folder:
```
Parameters:
chessboard_width=9
chessboard_height=6
chessboard_square_size=0.025
input_filename=image-%d.png
intrinsic_file=camera.xml
camera_name=Camera
Found camera with name: "Camera"
cam:
Camera parameters for perspective projection with distortion:
  px = 945.590452	 py = 949.0459782
  u0 = 397.8410476	 v0 = 396.2899592
  kud = -0.002582655301
  kdu = 0.002536841551
```
I will try to debug it, but maybe it is worth mentioning I tried to build it again, but each time I have these errors, after which I try again and the error happens again just further in the process until I get to the end (100%) and it is completely built, but I do not know if this affects the built binaries:

> $ make -j8
> [  0%] Built target developer_scripts
> [  0%] Built target visp_clipper
> [  0%] Built target visp_qbdevice
> [  0%] Built target man
> [  0%] Built target visp_pugixml
> [  0%] Built target visp_reflex_takktile2
> [  3%] Built target visp_apriltag
> [ 11%] Built target visp_core
> [ 11%] Built target visp_klt
> [ 11%] Built target visp_blob
> [ 12%] Built target visp_me
> [ 12%] Built target visp_io
> [ 13%] Built target visp_gui
> [ 14%] Built target visp_imgproc
> [ 14%] Built target testThread2
> [ 14%] Built target testMath
> [ 14%] Built target visp_ar
> [ 14%] Built target testEigenConversion
> [ 14%] Built target testColVector
> [ 15%] Built target visp_sensor
> [ 15%] Built target testUndistortImage
> [ 18%] Built target visp_visual_features
> [ 18%] Built target testKalmanAcceleration
> [ 18%] Built target testArray2D
> [ 19%] Built target testCPUFeatures
> [ 19%] Built target testMatrixConditionNumber
> [ 19%] Built target perfMatrixMultiplication
> [ 19%] Built target testRotation
> [ 19%] Built target testImageOwnership
> [ 20%] Built target perfImageWarp
> [ 20%] Built target testTime
> [ 20%] Built target testCameraParametersConversion
> [ 20%] Built target testImageResize
> [ 20%] Built target perfColorConversion
> [ 20%] Built target testIoPPM
> [ 20%] Built target testImageWarp
> [ 20%] Built target testImagePrint
> [ 20%] Built target testMatrixPseudoInverse
> [ 20%] Built target testImageMorphology
> [ 20%] Built target testXmlParserCamera
> [ 21%] Built target testPerformanceLUT
> [ 22%] Built target testMatrix
> [ 22%] Built target testImageBinarise
> [ 23%] Built target testIoPGM
> [ 23%] Built target testEndian
> [ 23%] Built target testCrop
> [ 23%] Built target testCropAdvanced
[... 114 lines cut ...]
> [ 42%] Building CXX object modules/mbt/CMakeFiles/visp_mbt.dir/src/hybrid/vpMbEdgeKltTracker.cpp.o
> [ 42%] Linking CXX executable testAfma4
> ../../lib/libvisp_vision.so.3.3.1: file not recognized: File truncated
> collect2: error: ld returned 1 exit status
> modules/robot/CMakeFiles/testAfma4.dir/build.make:514: recipe for target 'modules/robot/testAfma4' failed
> make[2]: *** [modules/robot/testAfma4] Error 1
> CMakeFiles/Makefile2:7630: recipe for target 'modules/robot/CMakeFiles/testAfma4.dir/all' failed
> make[1]: *** [modules/robot/CMakeFiles/testAfma4.dir/all] Error 2
> [ 42%] Building CXX object modules/mbt/CMakeFiles/visp_mbt.dir/src/klt/vpMbKltTracker.cpp.o
> [ 42%] Building CXX object modules/mbt/CMakeFiles/visp_mbt.dir/src/klt/vpMbtDistanceKltCylinder.cpp.o
> [ 43%] Building CXX object modules/mbt/CMakeFiles/visp_mbt.dir/src/klt/vpMbtDistanceKltPoints.cpp.o
> x86_64-linux-gnu-g++-7: internal compiler error: Killed (program cc1plus)
> Please submit a full bug report,
> with preprocessed source if appropriate.
> See <file:///usr/share/doc/gcc-7/README.Bugs> for instructions.
> modules/mbt/CMakeFiles/visp_mbt.dir/build.make:302: recipe for target 'modules/mbt/CMakeFiles/visp_mbt.dir/src/hybrid/vpMbEdgeKltTracker.cpp.o' failed
> make[2]: *** [modules/mbt/CMakeFiles/visp_mbt.dir/src/hybrid/vpMbEdgeKltTracker.cpp.o] Error 4
> make[2]: *** Waiting for unfinished jobs....
> CMakeFiles/Makefile2:10242: recipe for target 'modules/mbt/CMakeFiles/visp_mbt.dir/all' failed
> make[1]: *** [modules/mbt/CMakeFiles/visp_mbt.dir/all] Error 2
> Makefile:140: recipe for target 'all' failed
> make: *** [all] Error 2
> 

Thanks

mbajlo, 2020-05-12:
Hello,

After debugging the `tutorial-chessboard-pose.cpp` the problem is it doesn't go into the while loop, because the `reader.end()` is true before it comes to the [while loop](https://github.com/lagadic/visp/blob/8e8179d5c9df001c27d2cea20ed3747477215708/tutorial/calibration/tutorial-chessboard-pose.cpp#L106). The `image-%d.png` image files are in the same directory and there are 8 images, so the reader input shouldn't be empty.
Do you know why this happens?

fspindle, 2020-05-13:
Use `./tutorial-chessboard-pose --square_size 0.025 --input image-%02d.png`

mbajlo, 2020-05-13:
I tried to remove everything and start from the beginning and it looks like it works perfectly. Probably something went wrong through the last installation. Thanks a lot for your support.

closed 2020-05-13 by mbajlo