lagadic/visp #761: VISP build and openCV version selection

mbajlo, 2020-05-17:
Hello,

I have installed the Ubuntu 18.04 on the new PC. Then I have installed the `openCV 4.3` from source, because I wanted to try some new things. Just to let you know there is already `openCV 3.2` preinstalled on the Ubuntu 18.04. 
So as I said, I installed `openCV4.3` and after that I have cloned the `VISP` and configured it, `VISP` automatically recognized the `openCV4.3` and not `openCV3.2`, is it because `openCV4.3` has been installed in `usr/local/` or for some other reason?

Then I managed to build my code which was working on the old PC, but the problem happens when I try to detect the object using the:
`bool objectFound = _keypoint_detection.matchPoint(_vpImageCamera, _cam, _cMo, _detectionError, _elapsedTime);`

I get the error:

> OpenCV Error: Assertion failed (tlsSlots.size() > slotIdx) in releaseSlot, file /build/opencv-L2vuMj/opencv-3.2.0+dfsg/modules/core/src/system.cpp, line 1092
> terminate called after throwing an instance of 'cv::Exception'
>   what():  /build/opencv-L2vuMj/opencv-3.2.0+dfsg/modules/core/src/system.cpp:1092: error: (-215) tlsSlots.size() > slotIdx in function releaseSlot

I would say there is some kind of mixing of the `openCV` versions.

you can see in the call stack at the end of the text there is a `libopencv_core.so.3.2!cv::TLSDataContainer::release() (Unknown Source:0)` and the source is unknown.

Do you have any idea why this happens?
Why there is a `openCV 3.2` library even used when the `VISP` is built with the CMAKE and it has recognized `openCV4.3`?
Is there a way to define which openCV version the VISP will build with?
Just to let you know, I need `openCV3.2` for some other packages, like `cv_bridge` which by default uses the `openCV3.2`

Call stack looks like this:
> libc.so.6!__GI_raise(int sig) (/build/glibc-OTsEL5/glibc-2.27/sysdeps/unix/sysv/linux/raise.c:51)
> libc.so.6!__GI_abort() (/build/glibc-OTsEL5/glibc-2.27/stdlib/abort.c:79)
> libstdc++.so.6![Unknown/Just-In-Time compiled code] (Unknown Source:0)
> libstdc++.so.6!__gxx_personality_v0 (Unknown Source:0)
> libgcc_s.so.1![Unknown/Just-In-Time compiled code] (Unknown Source:0)
> libgcc_s.so.1!_Unwind_Resume (Unknown Source:0)
> **libopencv_core.so.3.2!cv::TLSDataContainer::release() (Unknown Source:0)**
> libopencv_features2d.so.4.3!cv::TLSData<std::vector<cv::KeyPoint, std::allocator<cv::KeyPoint> > >::~TLSData(cv::TLSData<std::vector<cv::KeyPoint, std::allocator<cv::KeyPoint> > > * const this) (/home/ros-industrial/Desktop/openCV/opencv/modules/core/include/opencv2/core/utils/tls.hpp:65)
> libopencv_features2d.so.4.3!cv::TLSDataAccumulator<std::vector<cv::KeyPoint, std::allocator<cv::KeyPoint> > >::~TLSDataAccumulator(cv::TLSDataAccumulator<std::vector<cv::KeyPoint, std::allocator<cv::KeyPoint> > > * const this) (/home/ros-industrial/Desktop/openCV/opencv/modules/core/include/opencv2/core/utils/tls.hpp:97)
> libopencv_features2d.so.4.3!cv::SIFT_Impl::findScaleSpaceExtrema(const cv::SIFT_Impl * const this, const std::vector<cv::Mat, std::allocator<cv::Mat> > & gauss_pyr, const std::vector<cv::Mat, std::allocator<cv::Mat> > & dog_pyr, std::vector<cv::KeyPoint, std::allocator<cv::KeyPoint> > & keypoints) (/home/ros-industrial/Desktop/openCV/opencv/modules/features2d/src/sift.cpp:696)
> libopencv_features2d.so.4.3!cv::SIFT_Impl::detectAndCompute(cv::SIFT_Impl * const this, cv::InputArray _image, cv::InputArray _mask, std::vector<cv::KeyPoint, std::allocator<cv::KeyPoint> > & keypoints, cv::OutputArray _descriptors, bool useProvidedKeypoints) (/home/ros-industrial/Desktop/openCV/opencv/modules/features2d/src/sift.cpp:1150)
> libopencv_features2d.so.4.3!cv::Feature2D::detect(cv::Feature2D * const this, cv::InputArray image, std::vector<cv::KeyPoint, std::allocator<cv::KeyPoint> > & keypoints, cv::InputArray mask) (/home/ros-industrial/Desktop/openCV/opencv/modules/features2d/src/feature2d.cpp:70)
> libvisp_vision.so.3.3!vpKeyPoint::detect(vpKeyPoint * const this, const cv::Mat & matImg, std::vector<cv::KeyPoint, std::allocator<cv::KeyPoint> > & keyPoints, double & elapsedTime, const cv::Mat & mask) (/home/ros-industrial/Desktop/VISP_github/visp/modules/vision/src/key-point/vpKeyPoint.cpp:1377)
> libvisp_vision.so.3.3!vpKeyPoint::detect(vpKeyPoint * const this, const vpImage<unsigned char> & I, std::vector<cv::KeyPoint, std::allocator<cv::KeyPoint> > & keyPoints, double & elapsedTime, const vpRect & rectangle) (/home/ros-industrial/Desktop/VISP_github/visp/modules/vision/src/key-point/vpKeyPoint.cpp:1330)
> libvisp_vision.so.3.3!vpKeyPoint::matchPoint(vpKeyPoint * const this, const vpImage<unsigned char> & I, const vpCameraParameters & cam, vpHomogeneousMatrix & cMo, double & error, double & elapsedTime, bool (*)(const vpHomogeneousMatrix &) func, const vpRect & rectangle) (/home/ros-industrial/Desktop/VISP_github/visp/modules/vision/src/key-point/vpKeyPoint.cpp:3672)
> libpkg_libraries.so!VISP_Detection::detectObject(VISP_Detection * const this) (/home/ros-industrial/Desktop/ROS_VISP_ROBOTIQ_WS/src/libraries/pkg_libraries/src/VISP_Detection.cpp:405)
> libpkg_libraries.so!VISP_Detection::imageCb(VISP_Detection * const this, const sensor_msgs::ImageConstPtr & msg) (/home/ros-industrial/Desktop/ROS_VISP_ROBOTIQ_WS/src/libraries/pkg_libraries/src/VISP_Detection.cpp:200)
> libpkg_libraries.so!boost::_mfi::mf1<void, VISP_Detection, boost::shared_ptr<sensor_msgs::Image_<std::allocator<void> > const> const&>::operator()(const boost::_mfi::mf1<void, VISP_Detection, boost::shared_ptr<sensor_msgs::Image_<std::allocator<void> > const> const&> * const this, VISP_Detection * p, const boost::shared_ptr<sensor_msgs::Image_<std::allocator<void> > const> & a1) (/usr/include/boost/bind/mem_fn_template.hpp:165)
> libpkg_libraries.so!boost::_bi::list2<boost::_bi::value<VISP_Detection*>, boost::arg<1> >::operator()<boost::_mfi::mf1<void, VISP_Detection, boost::shared_ptr<sensor_msgs::Image_<std::allocator<void> > const> const&>, boost::_bi::rrlist1<boost::shared_ptr<sensor_msgs::Image_<std::allocator<void> > const> const&> >(boost::_bi::list2<boost::_bi::value<VISP_Detection*>, boost::arg<1> > * const this, boost::_mfi::mf1<void, VISP_Detection, boost::shared_ptr<sensor_msgs::Image_<std::allocator<void> > const> const&> & f, boost::_bi::rrlist1<boost::shared_ptr<sensor_msgs::Image_<std::allocator<void> > const> const&> & a) (/usr/include/boost/bind/bind.hpp:319)
> libpkg_libraries.so!boost::_bi::bind_t<void, boost::_mfi::mf1<void, VISP_Detection, boost::shared_ptr<sensor_msgs::Image_<std::allocator<void> > const> const&>, boost::_bi::list2<boost::_bi::value<VISP_Detection*>, boost::arg<1> > >::operator()<boost::shared_ptr<sensor_msgs::Image_<std::allocator<void> > const> const&>(boost::_bi::bind_t<void, boost::_mfi::mf1<void, VISP_Detection, boost::shared_ptr<sensor_msgs::Image_<std::allocator<void> > const> const&>, boost::_bi::list2<boost::_bi::value<VISP_Detection*>, boost::arg<1> > > * const this, const boost::shared_ptr<sensor_msgs::Image_<std::allocator<void> > const> & a1) (/usr/include/boost/bind/bind.hpp:1306)

Thanks

fspindle, 2020-05-18:
To select OpenCV version use something like:
```
$ cd visp-build
$ cmake ../visp -DOpenCV_DIR=/usr/local/share/OpenCV/ 
```
Check in `ViSP-third-party.txt` file that the right OpenCV version is used.
Once build, use `ldd <your-binary> | grep libopencv_core` to check that the right OpenCV version will be used during runtime

mbajlo, 2020-05-18:
Hello @fspindle ,

I have checked the `ViSP-third-party.txt` and I have the `Use OpenCV:                  yes (ver 4.3.0)` everywhere where the Use OpenCV is mentioned. After I have built and installed the `VISP`, I went to my ROS workspace and I made the clean and build again. I have checked the `ldd <your-binary> | grep libopencv_core` and it returned this:

> 	libopencv_core.so.3.2 => /usr/lib/x86_64-linux-gnu/libopencv_core.so.3.2 (0x00007fe9de973000)
> 	libopencv_core.so.4.3 => /usr/local/lib/libopencv_core.so.4.3 (0x00007fe9db2e2000)
> 

For some reason it uses the `libopencv_core.so.3.2` still, but I have no idea why. I have included just visp directories and libraries in my node `cmakelist.txt`

Do you know what could have caused that 2 versions of the libraries have been used?

mbajlo, 2020-05-19:
Hello @fspindle ,

I have been thinking and trying to figure out what could have caused this and what could be a possible workaround. I am kinda of stuck in the situation where I have a `image subscriber` which in it's callback is getting `cv_bridge::CvImagePtr`  from the `sensor_msgs::ImageConstPtr` which is basically the image from the subscribed image topic. The code looks like this:
```
_cv_ptr = cv_bridge::toCvCopy(msg, sensor_msgs::image_encodings::BGR8);
_vpImageCamera  = visp_bridge::toVispImage(*(_cv_ptr->toImageMsg()));
```
Do you think this could cause the issue? 
Because the `cv_bridge` is using the openCV 3.2(by default) and `VISP` is now installed with the new `openCV 4.3` and those are used in the same node? 
So even if I have setup the `openCV` path in the `CMakeList.txt` to look in the folder where the `openCV4.3` config file is (`OpenCVConfig.cmake`) the node will have a library conflict, since the `cv_bridge` has included the `openCV 3.2 libopencv_core.so.3.2` and the `VISP` is trying to use the `openCV4.3 libopencv_core.so.4.3`. 
If this is a problem, is there a way to solve it? how to explain the node that with VISP it has to use explicitely the openCV4.3 and the cv_bridge the openCV3.2 like it is doing already?

Thanks

fspindle, 2020-05-20:
A binary cannot be linked with 2 different versions of a same library.
You need to choose to use either OpenCV 3.2 or 4.3.
- If you choose 4.3, you need to uninstall Ubuntu packages that depend on OpenCV 3.2 and build these packages from source with OpenCV 4.3
- If you choose 3.2 that seems used by ROS packages, you need just to build ViSP with this version

mbajlo, 2020-05-20:
Thanks @fspindle ,
This is what I was afraid of, I thought there is some workaround since it the cv_bridge is hard coded with the version 3.2.

I will try to figure out, because I need opencv version greater than the 3.4.
Thanks a lot.

closed 2020-05-20 by mbajlo