lagadic/visp #762: Edge detector

fmaglia, 2020-05-18:
Hi,
I followed the tutorial Markerless generic model-based tracking using a color camera and I correctly implemented the tracker.
I prefer the edge one. What is the edge detector used in this case? Harris corners detector?
I tried to draw these features on the image with the tracker.getFeaturesForDisplay(), but I didn't find how to extract these features on the image. 
Do you know how can I extract (not draw and display) these features from the image?
Regards,
Federico

s-trinh, 2020-05-21:
Hi,

You should find more information about `vpMbGenericTracker::EDGE_TRACKER` and moving-edges tracking in the following papers:
- [A maximum-likelihood framework for determining moving edges in image sequences](https://hal.inria.fr/inria-00075857/document), Patrick Bouthemy
- [Real-time markerless tracking for augmented reality:the virtual visual servoing framework](https://hal.inria.fr/inria-00161250/document), Andrew Comport, Eric Marchand, Muriel Pressigout, François Chaumette, TVCG 2006

fmaglia, 2020-05-22:
Thanks.
I'm trying to automatically detect the 4 points to init the tracker.
Do you have any idea?
Is there something in VISP library?
Actually, I'm collecting the Shi-tomasi features of a bounding box selected by the user and then I remove useless combinations based on the geometry of the points.

s-trinh, 2020-05-22:
If your object is textured, you can use keypoints matching to compute 2D-3D correspondences and perform PnP pose estimation to set the initial pose for the MBT.
See the [following tutorial](https://visp-doc.inria.fr/doxygen/visp-daily/tutorial-detection-object.html).

Another solution is to use an AprilTag marker, see the [following tutorial](https://visp-doc.inria.fr/doxygen/visp-daily/tutorial-tracking-mb-generic-apriltag-live.html).

---

Here some tracking examples in commercial AR libraries, #670.
Looks like they use an user-assisted approach for initialisation. The user moves the camera to roughly align the projected CAD model with the real object, and the underlying methods register the CAD model contours with the current image to properly initialise the tracking.

If your object is textureless and you cannot use fiducial markers, for 6D object pose estimation:
- you can look at this challenge for a quick overview of the methods: [BOP: Benchmark for 6D Object Pose Estimation](https://bop.felk.cvut.cz/home/)
- if you are using a RGB-D camera, see methods that exploit pointcloud information
- or silouhette-based registration for pure 2D image

closed 2021-02-17 by fspindle