lagadic/visp #763: VISP does not support the latest PCL release 1.11.0

SergioRAgostinho, 2020-05-21:
Hey. I'm just opening the issue in case you intend to support the latest version of PCL. Currently 1.11.0 is failing in configuration. I bumped into the problem while trying to update PCL's formula in homebrew.

Platform: macOS Catalina 
CMake: 3.17.2

Dump:
```
CMake Error at modules/sensor/CMakeLists.txt:209 (get_target_property):
  INTERFACE_LIBRARY targets may only have whitelisted properties.  The
  property "IMPORTED_IMPLIB_NONE" is not allowed.

CMake Error at modules/sensor/CMakeLists.txt:211 (get_target_property):
  INTERFACE_LIBRARY targets may only have whitelisted properties.  The
  property "IMPORTED_LOCATION_NONE" is not allowed.

CMake Error at modules/sensor/CMakeLists.txt:209 (get_target_property):
  INTERFACE_LIBRARY targets may only have whitelisted properties.  The
  property "IMPORTED_IMPLIB_RELEASE" is not allowed.

CMake Error at modules/sensor/CMakeLists.txt:211 (get_target_property):
  INTERFACE_LIBRARY targets may only have whitelisted properties.  The
  property "IMPORTED_LOCATION_RELEASE" is not allowed.

CMake Error at modules/sensor/CMakeLists.txt:209 (get_target_property):
  INTERFACE_LIBRARY targets may only have whitelisted properties.  The
  property "IMPORTED_IMPLIB_DEBUG" is not allowed.

CMake Error at modules/sensor/CMakeLists.txt:211 (get_target_property):
  INTERFACE_LIBRARY targets may only have whitelisted properties.  The
  property "IMPORTED_LOCATION_DEBUG" is not allowed.

CMake Error at modules/sensor/CMakeLists.txt:209 (get_target_property):
  INTERFACE_LIBRARY targets may only have whitelisted properties.  The
  property "IMPORTED_IMPLIB_RELEASEWITHDEBINFO" is not allowed.

CMake Error at modules/sensor/CMakeLists.txt:211 (get_target_property):
  INTERFACE_LIBRARY targets may only have whitelisted properties.  The
  property "IMPORTED_LOCATION_RELEASEWITHDEBINFO" is not allowed.

CMake Error at modules/sensor/CMakeLists.txt:209 (get_target_property):
  INTERFACE_LIBRARY targets may only have whitelisted properties.  The
  property "IMPORTED_IMPLIB_RELWITHDEBINFO" is not allowed.

CMake Error at modules/sensor/CMakeLists.txt:211 (get_target_property):
  INTERFACE_LIBRARY targets may only have whitelisted properties.  The
  property "IMPORTED_LOCATION_RELWITHDEBINFO" is not allowed.

```

SergioRAgostinho, 2020-05-21:
The issue is being caused by pcl_2d and pcl_geometry, which currently are reported as interface modules (header only).

oKermorgant, 2020-05-25:
Hi, 
I have the same error on latest Ubuntu Focal (20.04).

The PR solves the issue that is probably coming from a CMake version.

PR #768 (2020-05-27): Fix compat with pcl 1.11.0 and ogre 1.9
This PR closes #763 and #767
commit 61c8beb8 (2020-05-27): Fix compat with pcl 1.11.0. Closes #763
Inspired from PR #765
closed 2020-05-27 by fspindle