lagadic/visp #766: Symbol Lookup Error: using visp generic-rgbd-realsense module

dvedova, 2020-05-25:
Hello,

I am attempting to build an run the executables for the tutorial on Markerless generic model-based tracking using a RGB-D camera [https://visp-doc.inria.fr/doxygen/visp-daily/tutorial-tracking-mb-generic-rgbd.html](url) 

When I follow the tutorial to install visp, and build everything, I get the following error:

`(base) dvedova@danny-ThinkPad-P52s:~/visp-tracking/generic-rgbd/build$ ./tutorial-mb-generic-tracker-rgbd-realsense 
./tutorial-mb-generic-tracker-rgbd-realsense: symbol lookup error: ./tutorial-mb-generic-tracker-rgbd-realsense: undefined symbol: _ZN12vpRealSense24openERKN3rs26configE
`

I am running ubuntu 18.04, and I installed visp from source. I have tried going through the tutorials several times, and I can get other modules to work, but for some reason, I cannot get the tutorial-mb-generic-tracker-rgbd or tutorial-mb-generic-tracker-rgbd-realsense executables to run without sending the symbol lookup error. 

I would really appreciate any assistance with this problem. Thank you!

s-trinh, 2020-05-25:
Hi,

Can you copy/paste the content of the `ViSP-third-party.txt` file located in the ViSP build directory?
Also, which version of `librealsense` are you using? How did you install the `librealsense` library?

dvedova, 2020-05-25:
Thanks for the reply!

I am using librealsesne 2.0, which I cloned from github this morning and installed by following the directions at [https://visp-doc.inria.fr/doxygen/visp-daily/tutorial-install-ubuntu.html](url) under the instructions for advanced visp installation. 

Here is the text from the `ViSP-third-party.txt` file:

`
==========================================================
General configuration information for ViSP 3.3.1

  Version control:               3.2.0-1046-g69f2a00f5

  Platform:
    Timestamp:                   2020-05-25T01:18:03Z
    Host:                        Linux 5.3.0-51-generic x86_64
    CMake:                       3.10.2
    CMake generator:             Unix Makefiles
    CMake build tool:            /usr/bin/make
    Configuration:               Release

  C/C++:
    Built as dynamic libs?:      yes
    C++ Compiler:                /usr/bin/c++  (ver 5.5.0)
    C++ flags (Release):         -Wall -Wextra -fopenmp -std=c++11 -fvisibility=hidden -msse2 -msse3 -mssse3 -fPIC -O3 -DNDEBUG
    C++ flags (Debug):           -Wall -Wextra -fopenmp -std=c++11 -fvisibility=hidden -msse2 -msse3 -mssse3 -fPIC -g
    C Compiler:                  /usr/bin/cc
    C flags (Release):           -Wall -Wextra -fopenmp -std=c++11 -fvisibility=hidden -msse2 -msse3 -mssse3 -fPIC -O3 -DNDEBUG
    C flags (Debug):             -Wall -Wextra -fopenmp -std=c++11 -fvisibility=hidden -msse2 -msse3 -mssse3 -fPIC -g
    Linker flags (Release):      
    Linker flags (Debug):        

  ViSP modules:
    To be built:                 core gui imgproc io java_bindings_generator klt me sensor ar blob robot visual_features vs vision detection mbt tt tt_mi
    Disabled:                    -
    Disabled by dependency:      -
    Unavailable:                 java

  Python (for build):            /home/dvedova/anaconda2/bin/python2.7

  Java:                          
    ant:                         NO
    JNI:                         /usr/lib/jvm/java-8-openjdk-amd64/include /usr/lib/jvm/java-8-openjdk-amd64/include/linux /usr/lib/jvm/java-8-openjdk-amd64/include

  Build options: 
    Build deprecated:            yes
    Build with moment combine:   no

  Mathematics: 
    Blas/Lapack:                 yes
    \- Use MKL:                  no
    \- Use OpenBLAS:             no
    \- Use Atlas:                no
    \- Use Netlib:               yes (ver 3.7.1)
    \- Use GSL:                  no
    \- Use Lapack (built-in):    no
    Use Eigen3:                  yes (ver 3.3.4)
    Use OpenCV:                  yes (ver 3.2.0)

  Simulator: 
    Ogre simulator: 
    \- Use Ogre3D:               yes (ver 1.9.0)
    \- Use OIS:                  yes (ver 1.3.0)
    Coin simulator: 
    \- Use Coin3D:               yes (ver 4.0.0)
    \- Use SoWin:                no
    \- Use SoXt:                 no
    \- Use SoQt:                 no
    \- Use Qt5:                  no
    \- Use Qt4:                  no
    \- Use Qt3:                  no

  Media I/O: 
    Use JPEG:                    yes (ver 80)
    Use PNG:                     yes (ver 1.6.34)
    \- Use ZLIB:                 yes (ver 1.2.11)
    Use OpenCV:                  yes (ver 3.2.0)
    Use stb_image (built-in):    no

  Real robots: 
    Use Afma4:                   no
    Use Afma6:                   no
    Use Franka:                  no
    Use Viper650:                no
    Use Viper850:                no
    Use aria (Pioneer):          no
    Use PTU46:                   no
    Use Biclops PTU:             no
    Use Flir PTU SDK:            no
    Use Parrot ARSDK:            no
    \-Use ffmpeg:                no
    Use Virtuose:                no
    Use qbdevice (built-in):     yes (ver 2.6.0)
    Use takktile2 (built-in):    yes (ver 1.0.0)

  GUI: 
    Use X11:                     yes
    Use GTK:                     no
    Use OpenCV:                  yes (ver 3.2.0)
    Use GDI:                     no
    Use Direct3D:                no

  Cameras: 
    Use DC1394-2.x:              yes (ver 2.2.5)
    Use CMU 1394:                no
    Use V4L2:                    yes (ver 1.14.2)
    Use directshow:              no
    Use OpenCV:                  yes (ver 3.2.0)
    Use Flycapture:              no
    Use Pylon:                   no

  RGB-D sensors: 
    Use Realsense:               no
    Use Realsense2:              yes (ver 2.33.1)
    Use Kinect:                  no
    \- Use libfreenect:          no
    \- Use libusb-1:             yes (ver 1.0.21)
    \- Use pthread:              yes
    Use PCL:                     yes (ver 1.8.1)
    \- Use VTK:                  yes (ver 6.3.0)

  F/T sensors: 
    Use atidaq (built-in):       no
    Use comedi:                  no
    Use IIT SDK:                 no

  Detection: 
    Use zbar:                    yes (ver 0.10)
    Use dmtx:                    yes (ver 0.7.4)
    Use AprilTag (built-in):     yes (ver 3.1.1)
    \- Use AprilTag big family:  no

  Misc: 
    Use Clipper (built-in):      yes (ver 6.4.2)
    Use pugixml (built-in):      yes (ver 1.9.0)
    Use libxml2:                 yes (ver 2.9.4)

  Optimization: 
    Use OpenMP:                  yes
    Use pthread:                 yes
    Use pthread (built-in):      no
    Use cxx standard:            11

  Documentation: 
    Use doxygen:                 no

  Tests and samples:
    Use catch2 (built-in):       yes (ver 2.9.2)
    Tests:                       yes
    Demos:                       yes
    Examples:                    yes
    Tutorials:                   yes

  Install path:                  /usr/local

==========================================================`

fspindle, 2020-05-26:
I suspect that you have multiple visp versions installed, and at least one that is not linked against `librealsense`. This visp version was maybe installed either from `libvisp-dev` package or from `ros-<distro>-visp` package.

Check running:
```
$ ldd ~/visp-tracking/generic-rgbd/build/tutorial-mb-generic-tracker-rgbd-realsense | grep libvisp_core
$ sudo updatedb
$ locate libvisp_core.so
```

and reply with the results of these commands

dvedova, 2020-05-26:
I think you are right that there are multiple visp versions installed (several of the versions seem to be in the trash; this is because I tried deleting my visp-ws directory and retrying the installation instructions, but I think when I deleted the visp-ws directory I did not actually uninstall visp). 

Here are the results of running the aforementioned commands: 
`(base) dvedova@danny-ThinkPad-P52s:~/mbt_tracking_ws$ ldd ~/visp-tracking/generic-rgbd/build/tutorial-mb-generic-tracker-rgbd-realsense | grep libvisp_core
	libvisp_core.so.3.3 => /opt/ros/melodic/lib/libvisp_core.so.3.3 (0x00007f69f4ea4000)`

`(base) dvedova@danny-ThinkPad-P52s:~/mbt_tracking_ws$ sudo updatedb`

`(base) dvedova@danny-ThinkPad-P52s:~/mbt_tracking_ws$ locate libvisp_core.so
/home/dvedova/.local/share/Trash/files/visp-ws/visp-build/lib/libvisp_core.so
/home/dvedova/.local/share/Trash/files/visp-ws/visp-build/lib/libvisp_core.so.3.3
/home/dvedova/.local/share/Trash/files/visp-ws/visp-build/lib/libvisp_core.so.3.3.1
/home/dvedova/.local/share/Trash/files/visp-ws2/visp-build/lib/libvisp_core.so
/home/dvedova/.local/share/Trash/files/visp-ws2/visp-build/lib/libvisp_core.so.3.3
/home/dvedova/.local/share/Trash/files/visp-ws2/visp-build/lib/libvisp_core.so.3.3.1
/home/dvedova/catkin_ws/src/visp-release/lib/libvisp_core.so
/home/dvedova/catkin_ws/src/visp-release/lib/libvisp_core.so.3.3
/home/dvedova/catkin_ws/src/visp-release/lib/libvisp_core.so.3.3.0
/home/dvedova/visp-ws/visp-build/lib/libvisp_core.so
/home/dvedova/visp-ws/visp-build/lib/libvisp_core.so.3.3
/home/dvedova/visp-ws/visp-build/lib/libvisp_core.so.3.3.1
/opt/ros/melodic/lib/libvisp_core.so
/opt/ros/melodic/lib/libvisp_core.so.3.3
/opt/ros/melodic/lib/libvisp_core.so.3.3.0
(base) dvedova@danny-ThinkPad-P52s:~/mbt_tracking_ws$ `

dvedova, 2020-06-03:
What would be the best way of solving the issue? Should I delete all of the versions I have and reinstall one more time?

s-trinh, 2020-06-03:
```
ldd ~/visp-tracking/generic-rgbd/build/tutorial-mb-generic-tracker-rgbd-realsense | grep libvisp_core libvisp_core.so.3.3 => /opt/ros/melodic/lib/libvisp_core.so.3.3 (0x00007f69f4ea4000)
```

What I would do:
- uninstall ViSP installed by ROS, but before confirming
- note the ROS packages that will be uninstalled because they depend on ROS ViSP
- uninstall ROS ViSP
- clone from source the ROS packages that have been uninstalled
- build them using your version of ViSP built from source

cheacheavpor, 2020-11-09:
> What would be the best way of solving the issue? Should I delete all of the versions I have and reinstall one more time
How did you finally solve your problem? reinstall it? I will appreciate it if you could share me some experience. I met the absolutely same issue as yours.

fspindle, 2020-11-09:
1. Identify all the ViSP versions you have installed. There should be one in `/opt/ros/distro`and maybe an other one build from source in `$HOME/visp-ws/visp-build` and if you did make install the corresponding installation in `/usr/local`.

    ```
    $ sudo updatedb
    $ locate libvisp_core.so
    ```

2. Identify all the ROS packages that depend on ROS ViSP installed in `/opt/ros/distro`. Use `sudo synaptic`and search for "visp" keyword. Here you could find the `libvisp-dev` package and also ROS packages `ros-distro-visp`, `ros-distro-vision-visp`...

3. Remove all installed ubuntu packages that mention `visp`: `libvisp-dev`, `ros-distro-visp`, `ros-distro-vision-visp`...

4. Do again step 1. Here you should find only one installation, the one build from source in `$HOME/visp-ws/visp-build` and if you did make install the corresponding installation in `/usr/local`.

5. Update ViSP from source to the last github version
    ```
    $ cd $HOME/visp-ws/visp
    $ git pull
    $ cd ../visp-build
    $ make -j4
    $ sudo make install
    ```

6. Install from source all ROS packages that you are using

    ```
    $ cd $HOME/catkin_ws/src
    $ git clone https://github.com/lagadic/vision_visp
    $ cd ..
    $ catkin_make -j4
    ```

Keep in mind that the objective of all these steps is:
1. to have a single up to date version of visp installed in `/usr/local`
2. build all the packages that are using ViSP from source and link them to the version installed in `/usr/local`

cheacheavpor, 2020-11-10:
> 1. Identify all the ViSP versions you have installed. There should be one in `/opt/ros/distro`and maybe an other one build from source in `$HOME/visp-ws/visp-build` and if you did make install the [...]

Hi Fabien:
Thanks for replying. Your instruction was clear and detailed.
Many thanks,

closed 2021-02-17 by fspindle