lagadic/visp #772: Using the vpDisplay::Init() function causes a std::bad_alloc error.

binit400, 2020-06-04:
Hello. 

I am attempting to use ViSP for a project in Unity so I can do model tracking with a depth sensor. I followed the tutorials for setting up ViSP and creating a plugin for Unity without a problem. I am now trying to create a plugin using the RGB-D tracking example. I am splitting the main function up into an initialize function and the tracking loop, so that way I can feed it my camera data from Unity.

I've run into an issue where the Initialize function was crashing Unity. The line that was causing to crash was the vpDisplay::Init() function. At first I thought the data I was passing it wasn't converting correctly, but I tried loading a test image in its place and it still throws the error. Visual Studio is saying that it is a std::bad_alloc error. I looked at a couple different tutorials, and it doesn't look like there is any step I am missing before I try to call the Init function, so I am not sure why it would cause a crash. 

I have created a separate Unity project where C# simply calls the plugin without sending it camera data, and the plugin (still) loads the test image to use as data. Strangely, this project fails on the same line of code, but unlike the main project, It was actually throwing a vpException rather than just crashing. The exception says "Image not Initialized". I've tried to look through the documentation again to see what I was missing, but from my understanding there doesn't seem to be a way to "initialize" a vpImage before using it in vpDisplay::Init(), which leaves me in the same place as having no error message.

Below is the code in my C# script that calls the C++ plugin from Unity, and then the full code for the Plugin. I have not finished reworking the code, so there may be some errors towards the bottom of my functions that I have hardly looked at, but right now I am just trying to solve the issue with vpDisplay. If anybody can point out a step I am missing or give me some insight on this problem, it would be much appreciated.

C# code
```
using System;
using System.Collections;
using System.Collections.Generic;
using System.IO;
using System.Runtime.InteropServices;
using UnityEngine;

public class ExecutePlugin : MonoBehaviour
{
    [DllImport("ViSPUnity", CallingConvention = CallingConvention.StdCall, EntryPoint = "InitializeTrackRGBD")]
    public static extern void InitializeTrackRGBD(Byte[] colorMat, UInt16[] depthBitmap);
    [DllImport("ViSPUnity", CallingConvention = CallingConvention.StdCall, EntryPoint = "TrackRGBD")]
    public static extern void TrackRGBD(Byte[] colorMat, UInt16[] depthBitmap, UInt16[] pointCloud);

    public void OnPress_ExecutePlugin()
    {
        byte[] bArray = new byte[1];
//For now, I am reading depth data that I wrote to a file and passing that to the plugin.
        byte[] bytesFromFile = File.ReadAllBytes("DepthData.data");

        UInt16[] uInt16Array = new UInt16[bytesFromFile.Length / 2];

        int j = -2;
        for (int i = 0; i < uInt16Array.Length; i++)
        {
            uInt16Array[i] = BitConverter.ToUInt16(bytesFromFile, j += 2);
        }

        InitializeTrackRGBD(bArray, uInt16Array);
    }
}
```

C++ code
```
#include "ViSPUnity.h"

#include <visp3/gui/vpDisplayGDI.h>
#include <visp3/gui/vpDisplayOpenCV.h>
#include <visp3/gui/vpDisplayX.h>
#include <iostream>
#include <visp3/core/vpDisplay.h>
#include <visp3/core/vpIoTools.h>
#include <visp3/io/vpImageIo.h>
#include <visp3/core/vpXmlParserCamera.h>
#include <visp3/gui/vpDisplayGDI.h>
#include <visp3/gui/vpDisplayX.h>
#include <visp3/mbt/vpMbGenericTracker.h>
#if defined (VISP_HAVE_PCL)
#include <pcl/common/common.h>
#include <pcl/io/pcd_io.h>
#include <pcl/visualization/cloud_viewer.h>
#endif

extern "C" {

#if defined (VISP_HAVE_PCL)
	struct rs_intrinsics {
		float ppx;
		float ppy;
		float fx;
		float fy;
		float coeffs[5];
	};
	std::string input_directory = "Assets/data"; // location of the data (images, depth_map, point_cloud)
	std::string config_color = "Assets/cube/cube.xml", config_depth = "Assets/cube/cube_depth.xml";
	std::string model_color = "Assets/cube/cube.cao", model_depth = "Assets/cube/cube.cao";
	std::string init_file = "Assets/cube/cube.init";
	int frame_cpt = 0;
	bool disable_depth = false;
	unsigned int _posx = 100, _posy = 50, _posdx = 10;
	vpCameraParameters cam_color, cam_depth;
	vpImage<vpRGBa> I_color;
	vpImage<unsigned char> I_gray, I_depth;
	vpImage<uint16_t> I_depth_raw;
	vpMbGenericTracker tracker;
	std::map<std::string, const vpImage<unsigned char> *> mapOfImages;
	vpHomogeneousMatrix depth_M_color;
	std::vector<double> times_vec;

	PLUGINEX(void) InitializeTrackRGBD(uint8_t *colorByteArray, uint16_t *depthBitmap) {
		ofstream stream("ErrorLog.txt");
		try {
			//cv::Size sz(1920, 1080);
			//cv::Mat colorMat = cv::Mat(sz, CV_8UC3, colorByteArray);
			cv::Mat colorMat = cv::imread("Assets/TestImage.jpg");
			vpImageConvert::convert(colorMat, I_color);
			I_depth_raw.resize(1024, 1024);
			I_depth_raw.bitmap = depthBitmap;
			vpImageConvert::convert(I_color, I_gray);
			vpImageConvert::createDepthHistogram(I_depth_raw, I_depth);

			vpDisplayOpenCV d1, d2;
			d1.init(I_gray, _posx, _posy, "Color stream");
			d2.init(I_depth, _posx + I_gray.getWidth() + _posdx, _posy, "Depth stream");
			vpDisplay::display(I_gray);
			vpDisplay::display(I_depth);
			vpDisplay::flush(I_gray);
			vpDisplay::flush(I_depth);
			std::vector<int> trackerTypes;
			trackerTypes.push_back(vpMbGenericTracker::EDGE_TRACKER);
			trackerTypes.push_back(vpMbGenericTracker::KLT_TRACKER);
			trackerTypes.push_back(vpMbGenericTracker::DEPTH_DENSE_TRACKER);
			tracker = vpMbGenericTracker(trackerTypes);
			tracker.loadConfigFile(config_color, config_depth);
			tracker.loadModel(model_color, model_depth);
			tracker.getCameraParameters(cam_color, cam_depth);

			std::cout << "Camera parameters for color camera (from XML file): " << cam_color << std::endl;
			std::cout << "Camera parameters for depth camera (from XML file): " << cam_depth << std::endl;
			tracker.setDisplayFeatures(true);
			{
				std::ifstream file((std::string(input_directory + "/depth_M_color.txt")).c_str());
				depth_M_color.load(file);
				file.close();
			}
			std::map<std::string, vpHomogeneousMatrix> mapOfCameraTransformations;
			mapOfCameraTransformations["Camera1"] = vpHomogeneousMatrix();
			mapOfCameraTransformations["Camera2"] = depth_M_color;
			tracker.setCameraTransformationMatrix(mapOfCameraTransformations);
			std::cout << "depth_M_color: \n" << depth_M_color << std::endl;
			mapOfImages["Camera1"] = &I_gray;
			mapOfImages["Camera2"] = &I_depth;
			std::map<std::string, std::string> mapOfInitFiles;
			mapOfInitFiles["Camera1"] = init_file;
			tracker.initClick(mapOfImages, mapOfInitFiles, true);
			mapOfImages.clear();

			std::cout << "\nProcessing time, Mean: " << vpMath::getMean(times_vec) << " ms ; Median: " << vpMath::getMedian(times_vec)
				<< " ; Std: " << vpMath::getStdev(times_vec) << " ms" << std::endl;
		}
		catch (const vpException &e) {
			stream << "Catch exception: " << e.getMessage() << std::endl;
		}
		stream.close();
	}
```

fspindle, 2020-06-05:
The line that is causing the issue is the following:
```
I_depth_raw.bitmap = depthBitmap;
```
you should rather do a memcpy to update the bitmap like
```
memcpy(I_depth_raw.bitmap, depthBitmap, I_depth_raw.getWidth()*I_depth_raw.getHeight()*sizeof(uint16_t));
```

There is a more elegant way avoing data copy by modifying the construction of `I_depth_raw` in order that the bitmap is pointing to `depthBitmap`:
```
#include <visp3/gui/vpDisplayOpenCV.h>
#include <visp3/core/vpIoTools.h>

int main()
{
  uint16_t *depthBitmap;
  unsigned int depth_width = 1024, depth_height = 1024;
  {
    // Data updated by the sensor
    depthBitmap = new uint16_t [depth_width * depth_height];
    for (unsigned int i = 0; i < depth_width * depth_height; i++) {
      depthBitmap[i] = 255;
    }
  }

  vpImage<vpRGBa> I_color;
  vpImage<unsigned char> I_gray, I_depth;
  vpImage<uint16_t> I_depth_raw(depthBitmap, 1024, 1024, false);
  unsigned int _posx = 100, _posy = 50, _posdx = 10;
  cv::Mat colorMat = cv::imread("Assets/TestImage.jpg");
  vpImageConvert::convert(colorMat, I_color);
  //I_depth_raw.resize(1024, 1024);
  //I_depth_raw.bitmap = depthBitmap;
  vpImageConvert::convert(I_color, I_gray);
  vpImageConvert::createDepthHistogram(I_depth_raw, I_depth);

  vpDisplayOpenCV d1, d2;
  d1.init(I_gray, _posx, _posy, "Color stream");
  d2.init(I_depth, _posx + I_gray.getWidth() + _posdx, _posy, "Depth stream");
  vpDisplay::display(I_gray);
  vpDisplay::display(I_depth);
  vpDisplay::flush(I_gray);
  vpDisplay::flush(I_depth);
  vpDisplay::getClick(I_gray);
  vpDisplay::getClick(I_depth);

  delete [] depthBitmap;
  std::cout << "The end" << std::endl;
}
```

binit400, 2020-06-05:
Thank you for responding to my issue so quickly. I tried using your method for copying the depth data, but it still crashes in the same way. I am fairly certain that the problem has to do with converting and displaying the RGB data. I tried commenting out all the code that has to do with the depth data, and the program still crashes in the same way. It seems to be crashing once it hits this line.
```
d1.init(I_gray, _posx, _posy, "Color stream");
```
I can't hit breakpoints in the plugin when I attach VS to Unity; so the way I determined the line it crashed at was by writing to a text file in between every line of code. I run the program with VS attached, and it takes me to throw_bad_alloc.cpp and displays a text box that says that my code caused a std::bad_alloc error. I am sorry if my methods are messy. I am used to working with C# and this was the best I could do to narrow down the problem.

In Unity, I loaded my color data to a texture immediately before calling the plugin to make sure the data I am passing is correct. It looks like I am converting my RGB data the same way some of the tracking examples were, so I am not sure what I need to change.

fspindle, 2020-06-06:
2 ideas to investigate:

1. It may be possible that the image size from texture is not updated when calling `d1.init()`. A work arround is to introduce a "warmup" as done [here](https://github.com/lagadic/visp_unity/blob/master/unityProject/Assets/Scripts/Script_ar.cs#L183).

2. To debug write a simple as possible test in C++ without Unity to see if you are able to reproduce the issue.