lagadic/visp #782: Issue with loading files when using ViSP in a native plugin for Unity

binit400, 2020-06-25:
Hello again,

I am using ViSP in a plugin that I am trying to use in Unity. I followed the tutorial on how to create a plugin for Unity, and now I am trying to use the code in the "model based tracking using an rgd-d camera" example in my plugin so I can perform the tracking in  a Unity project.  

I have run into an issue with loading files in the vpMbGenericTracker functions. First I ran into an issue when I tried to set the config settings using the loadConfigFile() function. When I tried this function just using the cube.xml and cube_depth.xml files provided in the example, it would give me a vpException that says "Cannot open XML file "(garbage data)", where the (garbage data) was a short string of random characters (example: "àÊ_"). Since the file is formatted correctly, I am not sure why I would be getting this error, nor why it would give us garbage data rather than the name of the file. We have tried putting the files in a different folder to see if it was a permissions error with Unity, but it would still give us the same error. To work around this, I used the individual functions on the vpMbGenericTracker to set the config settings. We then ran into the same error when trying to use the loadModel() function. It would put the same garbage data in the error message. I tried this both with the provided cube.cao file, and with a .wrl file I made of an arduino. Since we got the same type of error in two places where we were trying to load a file, we assumed that the problem was either caused by Unity interacting with ViSP, or a bug in the code. I tried to create a c++ console app to interact with the plugin in Unity's place in order to narrow down the problem. Unfortunately, I had some issues setting up the references, and my manager wants me to move on from this issue and work on another solution. I can still answer questions in regards to this problem.

fspindle, 2020-07-01:
Did you try to run a ViSP example without Unity, like the one in `tutorial/tracking/model-based/generic/tutorial-mb-generic-tracker.cpp`?

binit400, 2020-07-01:
I cannot get cmake to build the tutorial. It gives me the error: 

> CMake Error at CMakeLists.txt:5 (find_package):
>   By not providing "FindVISP.cmake" in CMAKE_MODULE_PATH this project has
>   asked CMake to find a package configuration file provided by "VISP", but
>   CMake did not find one.
> 
>   Could not find a package configuration file provided by "VISP" with any of
>   the following names:
> 
>     VISPConfig.cmake
>     visp-config.cmake
> 
>   Add the installation prefix of "VISP" to CMAKE_PREFIX_PATH or set
>   "VISP_DIR" to a directory containing one of the above files.  If "VISP"
>   provides a separate development package or SDK, be sure it has been
>   installed.

I have the "VISP_DIR" environment variable set to the folder "C:\visp-ws\visp-build-vc15\install", which contains VISPConfig.cmake file, so I am not sure what the issue is.