lagadic/visp #783: Version 3.3.0 fails to build with pcl-1.11.0: error: no member named 'isFinite' in namespace 'pcl'

yurivict, 2020-06-26:
```
/wrkdirs/usr/ports/misc/visp/work/visp-3.3.0/modules/tracker/mbt/src/depth/vpMbtFaceDepthDense.cpp:216:53: error: no member named 'isFinite' in namespace 'pcl'
        if (vpMeTracker::inMask(mask, i, j) && pcl::isFinite((*point_cloud)(j, i)) && (*point_cloud)(j, i).z > 0) {
                                               ~~~~~^
1 error generated.
```

FreeBSD 12.1
clang-9

fspindle, 2020-07-02:
@yurivict Thank's for your feedback. I'm able to reproduce the error on FreeBSD 12.1 with PCL 1.11.0 and ViSP 3.3.0.

This error doesn't occur on FreeBSD 12.1 with PCL 1.11.0.99 installed from source (current github master version) or 1.11.0 release build from source and ViSP from master branch.

I will investigate to see what is wrong in VISP 3.3.0 and what was changed in ViSP since 3.3.0 that fixes the build.

fspindle, 2020-07-03:
This issue was fixed in 61c8beb8442f9

If you can add a patch in `modules/tracker/mbt/src/depth/vpMbtFaceDepthDense.cpp` that adds the following lines it should do the trick:

```
--- /Users/fspindle/soft/ViSP/visp-3.3.0/modules/tracker/mbt/src/depth/vpMbtFaceDepthDense.cpp  2020-02-14 14:41:31.000000000 +0100
+++ vpMbtFaceDepthDense.cpp     2020-06-12 08:29:33.000000000 +0200
@@ -36,6 +36,10 @@
 #include <visp3/core/vpCPUFeatures.h>
 #include <visp3/mbt/vpMbtFaceDepthDense.h>
 
+#ifdef VISP_HAVE_PCL
+#include <pcl/common/point_tests.h>
+#endif
+
 #if defined __SSE2__ || defined _M_X64 || (defined _M_IX86_FP && _M_IX86_FP >= 2)
 #include <emmintrin.h>
 #define VISP_HAVE_SSE2 1
```

yurivict, 2020-08-05:
This works, thanks!

closed 2020-08-05 by yurivict