lagadic/visp #790: Build with GCC aarch64

ArMourMag, 2020-07-16:
Hello,

I'm an error when I built VISP with gcc for aarch64 (aarch64-linux-gnu-g++  (ver 7.5.0)).

```
make --jobserver-fds=7,8 -j
in directory /home/esrocos_workspace/external/visp_3_3_0/build_aarch64
[  0%] Built target developer_scripts
[  0%] Built target man
[  0%] Built target visp_reflex_takktile2
[  1%] Building CXX object 3rdparty/simdlib/CMakeFiles/visp_simdlib.dir/Simd/SimdBaseCustomFunctions.cpp.o
[  1%] Building CXX object 3rdparty/simdlib/CMakeFiles/visp_simdlib.dir/Simd/SimdLib.cpp.o
[  1%] Building CXX object 3rdparty/simdlib/CMakeFiles/visp_simdlib.dir/Simd/SimdBaseBgrToRgba.cpp.o
[  1%] Building CXX object 3rdparty/simdlib/CMakeFiles/visp_simdlib.dir/Simd/SimdBaseBgraToGray.cpp.o
[  1%] Building CXX object 3rdparty/simdlib/CMakeFiles/visp_simdlib.dir/Simd/SimdBaseBgraToBgr.cpp.o
[  1%] Building CXX object 3rdparty/simdlib/CMakeFiles/visp_simdlib.dir/Simd/SimdBaseCopy.cpp.o
aarch64-linux-gnu-g++: error: unrecognized command line option ‘-mfpu=neon’
3rdparty/simdlib/CMakeFiles/visp_simdlib.dir/build.make:62: recipe for target '3rdparty/simdlib/CMakeFiles/visp_simdlib.dir/Simd/SimdLib.cpp.o' failed
make[2]: *** [3rdparty/simdlib/CMakeFiles/visp_simdlib.dir/Simd/SimdLib.cpp.o] Error 1
make[2]: *** Waiting for unfinished jobs....
[ 27%] Built target visp_lapack
CMakeFiles/Makefile2:349: recipe for target '3rdparty/simdlib/CMakeFiles/visp_simdlib.dir/all' failed
make[1]: *** [3rdparty/simdlib/CMakeFiles/visp_simdlib.dir/all] Error 2
Makefile:129: recipe for target 'all' failed
make: *** [all] Error 2
Exit: pid 118710 exit 2
```

Thanks.

fspindle, 2020-07-16:
@ArMourMag Thanks for this feedback. In order to reproduce this issue, could you precise what kind of device and OS you are using ?

A work around to fix this build issue is to modify the `3rdparty/simdlib/CMakeLists.txt` file [line 160](https://github.com/lagadic/visp/blob/master/3rdparty/simdlib/CMakeLists.txt#L160) with 
```
set(CXX_NEON_FLAG "")
```

ArMourMag, 2020-07-16:
The issue is fixed. In my cmake toolchain file I forgot to change CMAKE_SYSTEM_PROCESSOR to aarch64. I thought that the compiler name was enough.

Thanks.

closed 2020-07-16 by ArMourMag