lagadic/visp #797: 4 points problem fitting (Markerless generic model-based tracking)

andreabonacini, 2020-07-30:
Hi, 

I'm trying to use the Markerless generic model-based tracking tool for the edge detection of some objects. The aim is to reach the detection using 4 points defined on the same face but sometimes it does not work. 

Below i'm showing you an example with a teabox.

**_Points numeration:_**
[image]

**_Error:_**
[image]

Do you know this kind of problem? If yes, why is this happening and is there some way to solve it? Thanks in advance.

Regards,
Andrea

s-trinh, 2020-07-30:
Hi,

This is due to the planar pose estimation ambiguity:

[image]

Small uncertainties in the 2D corner locations and the pose can be ambiguous in certain viewpoint.

To solve the issue, try to use a 4th point not on the same plane. Or add a 5th point not on the same plane.

andreabonacini, 2020-07-31:
Thanks for your reply.

I have 2 more questions:
1) Is there any method to implement the object fitting without using the whole tracking method?
2) Is there any possibility to evaluate more 4 points combinations fastly instead of repeating many times the initFromPoints() + the tracking function? Because we have a list of 4 points combinations to evaluate in order to fastly get the best fitting/pose of the object (that corresponds to the best 4 points combination)

Regards,
Andrea

s-trinh, 2020-08-02:
Can you describe a bit more what you want to achieve? For instance, do you have a fixed list of 3D coordinates and unknown correspondind 2D coordinates?

---

> Is there any method to implement the object fitting without using the whole tracking method?

I don't think so, but if you can provide more information about your use case.

> Is there any possibility to evaluate more 4 points combinations fastly instead of repeating many times the initFromPoints() + the tracking function? Because we have a list of 4 points combinations [...]

`initFromPoints()` is used to compute the initial pose to initialise the tracker. If you have another method that provides this info, you can use [`initFromPose()`](https://visp-doc.inria.fr/doxygen/visp-daily/classvpMbGenericTracker.html#a6205351b5de665aab9aa9b655404db20) to initialise the tracker.

Getting this initial pose is tricky. `initFromPoints()` is an easy way to let the user click on specific points to perform the pose computation and initialise the tracker. [Keypoints](https://visp-doc.inria.fr/doxygen/visp-daily/tutorial-detection-object.html) can be a solution. For state of the art methods, see [6th International Workshop on Recovering 6D Object Pose](http://cmp.felk.cvut.cz/sixd/workshop_2020/).

If you have a method that provides the 2d coordinates, you can use [`vpPose`](https://visp-doc.inria.fr/doxygen/visp-daily/classvpPose.html) to compute the pose from 2D/3D information. Since you have mentioned "a list of 4 points combinations to evaluate", maybe what you are doing is similar to [`findMatch()`](https://visp-doc.inria.fr/doxygen/visp-daily/classvpPose.html#adc0154fe830ba9a6566f9217621e6ef3)?

andreabonacini, 2020-08-04:
Thanks for the tips.

I think that initFromPose() is not what we are looking for, since the pose matrix is the real aim. 

Our goal is to compute the model fitting of an object starting from four 3D points (on the same plane) in the object frame and the four corresponding 2D points on the image plane (provided by our vision algorithm) in order to get the pose in the camera frame.
 In this way, it seems to me that findMatch() could be enough, isn't it?

s-trinh, 2020-08-04:
> In this way, it seems to me that findMatch() could be enough, isn't it?

Yes, I think so.

You will need to tweak the RANSAC parameters like the maximum number of iterations, reprojection error threshold, etc. to have good results.

andreabonacini, 2020-08-26:
Thanks again. 

Also, about the vector of 2D points: are they the pixel points on the image plane or are they expressed in meters?
I'm asking you this because i don't get where the function findMatch() exploits the intrinsic matrix of the camera.

I imagine that it need that matrix in order to get the correct cMo.

fspindle, 2020-08-26:
The vector of 2D points `p2D` used in [findMatch()](file:///Users/fspindle/soft/ViSP/ViSP-dev/test-pr/visp-fspindle-build/doc/html/classvpPose.html#adc0154fe830ba9a6566f9217621e6ef3) needs `x,y` 2D values in meter.

The conversion from pixel to meter has to be done prior calling this function using something similar to:
```
for (unsigned int i = 0; ...) {
    double x = 0, y = 0;
    // u, v are the coordinates in pixels (u horizontal, v vertical coordinates)
    vpPixelMeterConversion::convertPoint(cam, u, v, x, y);
    p2D[i].set_x(x);
    p2D[i].set_y(y);
}
```

closed 2021-02-17 by fspindle