lagadic/visp #798: Inconsistant handling of Exif Orientation between libjpeg and opencv

pgchatela, 2020-08-04:
The result of `vpImageIo::read()` on a JPEG image containing Exif metadata depends on the value of `USE_JPEG`:
- With `USE_JPEG=OFF` and `USE_OPENCV=ON`, `vpImageIo::read()` uses OpenCV as backend, and `cv::imread` considers Exif Orientation and rotates the image accordingly (see https://docs.opencv.org/3.4.11/d4/da8/group__imgcodecs.html#ga288b8b3da0892bd651fce07b3bbd3a56)
- With `USE_JPEG=ON` and `USE_OPENCV=ON`, `vpImageIo::read()` uses libjpeg as backend, which ignores Exif metadata.

Minimal example to reproduce the issue:
```c++
// example.cpp

#include <visp3/core/vpImage.h>
#include <visp3/io/vpImageIo.h>

int main(int argc, const char **argv)
{
  vpImage<vpRGBa> I;
  vpImageIo::read(I, "landscape_6.jpg");
  std::cout << "Image size: " << I.getHeight() << "x" << I.getWidth() << std::endl;
}
```

Then download the sample image and run the code with and without `USE_JPEG`:
```bash
> wget https://raw.githubusercontent.com/ianare/exif-samples/master/jpg/orientation/landscape_6.jpg
> cmake -DUSE_JPEG=OFF -DUSE_OPENCV=ON .
> make -j
> ./example
Image size: 450x600
> cmake -DUSE_JPEG=ON -DUSE_OPENCV=ON .
> make -j
> ./example
Image size: 600x450
```

I'd suggest to make the behavior of `vpImageIo::read()` independent of third-party backends. Options are:
- Always consider Exif Orientation
- Always ignore Exif Orientation
- Add a flag to explicitly consider/ignore Exif Orientation (like the `cv::IMREAD_IGNORE_ORIENTATION` flag)

s-trinh, 2020-08-06:
From my point of view, I would suggest to "Always ignore Exif Orientation". If in the future there is a need for this, it can be added later.

fspindle, 2020-08-24:
Thanks for this feedback.

commit 96cc870c (2020-08-24): Fix inconsistant handling of Exif Orientation between libjpeg and opencv. Closes #798
closed 2020-08-24 by fspindle