lagadic/visp #800: computePose gets worse with more data

gchabert, 2020-08-13:
I've experienced the following unexpected behavior of computePose. I'm using VISP 3.1.0.

In my problem, I have 16 points and corresponding pixels.
When I run computePose with the 8 first or the 8 last points, I get a good estimation (small residual) and the resulting pose is  similar in both cases and close to the truth (see trace below).
But when I run computePose with the 16 points, the estimation fails.

Here is the code:
```
	double pixels[][2] = {
	{601.945373535,1289.42102051}, 
	{644.037902832,1397.66699219}, 
	{750.433105469,1443.87646484}, 
	{857.745056152,1400.93884277}, 
	{903.294006348,1293.63574219}, 
	{860.430969238,1186.04992676}, 
	{753.658508301,1139.24182129}, 
	{647.670166016,1182.48974609}, 
	{96.4898147583,1273.78283691}, 
	{128.912887573,1357.13171387}, 
	{210.612564087,1393.00866699}, 
	{293.219482422,1360.33862305}, 
	{328.785705566,1277.67675781}, 
	{296.352325439,1194.15332031}, 
	{214.422058105,1158.0090332}, 
	{132.116638184,1190.80993652}, 
	};
	
	double points[][3] = {
	{0.01,-0.164,0.0305}, 
	{0.0315667568262,-0.164,0.0215667568262}, 
	{0.0405,-0.164,0}, 
	{0.0315667568262,-0.164,-0.0215667568262}, 
	{0.01,-0.164,-0.0305}, 
	{-0.0115667568262,-0.164,-0.0215667568262}, 
	{-0.0205,-0.164,-3.7351727374e-18}, 
	{-0.0115667568262,-0.164,0.0215667568262}, 
	{0.01,-0.157,0.13503}, 
	{0.0270412734266,-0.157,0.127971273427}, 
	{0.0341,-0.157,0.11093}, 
	{0.0270412734266,-0.157,0.0938887265734}, 
	{0.01,-0.157,0.08683}, 
	{-0.0070412734266,-0.157,0.0938887265734}, 
	{-0.0141,-0.157,0.11093}, 
	{-0.0070412734266,-0.157,0.127971273427}, 
	};
	
	double u0 = 949.2645803;
	double v0 = 557.9620627;
	double px = 1465.209686;
	double py = 1465.047608;

	vpCameraParameters cam_params(px, py, u0, v0);

	std::vector<vpImagePoint> points2D; 
	std::vector<vpPoint> points3D;      

	vpHomogeneousMatrix cMo;
	bool init = true;

	// ========================= case 1 ==============================
	for (int i=0; i<8; i++) {
		points2D.push_back(vpImagePoint(pixels[i][0],pixels[i][1]));
		points3D.push_back(vpPoint(points[i][0],points[i][1],points[i][2]));
	}
	
   	cout << "residual 1 =" << compute_pose(points3D, points2D, cam_params, init, cMo) << endl;
	cout << "pose 1 =" << cMo << endl;
	points2D.clear();
	points3D.clear();
    
	// ========================= case 2 ==============================
	for (int i=8; i<16; i++) {
		points2D.push_back(vpImagePoint(pixels[i][0],pixels[i][1]));
		points3D.push_back(vpPoint(points[i][0],points[i][1],points[i][2]));
	}
	
   	cout << "residual 2 =" << compute_pose(points3D, points2D, cam_params, init, cMo) << endl;
   	cout << "pose 2 =" << cMo << endl;
	points2D.clear();
	points3D.clear();
    
	// ========================= case 3 ==============================
	for (int i=0; i<16; i++) {
		points2D.push_back(vpImagePoint(pixels[i][0],pixels[i][1]));
		points3D.push_back(vpPoint(points[i][0],points[i][1],points[i][2]));
	}
	
	cout << "residual 3 =" << compute_pose(points3D, points2D, cam_params, init, cMo) << endl;
	cout << "pose 3 =" << cMo << endl;
	points2D.clear();
	points3D.clear();

```

The function ``compute_pose`` is copy-pasted from the VISP tutorial:
```
double compute_pose(std::vector<vpPoint> &point, const std::vector<vpImagePoint> &ip,
                 const vpCameraParameters &cam, bool init, vpHomogeneousMatrix &cMo) {
  vpPose pose;
  double x = 0, y = 0;
  for (unsigned int i = 0; i < point.size(); i++) {
    vpPixelMeterConversion::convertPoint(cam, ip[i], x, y);
    point[i].set_x(x);
    point[i].set_y(y);
    pose.addPoint(point[i]);
  }

  double residual;

  if (init == true) {
    vpHomogeneousMatrix cMo_dem;
    vpHomogeneousMatrix cMo_lag;
    pose.computePose(vpPose::DEMENTHON, cMo_dem);
    pose.computePose(vpPose::LAGRANGE, cMo_lag);
    double residual_dem = pose.computeResidual(cMo_dem);
    double residual_lag = pose.computeResidual(cMo_lag);
    if (residual_dem < residual_lag) {
    	residual = residual_dem;
    	cMo = cMo_dem;
    } else {
    	residual = residual_lag;
    	cMo = cMo_lag;
    }
  }
  pose.computePose(vpPose::VIRTUAL_VS, cMo);
  return residual;
}

```

The trace is:

```
residual 1 =0.000132234
pose 1 =0.9992556991  0.03447843231  -0.01729987339  0.06475446827
-0.01696647264  -0.009939086463  -0.999806658  0.03782748545
-0.03464371112  0.9993560188  -0.009346711413  0.4601354556
0  0  0  1

residual 2 =0.000153965417
pose 2 =0.9997122638  0.02057726168  -0.01232744463  0.0618087799
-0.0128543738  0.02567460402  -0.9995877049  0.04386885996
-0.02025227552  0.9994585489  0.02593172432  0.4560932516
0  0  0  1

residual 3 =7.545631406
pose 3 =0.7075585169  0.6694251665  -0.2263424211  0.1138701242
0.7011801312  -0.7048812768  0.1071858625  -0.119802483
-0.08779162091  -0.2345470784  -0.968132377  0.0210207096
0  0  0  1
```

s-trinh, 2020-08-19:
I have tested your code with latest ViSP version.

I am getting:

```
residual 1 =2.08991e-06
pose 1 =0.9993124432  0.03282840004  -0.01723186103  0.06448262297
-0.01693053471  -0.009434820678  -0.9998121529  0.03790581333
-0.03298481284  0.9994164699  -0.008872531337  0.4601164776
0  0  0  1
residual 2 =6.597334557e-05
pose 2 =0.9997145473  0.02045606245  -0.01234396156  0.06178846988
-0.01288477486  0.02652035626  -0.9995652321  0.04400300988
-0.02011980256  0.9994389527  0.02677635771  0.4559609697
0  0  0  1
residual 3 =0.001636083382
pose 3 =0.999490434  0.02791469582  -0.01548038008  0.06371174648
-0.01561505415  0.004612957584  -0.9998674366  0.04020997634
-0.02783958502  0.9995996651  0.005046496464  0.4602618745
0  0  0  1
```

There were some changes in Dementhon / Lagrange pose estimation in #439 that seems to have improved the pose estimation with Dementhon and Lagrange methods. But I don't know why the VVS now converges.

ViSP 3.1.0 is pretty old (22 Dec 2017). In any case, you should update to a more recent ViSP version.Building ViSP manually from source should work flawlessly.

gchabert, 2020-08-21:
Great. Many thanks for your answer.
I'm going to install the lastest release.

closed 2020-08-21 by gchabert