lagadic/visp #801: build issue with visp_java

jeremi, 2020-08-16:
I'm on arch linux, up to date version. I'm using the latest `master` with the PR #795 applied.

The PR #795 fixed the openCV issue for me, but I'm still having an issue building the java part (I've tried with openJDK 8 & 11).

Here is the log:
```In file included from /usr/include/c++/10.1.0/bits/move.h:57,
                 from /usr/include/c++/10.1.0/bits/nested_exception.h:40,
                 from /usr/include/c++/10.1.0/exception:148,
                 from /usr/include/c++/10.1.0/stdexcept:38,
                 from /home/jeremi/tmp/visp/visp/modules/java/generator/src/cpp/common.h:6,
                 from /home/jeremi/tmp/visp/visp-build/modules/java_bindings_generator/gen/cpp/core.inl.hpp:5,
                 from /home/jeremi/tmp/visp/visp-build/modules/java_bindings_generator/gen/cpp/visp_jni.hpp:1,
                 from /home/jeremi/tmp/visp/visp/modules/java/generator/src/cpp/visp_java.cpp:4:
/usr/include/c++/10.1.0/type_traits: In instantiation of ‘struct std::is_move_constructible<vpMbHiddenFaces<vpMbtPolygon> >’:
/usr/include/c++/10.1.0/type_traits:138:12:   required from ‘struct std::__and_<std::is_move_constructible<vpMbHiddenFaces<vpMbtPolygon> >, std::is_move_assignable<vpMbHiddenFaces<vpMbtPolygon> > >’
/usr/include/c++/10.1.0/type_traits:143:12:   required from ‘struct std::__and_<std::__not_<std::__is_tuple_like<vpMbHiddenFaces<vpMbtPolygon> > >, std::is_move_constructible<vpMbHiddenFaces<vpMbtPolygon> >, std::is_move_assignable<vpMbHiddenFaces<vpMbtPolygon> > >’
/usr/include/c++/10.1.0/type_traits:2195:11:   required by substitution of ‘template<class ... _Cond> using _Require = std::__enable_if_t<std::__and_< <template-parameter-1-1> >::value> [with _Cond = {std::__not_<std::__is_tuple_like<vpMbHiddenFaces<vpMbtPolygon> > >, std::is_move_constructible<vpMbHiddenFaces<vpMbtPolygon> >, std::is_move_assignable<vpMbHiddenFaces<vpMbtPolygon> >}]’
/usr/include/c++/10.1.0/bits/move.h:189:5:   required by substitution of ‘template<class _Tp> std::_Require<std::__not_<std::__is_tuple_like<_Tp> >, std::is_move_constructible<_Tp>, std::is_move_assignable<_Tp> > std::swap(_Tp&, _Tp&) [with _Tp = vpMbHiddenFaces<vpMbtPolygon>]’
/home/jeremi/tmp/visp/visp/modules/tracker/mbt/include/visp3/mbt/vpMbHiddenFaces.h:99:15:   required from ‘class vpMbHiddenFaces<vpMbtPolygon>’
/home/jeremi/tmp/visp/visp/modules/tracker/mbt/include/visp3/mbt/vpMbTracker.h:143:33:   required from here
/usr/include/c++/10.1.0/type_traits:960:52: error: static assertion failed: template argument must be a complete class or an unbounded array
  960 |       static_assert(std::__is_complete_or_unbounded(__type_identity<_Tp>{}),
      |                     ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~
make[2]: *** [modules/java/jni/CMakeFiles/visp_java.dir/build.make:134: modules/java/jni/CMakeFiles/visp_java.dir/__/generator/src/cpp/visp_java.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:1983: modules/java/jni/CMakeFiles/visp_java.dir/all] Error 2
make: *** [Makefile:149: all] Error 2```

Do you have any idea of what is wrong?

s-trinh, 2020-08-19:
Can you post the output of the `ViSP-third-party.txt` in the ViSP build folder and the output of `gcc --version`?

fspindle, 2020-08-20:
This issue is also reported in gentoo [[link](https://bugs.gentoo.org/725392)]

I'm able to reproduce it with the code from master without PR #795 

```
$ c++ --version
c++ (GCC) 10.1.0
```

jeremi, 2020-08-21:
Hi @s-trinh ,

```gcc --version
gcc (GCC) 10.1.0
```