lagadic/visp #818: Model-based tracking Issue using RGBD-data on Windows

lugert23, 2020-09-21:
Hello, 

I am trying to run VISP on Windows 10 using RGBD data. For a first trial I ran the tutorials with the help of the guidance on your website (https://visp-doc.inria.fr/doxygen/visp-daily/tutorial-tracking-mb-generic-rgbd.html). After executing the command: `tutorial-mb-generic-tracker-rgbd.exe` the init window appears, so i can set the initialization-points. But when starting the tracking process, the program aborts without any error. The same problem occurs while using `tutorial-mb-generic-tracker-rgbd-realsense.exe`. Unlike the RGBD- tutorials, the generic und stereo versions are working.
Running `testGenericTracker.exe` (in modules/mbt) doesn't work reliable as well for me. In average it passes every third time, otherwise it aborts.

I tried different configurations, with or without OpenBLAS, different versions of PCL and OpenCV, but nothing changed. 

I hope my issue gets clear and you have an idea how to solve it.

My current third-parties are the following:

```
==========================================================
General configuration information for ViSP 3.3.1

  Version control:               3.2.0-1161-gd3be97143-dirty

  Platform:
    Timestamp:                   2020-09-20T18:00:52Z
    Host:                        Windows 10.0.19041 AMD64
    CMake:                       3.16.8
    CMake generator:             Visual Studio 15 2017
    CMake build tool:            C:/Program Files (x86)/Microsoft Visual Studio/2017/Community/MSBuild/15.0/Bin/MSBuild.exe
    MSVC:                        1916

  C/C++:
    Built as dynamic libs?:      yes
    C++ Compiler:                C:/Program Files (x86)/Microsoft Visual Studio/2017/Community/VC/Tools/MSVC/14.16.27023/bin/Hostx86/x64/cl.exe  (ver 19.16.27043.0)
    C++ flags (Release):         /DWIN32 /D_WINDOWS /W3 /GR /EHsc  /MP4   /EHa -openmp -std=c++11 /Gy /bigobj /MD /O2 /Ob2 /DNDEBUG
    C++ flags (Debug):           /DWIN32 /D_WINDOWS /W3 /GR /EHsc  /MP4   /EHa -openmp -std=c++11 /Gy /bigobj /MDd /Zi /Ob0 /Od /RTC1
    C Compiler:                  C:/Program Files (x86)/Microsoft Visual Studio/2017/Community/VC/Tools/MSVC/14.16.27023/bin/Hostx86/x64/cl.exe
    C flags (Release):           /DWIN32 /D_WINDOWS /W3    /MP4   /EHa -openmp -std=c++11 /MD /O2 /Ob2 /DNDEBUG
    C flags (Debug):             /DWIN32 /D_WINDOWS /W3    /MP4   /EHa -openmp -std=c++11 /MDd /Zi /Ob0 /Od /RTC1
    Linker flags (Release):      /machine:x64  /INCREMENTAL:NO
    Linker flags (Debug):        /machine:x64  /debug /INCREMENTAL /ignore:4099

  ViSP modules:
    To be built:                 core gui imgproc io java_bindings_generator klt me sensor ar blob robot visual_features vs vision detection mbt tt tt_mi
    Disabled:                    -
    Disabled by dependency:      -
    Unavailable:                 java

  Windows RT support:            no

  Python (for build):            C:/Users/Steven/AppData/Local/Programs/Python/Python38/python.exe

  Java:                          
    ant:                         NO
    JNI:                         NO

  Build options: 
    Build deprecated:            yes
    Build with moment combine:   no

  Mathematics: 
    Blas/Lapack:                 yes
    \- Use MKL:                  yes
    \- Use OpenBLAS:             no
    \- Use Atlas:                no
    \- Use Netlib:               no
    \- Use GSL:                  no
    \- Use Lapack (built-in):    no
    Use Eigen3:                  yes (ver 3.3.7)
    Use OpenCV:                  yes (ver 4.3.0)

  Simulator: 
    Ogre simulator: 
    \- Use Ogre3D:               no
    \- Use OIS:                  no
    Coin simulator: 
    \- Use Coin3D:               yes (ver 4.0.1)
    \- Use SoWin:                no
    \- Use SoXt:                 no
    \- Use SoQt:                 no
    \- Use Qt5:                  no
    \- Use Qt4:                  no
    \- Use Qt3:                  no

  Media I/O: 
    Use JPEG:                    no
    Use PNG:                     no
    \- Use ZLIB:                 no
    Use OpenCV:                  yes (ver 4.3.0)
    Use stb_image (built-in):    no

  Real robots: 
    Use Afma4:                   no
    Use Afma6:                   no
    Use Franka:                  no
    Use Viper650:                no
    Use Viper850:                no
    Use aria (Pioneer):          no
    Use PTU46:                   no
    Use Biclops PTU:             no
    Use Flir PTU SDK:            no
    Use Parrot ARSDK:            no
    \-Use ffmpeg:                no
    Use Virtuose:                no
    Use qbdevice (built-in):     yes (ver 2.6.0)
    Use takktile2 (built-in):    no

  GUI: 
    Use X11:                     no
    Use GTK:                     no
    Use OpenCV:                  yes (ver 4.3.0)
    Use GDI:                     yes
    Use Direct3D:                no

  Cameras: 
    Use DC1394-2.x:              no
    Use CMU 1394:                no
    Use V4L2:                    no
    Use directshow:              no
    Use OpenCV:                  yes (ver 4.3.0)
    Use Flycapture:              no
    Use Pylon:                   no

  RGB-D sensors: 
    Use Realsense:               no
    Use Realsense2:              yes (ver 2.38.1)
    Use Kinect:                  no
    \- Use libfreenect:          no
    \- Use libusb-1:             no
    \- Use pthread:              no
    Use PCL:                     yes (ver 1.9.1)
    \- Use VTK:                  yes (ver 8.1.2)

  F/T sensors: 
    Use atidaq (built-in):       no
    Use comedi:                  no
    Use IIT SDK:                 no

  Detection: 
    Use zbar:                    no
    Use dmtx:                    no
    Use AprilTag (built-in):     yes (ver 3.1.1)
    \- Use AprilTag big family:  no

  Misc: 
    Use Clipper (built-in):      yes (ver 6.4.2)
    Use pugixml (built-in):      yes (ver 1.9.0)
    Use libxml2:                 no

  Optimization: 
    Use OpenMP:                  yes
    Use pthread:                 no
    Use pthread (built-in):      yes (ver 3.0.1)
    Use cxx standard:            11

  Documentation: 
    Use doxygen:                 yes

  Tests and samples:
    Use catch2 (built-in):       yes (ver 2.9.2)
    Tests:                       yes
    Demos:                       yes
    Examples:                    yes
    Tutorials:                   yes

  Install path:                  C:/visp-ws/visp-build-vc15/install

==========================================================
```

s-trinh, 2020-09-21:
@lugert23 

PR #819 should fix that. Can you test if it solves your issue?

lugert23, 2020-09-22:
@s-trinh 
Thank you very much, that works for me. The tutorals are passing now.

PR #819 (2020-09-22): Revert "Modify vpMbHiddenFaces copy assignment operator to pass the p…
…arameter by reference instead by value"

This reverts commit e54c063f1a1bdb5547f05984e06be66854c6d5b4.

---

resolves #818 

This is the [copy-and-swap idiom](https://stackoverflow.com/questions/3279543/what-is-the-copy-and-swap-idiom/3279550#3279550).
closed 2020-09-22 by lugert23