lagadic/visp #834: Fitting issue of mb-apriltag-tracker using depth information

lugert23, 2020-10-26:
Hello,

I am using the tutorial-mb-generic-tracker-apriltag-rs2 as a basis for my model-based tracker- example, utilizing a realsense D435i depth camera. Tracking a couple of packages with referenced apriltags for initialization leads to the following results. When I only use moving edge features the tracker works well and reliable. The estimated edges fit with the real contours.

[image]

Activating texture features doesn't make a difference, the tracking also works well.
But when the depth features are added, the reliability is gone. The estimated edges doesn't fit (only some) and the tracking process aborts fastly. It looks like a scaling problem of the depth stream, but that should be included in depth_M_color transformation. The lager the model, the more noticeable is the deviation.

[image]

I hope you can clear up this issue. Thanks in advance.

fspindle, 2020-10-26:
I don't know what is occuring here, but I'm not completly sure that your CAD model fit your object using moving edges, see drawings in blue.

[image]

I would suggest to check the model again.

lugert23, 2020-10-26:
I checked the cao-model and it should be fine. I hided some more Apriltags under a sheet of paper in the examples above to avoid tracking failures. I think that was a bit irritating.

I also compared my results to the original tutorial and both had the same issue.

**Example 2:**
without depth:
[image] 
with depth:
[image]

**Example 3:**
without depth:
[image] 
with depth:
[image]

s-trinh, 2020-10-27:
You should check the following:

- color camera intrinsic parameters
- infrared camera intrinsic parameters
- transformation between color and infrared cameras
- depth scale
- check that the images are not registered, for instance the depth image is not registered with the color camera

To debug, I would:

- use a static camera
- use only one box
- track using only color camera, record the estimated pose `cMo`
- track using only the depth camera, record the estimated pose `dMo`
- if everything is ok, you should have `cMo = cMd x dMo` with `cMd` the transformation from the depth to the color camera

lugert23, 2020-10-29:
Thank you for your answer.

The tracking of one single box seems to be correct, just like the camera parameters. I checked the camera transformation the way you suggested, the calculated value for dMc had the correct dimensions (compared to the realsense.getTransformation() value.
I tested the tutorial without any modifications:

<details>
  <summary>tutorial-mb-generic-apriltag-rs2</summary>

``` 
//! \example tutorial-mb-generic-tracker-apriltag-rs2.cpp
#include <fstream>
#include <ios>
#include <iostream>

#include <visp3/gui/vpDisplayGDI.h>
#include <visp3/gui/vpDisplayOpenCV.h>
#include <visp3/gui/vpDisplayX.h>
#include <visp3/core/vpXmlParserCamera.h>
#include <visp3/sensor/vpRealSense2.h>
#include <visp3/detection/vpDetectorAprilTag.h>
#include <visp3/mbt/vpMbGenericTracker.h>

typedef enum {
  state_detection,
  state_tracking,
  state_quit
} state_t;

#if defined(VISP_HAVE_APRILTAG)
state_t detectAprilTag(const vpImage<unsigned char> &I, vpDetectorAprilTag &detector,
                       double tagSize, const vpCameraParameters &cam, vpHomogeneousMatrix &cMo)
{
  std::vector<vpHomogeneousMatrix> cMo_vec;

  // Detection
  bool ret = detector.detect(I, tagSize, cam, cMo_vec);

  // Display camera pose
  for (size_t i = 0; i < cMo_vec.size(); i++) {
    vpDisplay::displayFrame(I, cMo_vec[i], cam, tagSize / 2, vpColor::none, 3);
  }

  vpDisplay::displayText(I, 40, 20, "State: waiting tag detection", vpColor::red);

  if (ret && detector.getNbObjects() > 0) { // if tag detected, we pick the first one
    cMo = cMo_vec[0];
    return state_tracking;
  }

  return state_detection;
}
#endif // #if defined(VISP_HAVE_APRILTAG)

state_t track(const vpImage<unsigned char> &I, vpMbGenericTracker &tracker,
              double projection_error_threshold, vpHomogeneousMatrix &cMo)
{
  vpCameraParameters cam;
  tracker.getCameraParameters(cam);

  // Track the object
  try {
    tracker.track(I);
  }
  catch (...) {
    return state_detection;
  }

  tracker.getPose(cMo);

  // Detect tracking error
[... 417 lines cut ...]

<details>
  <summary>additions</summary>
  
```
      realsense.acquire(reinterpret_cast<unsigned char *>(I_color_aligned.bitmap),
                        reinterpret_cast<unsigned char *>(I_depth_raw_aligned.bitmap), NULL, NULL,
                        (rs2::align *)&align_to_color);

        depthMap.resize(I_depth_raw_aligned.getHeight(), I_depth_raw_aligned.getWidth());
        for (unsigned int i = 0; i < I_depth_raw_aligned.getHeight(); i++) {
          for (unsigned int j = 0; j < I_depth_raw_aligned.getWidth(); j++) {
            if (I_depth_raw_aligned[i][j]) {
              float Z = I_depth_raw_aligned[i][j] * depth_scale;
              depthMap[i][j] = Z;
            } else {
              depthMap[i][j] = 0;
            }
          }
        }
```
</details>

But these changes did not lead to the desired result. The fitting problem stays the same.
Perhaps the depth-dense tracker is not suitable for this kind of task including multiple model parts (or this specific model does not provide enough contours for the depth registration). It also is possible that the camera distortion leads to a wrong registration at the edges of the image resulting in incorrect scale and lower tracking accuracy.

s-trinh, 2020-10-29:
> The tracking of one single box seems to be correct, just like the camera parameters. I checked the camera transformation the way you suggested, the calculated value for dMc had the correct dimensions (compared to the realsense.getTransformation() value.

So with one box the tracking with edges + depth features works correctly?
Another way to confirm that the color and depth intrinsic parameters + the transformation between depth and color cameras are ok would have been to:

- track with only the color camera
- use another `vpMbGenericTracker` and use the `display()` function to project the CAD model on the depth image using the depth intrinsics and using as pose `dMo = dMc x cMo`
- with correct parameters, the projected model with the pose estimated on the color camera should match on the depth image

If so, I think the issue comes from the CAD model. You should not add in the model faces that are not visible. Easiest thing would be to use one model that represents the whole object instead of the concatenation of multiple box models.
Otherwise, you can try the scan-line visibility test (see [Visibility settings](https://visp-doc.inria.fr/doxygen/visp-3.3.0/tutorial-tracking-mb-generic.html#mb_generic_settings_visibility) and [setScanLineVisibilityTest()](https://visp-doc.inria.fr/doxygen/visp-3.3.0/classvpMbTracker.html#a7980bac4eecc2c1876402eeecb244744)). It should eliminate these non visibile faces, but at the expensive of more computation time (so the tracking will be degraded, typically with fast camera or object motion).

> or this specific model does not provide enough contours for the depth registration

Depth features work well with flat objects and with dense depth information. It does not use contours information. To fully recover the 6D pose, with depth features the idea is to have 3 orthogonal flat surfaces, otherwise some degrees of freedom cannot be estimated.
Edge features work well with objects whose contours are clearly visible.
KLT features work well with textured objects.

> It also is possible that the camera distortion leads to a wrong registration at the edges of the image resulting in incorrect scale and lower tracking accuracy.

I think the distortion on the D435 should be ok.

s-trinh, 2020-10-30:
Looking very quickly to the first code, I see:

`realsense.acquire((unsigned char *)I_color.bitmap, (unsigned char *)I_depth_raw.bitmap, &pointcloud, NULL,
                          (rs2::align *)&align_to_color);`

and `mapOfCameraTransformations["Camera2"] = depth_M_color;`

If the depth stream is registered with the color stream, it should mean that the transformation between the depth and color image is idendity and the depth intrinsics is equal to the color intrinsics.

lugert23, 2020-11-02:
> So with one box the tracking with edges + depth features works correctly?

Yes, that works fast and reliable.. without problems. And the more boxes I use for tracking the more inaccurate does the tracking get.

> If so, I think the issue comes from the CAD model. You should not add in the model faces that are not visible. Easiest thing would be to use one model that represents the whole object instead of the concatenation of multiple box models.
Otherwise, you can try the scan-line visibility test (see Visibility settings and setScanLineVisibilityTest()). It should eliminate these non visibile faces, but at the expensive of more computation time (so the tracking will be degraded, typically with fast camera or object motion).

Thanks for all your suggestions I will check that as soon as I can!
I thought that the CAO model is suitable for my scene, cause the tracker with only using the edge features works quite perfect. But I will of cause test the changes as well.

> realsense.acquire((unsigned char *)I_color.bitmap, (unsigned char *)I_depth_raw.bitmap, &pointcloud, NULL, (rs2::align *)&align_to_color);
and mapOfCameraTransformations["Camera2"] = depth_M_color;

Yes you are right, but I am using PCL, so this part will not be reached.
This function here is relevant:

` realsense.acquire((unsigned char *) I_color.bitmap, (unsigned char *) I_depth_raw.bitmap, NULL, pointcloud, NULL);`

fspindle, 2020-11-02:
When the object becomes larger I'm wondering if the precision of the initialisation from the tag pose has an effect that could explain your behavior. Just to test, you could obtain more precision:
- using a tag that is as close as possible from the camera, meaning gluing the tag on the top of your object to get a larger tag in the image
- using `tutorial-mb-generic-tracker-rgbd-realsense.cpp` to initialize the tracker by user click with 4 points that are on your object corners. To this end you have to create a `.init` file with the 3D coordinates of the points to click 

An other point that is not clear to me is the influence of the ratio between the number of moving edges and the number of points in the depth map that are used as features. I suspect that you have too much depth features wrt moving edges when the object becomes larger. Reducing the number of depth features is possible modifying `_depth.xml` config file by increasing the value of `step_X` and `step_Y`:
```
  <depth_dense>
    <sampling_step>
      <step_X>4</step_X>
      <step_Y>4</step_Y>
    </sampling_step>
  </depth_dense>
```

lugert23, 2020-11-05:
> use another vpMbGenericTracker and use the display() function to project the CAD model on the depth image using the depth intrinsics and using as pose dMo = dMc x cMo

Tracking with only using the color camera and transforming into the depth image with dMo works. The projected contours are fitting with the model in the depth image.

> Otherwise, you can try the scan-line visibility test

This one has no recognizable effect on the tracking accuracy, only leads to a higher computation time.

For debugging I only took 3 boxes, the center box got the apriltag. The fitting issue occurs in direction of width, weather I used an assembly out of 3 subboxes, or one model wich combined the dimensions of the 3 boxes. So I can exlude the impact of the cad model.

> I suspect that you have too much depth features wrt moving edges when the object becomes larger. Reducing the number of depth features is possible modifying _depth.xml config file by increasing the value of step_X and step_Y:

Reducing the numer of samples by a tenth, so adding `tracker.setDepthDenseSamplingStep(40,40)` made the fitting-issue disappear. Can this lower amount of depth features effect the tracking behaviour? I think if the sampling step is high enough, the tracker will behave like only using the moving edges. Just for a better understanding: What does the the sample step defines in detail?

fspindle, 2020-11-05:
Concerning the last point, imagine that you have 500 moving-edges and 640x480/(4*4)=19200 points from the depth. The influence of the moving-edges will be neglected wrt the depth. This could explain the observed shift.

Now if you use 40 as depth sample step, you will get 640x480/(40*40)=192 points from the depth. Here the number of moving-edges and depth have the same order of size. 

The sample step is the distance between 2 points.

I will introduce in a comming PR some new functions to get the number of features (edges, klt, depth dense, depth normal) that are taken into account.

lugert23, 2020-11-05:
Thank you very much for your help @s-trinh & @fspindle ! You considerably improved my understanding of the tracker. I am very glad, that we found a solution for this issue.
 
For me this issue can be closed.
 
I only have one last question of understanding:
in the paper 
`S. Trinh, F. Spindler, E. Marchand, and F. Chaumette. A modular framework for model-based visual tracking using edge, texture and depth features. In IEEE/RSJ Int. Conf. on Intelligent Robots and Systems, IROS'18, Madrid, Spain, October 2018.`
you defined the point to plane distance for the depth features as 

[image]
 
My problem is to derive this formula from the given context. Do you can provide a link where this relation is derived or give a short explanation about that?

fspindle, 2020-11-05:
The equation of the plane is given by n_x X + n_y Y + n_z Z + D = 0 with (n_x,n_y,n_z) the normal to the plane and (X,Y,Z) a 3D point belonging to the plane. Given the perspective projection it comes x = X/Z and y = Y/Z. Just by replacing you will get the formula

lugert23, 2020-11-05:
Okay thanks. So the projection is given by the following equations, where d=1? Is d=1 allways assumed?

[image]

[image]

(source https://www.medien.ifi.lmu.de/lehre/ss12/cg1/vorlesung/cg1_5.pdf pp18)

s-trinh, 2020-11-05:
For me, there is no (satisfactory) reason why the fitting using depth leads to this issue.

If you have some time, can you record some data with one box (where it works well) and with 3 boxes (with the issue) with a static camera?

You can use this [program](https://github.com/lagadic/visp/blob/ad4e8b1373a92c2b01b7ed2c0ceef46df7017d4a/example/device/framegrabber/saveRealSenseData.cpp) to record the data (color + depth + pointcloud). You can check and read the data with this [program](https://github.com/lagadic/visp/blob/ad4e8b1373a92c2b01b7ed2c0ceef46df7017d4a/example/device/framegrabber/readRealSenseData.cpp). It will saves also the intrinsic and extrinsic parameters.

Additional data are CAD models and tracker config files.

lugert23, 2020-11-06:
> If you have some time, can you record some data with one box (where it works well) and with 3 boxes (with the issue) with a static camera?

I uploaded the test data (+config and CAD models) to google drive. You can use this link to access:

https://drive.google.com/drive/folders/1qhm5TZJQGlJ1WASLU0wKAR1QtW8O7j2y?usp=sharing

s-trinh, 2020-11-07:
@lugert23 

Thanks for the data. Can you copy/paste the output of this [program](https://github.com/lagadic/visp/blob/85dcf06883859cfb0ef6c350ee150fa321322869/example/device/framegrabber/getRealSense2Info.cpp) with the RealSense connected? The depth scale is missing to recover the 3D coordinates from the depth images.

[image]

Just to be completely sure. The 3 boxes all have the same dimension? And you have the issue with the 3 boxes?

Also, if you are using Apriltag, you should not remove the white border. Looks like also the pattern is blurry? If you are resizing the Apriltag, you have to use no interpolation mode (see [these hints](https://visp-doc.inria.fr/doxygen/visp-daily/tutorial-detection-apriltag.html#apriltag_detection_print)).

---

Looking at the images you uploaded:

- from this distance, CAD model registration with depth should be really good
- RealSense camera should have little to no lens distortion for IR cameras. Because when doing stereo-matching, you expect to find the pixel correspondence on the same row in the other image.
- balance/weighting between number of moving-edges and number of depth dense features is another story and **should not be the solution of the depth registration issue**. Rather than the direct number of features, maybe their "magnitude" or the magnitude of the type of features error should be more relevant.
- on the first images on the first post, possible reasons could be that maybe depth from the RealSense are imprecise if the objects are far (depth error scales with distance quadratically), distortion (I don't think for depth images, maybe for color images but should not since it is not visible on the images)
- `tracker.setDepthDenseSamplingStep(40,40)` --> this is not a solution. This means that depth registration is not good while it should be (at least with the 3 boxes data you uploaded, this use case should be perfectly handled by the depth features). A quicker solution would be to just use moving-edges features and not depth features.
- I will try to find some time to test with your data. Until then for your first use case, just use moving-edges features and not depth features

---

I only have this video at hand, but as you can see tracking with edges and depth features with objects perfectly flat are the perfect use case for the depth dense features.

[image]

lugert23, 2020-11-09:
thank you very much for your answer @s-trinh 
I added the Realsense2Info to the drive directory.
>  if you are using Apriltag, you should not remove the white border

Does it make a difference in detection and in scaling the x / y dimensions? Or does the border is just necessary for textured surroundings? I never paid attention to it.
All boxes have the same dimensions.

> tracker.setDepthDenseSamplingStep(40,40) --> this is not a solution. This means that depth registration is not good while it should be (at least with the 3 boxes data you uploaded, this use case [...]

okay I saw the Problem when I used a larger object, wich was placed more far away (like 2.5 Meters away from the camera). The tracker switched permanently between the correctly fitted objekt and the wrong scaled (smaller) one. 

> I will try to find some time to test with your data. Until then for your first use case, just use moving-edges features and not depth features

I think you will not see any issues, cause the moving edges tracker works fine.
--only occurs while using depth tracker.. and the larger the distance between camera and object is, the worse it gets.

So this is probably the answer:

> on the first images on the first post, possible reasons could be that maybe depth from the RealSense are imprecise if the objects are far (depth error scales with distance quadratically)

s-trinh, 2020-11-09:
> Does it make a difference in detection and in scaling the x / y dimensions? Or does the border is just necessary for textured surroundings? I never paid attention to it.

Apriltag will extract the corners from the black border for 36h11. See [this link](https://github.com/AprilRobotics/apriltag/wiki/AprilTag-User-Guide#pose-estimation) for the tag size to use for pose estimation. So for 36h11, it is the black square size.

The white border is needed for the detection algorithm:

[image]

closed 2020-12-15 by fspindle