lagadic/visp #838: vpImage::getValue() fails on Alpha arch

fspindle, 2020-11-02:
Following [debian build logs](https://buildd.debian.org/status/package.php?p=visp&suite=unstable), we can see that running `testImageGetValue` on Arch lead to the following error (see [here](https://buildd.debian.org/status/fetch.php?pkg=visp&arch=alpha&ver=3.3.0-4&stamp=1604034577&raw=0)):
```
 12/275 Test  #14: testImageGetValue .................................................***Failed    0.37 sec
checkPixelAccess<unsigned char>(3, 4, 2, 3): 11
checkPixelAccess<unsigned char>(3, 4, -2, -3): 
checkPixelAccess<unsigned char>(3, 4, 3, 4): 
checkPixelAccess<vpRGBa>(3, 4, 2, 3): (11,11,11,0)
checkPixelAccess<vpRGBa>(3, 4, -2, -3): 
checkPixelAccess<vpRGBa>(3, 4, 3, 4): 
checkPixelAccess<int>(3, 4, 2, 3): 11
checkPixelAccess<int>(3, 4, -2, -3): 
checkPixelAccess<int>(3, 4, 3, 4): 
checkPixelAccess<double>(3, 4, 2, 3): 11
checkPixelAccess<double>(3, 4, -2, -3): 
checkPixelAccess<double>(3, 4, 3, 4): 
diff_round: 2.56829e+06 ; meanDiffRound: 8.36031
diff: 2.41507e+06 ; meanDiff: 7.86156
Too much pixel difference between fixed-point vpImage::getValue(double, double) and old method.
```

@s-trinh I can't understand why on this specific arch there is more than 1 gray level mean difference. This [table](https://wiki.debian.org/ArchitectureSpecificsMemo) shows the differences between Debian archs.

s-trinh, 2020-11-02:
@fspindle 

Could be (again) an issue with endianness. What is your timeline for the next ViSP release?

---

From the [DEC Alpha](https://en.wikipedia.org/wiki/DEC_Alpha) wiki page, the hardware is bi-endianness. Looks like the [Debian OS](https://wiki.debian.org/ArchitectureSpecificsMemo) should force to use little-endian.

I don't think it will be possible to reproduce the issue easily, e.g. with Docker. My advices would be to:

- detect [Alpha architecture](https://sourceforge.net/p/predef/wiki/Architectures/) and do not use this code path:
https://github.com/lagadic/visp/blob/ad4e8b1373a92c2b01b7ed2c0ceef46df7017d4a/modules/core/include/visp3/core/vpImage.h#L1437

- somehow, run this [code](https://github.com/lagadic/visp/blob/ad4e8b1373a92c2b01b7ed2c0ceef46df7017d4a/modules/core/include/visp3/core/vpEndian.h) during the CMake checks and process it so that we can have the results in the CMake output (and directly in the build logs)

- there is [testEndian.cpp](https://github.com/lagadic/visp/blob/ad4e8b1373a92c2b01b7ed2c0ceef46df7017d4a/modules/core/test/tools/endian/testEndian.cpp), we should add in this file some printing to know which of `VISP_LITTLE_ENDIAN`, `VISP_BIG_ENDIAN` or `VISP_PDP_ENDIAN` has been defined. This way, we could run easily this program to debug on the target architecture.

fspindle, 2020-11-03:
@s-trinh 

Thanks for your feedback. 

In [vpEndian.h](https://github.com/lagadic/visp/blob/master/modules/core/include/visp3/core/vpEndian.h#L84) we set Alpha as little-endian. I suspect that for this specific architecture we should use the version of [reinterpret_cast_uchar_to_uint16_LE()](https://github.com/lagadic/visp/blob/master/modules/core/src/tools/endian/vpEndian.cpp#L115-L117) associated to big-endian architectures.

Since there is no Debian machine available for Alpha testing (see [here](https://db.debian.org/machines.cgi)), the simplest is to not use the optimized version when Alpha is detected.

PR #840 (2020-11-06): Various improvements
- Fix #838 around Alpha architecture endianness
- Improve tutorials (MBT, ROS, libdc1394-dev)
- Modify testEndian.cpp to print detected endianness
- Fix #841 issue with PCL 1.11.0 on OSX
- Fix #842 issue opencv2.framework not detected during iSO build
- remove std::random_shuffle usage deprecated in C++14
- start renaming external doxygen files from .doc to .dox
commit 7e946889 (2020-11-06): Disable optimization in vpImage::getValue() on Alpha arch (Closes: #838)
- testImageGetValue fails on Debian Alpha arch
- Alpha arch is bi-endianess. Little endian is detected but some operations could be big-endian
- There is no Debian machine available to test Alpha: https://db.debian.org/machines.cgi
- The solution is here to disable optimization when Alpha arch is detected
closed 2020-11-06 by fspindle