lagadic/visp #868: OpenCV 4

luigifaticoso, 2020-12-15:
Hello,
I just updated my openCV version. Visp should be compatible with openCV4 but when i run the make it stops at 32% with this output: 

`/home/visp/modules/vision/src/key-point/vpKeyPoint.cpp: In member function 'void vpKeyPoint::initDetector(const string&)':
/home/visp/modules/vision/src/key-point/vpKeyPoint.cpp:2272:66: error: 'cv::xfeatures2d::SIFT' has not been declared
     cv::Ptr<cv::FeatureDetector> siftDetector = cv::xfeatures2d::SIFT::create();
                                                                  ^~~~
/home/visp/modules/vision/src/key-point/vpKeyPoint.cpp: In member function 'void vpKeyPoint::initExtractor(const string&)':
/home/visp/modules/vision/src/key-point/vpKeyPoint.cpp:2450:52: error: 'cv::xfeatures2d::SIFT' has not been declared
     m_extractors[extractorName] = cv::xfeatures2d::SIFT::create();
                                                    ^~~~
modules/vision/CMakeFiles/visp_vision.dir/build.make:326: recipe for target 'modules/vision/CMakeFiles/visp_vision.dir/src/key-point/vpKeyPoint.cpp.o' failed
make[2]: *** [modules/vision/CMakeFiles/visp_vision.dir/src/key-point/vpKeyPoint.cpp.o] Error 1
CMakeFiles/Makefile2:9687: recipe for target 'modules/vision/CMakeFiles/visp_vision.dir/all' failed
make[1]: *** [modules/vision/CMakeFiles/visp_vision.dir/all] Error 2
Makefile:140: recipe for target 'all' failed
make: *** [all] Error 2`

This is the configuration information:
`-- ==========================================================
-- General configuration information for ViSP 3.3.0
-- 
--   Version control:               3.2.0-837-g24b3b2570
-- 
--   Platform:
--     Timestamp:                   2020-12-15T19:01:54Z
--     Host:                        Linux 5.4.0-56-generic x86_64
--     CMake:                       3.10.2
--     CMake generator:             Unix Makefiles
--     CMake build tool:            /usr/bin/make
--     Configuration:               Release
-- 
--   C/C++:
--     Built as dynamic libs?:      yes
--     C++ Compiler:                /usr/bin/c++  (ver 7.5.0)
--     C++ flags (Release):         -Wall -Wextra -fopenmp -std=c++11 -fvisibility=hidden -msse2 -msse3 -mssse3 -fPIC -O3 -DNDEBUG
--     C++ flags (Debug):           -Wall -Wextra -fopenmp -std=c++11 -fvisibility=hidden -msse2 -msse3 -mssse3 -fPIC -g
--     C Compiler:                  /usr/bin/cc
--     C flags (Release):           -Wall -Wextra -fopenmp -std=c++11 -fvisibility=hidden -msse2 -msse3 -mssse3 -fPIC -O3 -DNDEBUG
--     C flags (Debug):             -Wall -Wextra -fopenmp -std=c++11 -fvisibility=hidden -msse2 -msse3 -mssse3 -fPIC -g
--     Linker flags (Release):      
--     Linker flags (Debug):        
-- 
--   ViSP modules:
--     To be built:                 core gui imgproc io java_bindings_generator klt me sensor ar blob robot visual_features vs vision detection mbt tt tt_mi
--     Disabled:                    -
--     Disabled by dependency:      -
--     Unavailable:                 java
-- 
--   Python (for build):            /usr/bin/python2.7
-- 
--   Java:                          
--     ant:                         NO
--     JNI:                         NO
-- 
--   Build options: 
--     Build deprecated:            yes
--     Build with moment combine:   no
-- 
--   Mathematics: 
--     Use MKL:                     no
--     Use OpenBLAS:                no
--     Use Atlas:                   no
--     Use Netlib Lapack:           yes (ver 3.7.1)
--     Use Lapack (built-in):       no
--     Use Eigen3:                  yes (ver 3.3.7)
--     Use OpenCV:                  yes (ver 4.5.0)
--     Use GSL:                     yes (ver 2.4)
-- 
--   Simulator: 
--     Ogre simulator: 
--     \- Use Ogre3D:               no
--     \- Use OIS:                  no
--     Coin simulator: 
--     \- Use Coin3D:               no
--     \- Use SoWin:                no
--     \- Use SoXt:                 no
--     \- Use SoQt:                 no
--     \- Use Qt5:                  no
--     \- Use Qt4:                  no
--     \- Use Qt3:                  no
-- 
--   Media I/O: 
--     Use JPEG:                    yes (ver 80)
--     Use PNG:                     yes (ver 1.6.34)
--     \- Use ZLIB:                 yes (ver 1.2.11)
--     Use OpenCV:                  yes (ver 4.5.0)
--     Use stb_image (built-in):    no
-- 
--   Real robots: 
--     Use Afma4:                   no
--     Use Afma6:                   no
--     Use Franka:                  no
--     Use Viper650:                no
--     Use Viper850:                no
--     Use aria (Pioneer):          no
--     Use PTU46:                   no
--     Use Biclops PTU:             no
--     Use Flir PTU SDK:            no
--     Use Parrot ARSDK:            no
--     \-Use ffmpeg:                no
--     Use Virtuose:                no
--     Use qbdevice (built-in):     yes (ver 2.6.0)
-- 
--   GUI: 
--     Use X11:                     yes
--     Use GTK:                     no
--     Use OpenCV:                  yes (ver 4.5.0)
--     Use GDI:                     no
--     Use Direct3D:                no
-- 
--   Cameras: 
--     Use DC1394-2.x:              no
--     Use CMU 1394:                no
--     Use V4L2:                    no
--     Use directshow:              no
--     Use OpenCV:                  yes (ver 4.5.0)
--     Use Flycapture:              no
--     Use Pylon:                   no
-- 
--   RGB-D sensors: 
--     Use Realsense:               no
--     Use Realsense2:              yes (ver 2.40.0)
--     Use Kinect:                  no
--     \- Use libfreenect:          no
--     \- Use libusb-1:             no
--     \- Use pthread:              yes
--     Use PCL:                     yes (ver 1.9.1)
--     \- Use VTK:                  yes (ver 7.1.1)
-- 
--   F/T sensors: 
--     Use atidaq (built-in):       no
--     Use comedi:                  no
--     Use IIT SDK:                 no
-- 
--   Detection: 
--     Use zbar:                    yes (ver 0.10)
--     Use dmtx:                    no
--     Use AprilTag (built-in):     yes (ver 3.1.1)
--     \- Use AprilTag big family:  no
-- 
--   Misc: 
--     Use Clipper (built-in):      yes (ver 6.4.2)
--     Use pugixml (built-in):      yes (ver 1.9.0)
--     Use libxml2:                 yes (ver 2.9.4)
-- 
--   Optimization: 
--     Use OpenMP:                  yes
--     Use pthread:                 yes
--     Use pthread (built-in):      no
--     Use cxx standard:            11
-- 
--   Documentation: 
--     Use doxygen:                 no
-- 
--   Tests and samples:
--     Use catch2 (built-in):       yes (ver 2.9.2)
--     Tests:                       yes
--     Demos:                       yes
--     Examples:                    yes
--     Tutorials:                   yes
-- 
--   Install path:                  /usr/local
-- 
-- ==========================================================`

Do you know why it's giving me that problem ?
SIRF and SURF should not be available in OpenCV4

s-trinh, 2020-12-15:
SIFT has been moved to the main OpenCV repository since it is no more patented.

You should clone the current ViSP master for appropriate fixes. Else the fixes are the following: https://github.com/lagadic/visp/pull/795

luigifaticoso, 2020-12-15:
With the master branch it works, I hope the new release will be anytime soon!
Thank you

closed 2020-12-15 by luigifaticoso

s-trinh, 2020-12-15:
> I hope the new release will be anytime soon!

See also https://github.com/lagadic/visp/issues/866

---

@luigifaticoso 
Which functionalities of ViSP are you using?