lagadic/visp #872: Travis-ci.org migration to Travis-ci.com

s-trinh, 2020-12-28:
Next year, migration to travis-ci.com will be needed.

See this [FAQ](https://docs.travis-ci.com/user/migrate/open-source-repository-migration/#frequently-asked-questions):

> Q. When will the migration from travis-ci.org to travis-ci.com be completed? #
> 
> A. In an effort to ensure that all of our users - whether you build open-source, public or private repositories - receive regular feature updates, security patches and UX/UI enhancements, we are announcing that travis-ci.org will be officially closed down completely no later than December 31st, 2020, allowing us to focus all our efforts on bringing new features and fixes to travis-ci.com and all of our awesome users like yourself on the travis-ci.com domain.
> Q. What will happen to travis-ci.org after December 31st, 2020? #
> 
> A. Travis-ci.org will be switched to a read-only platform, allowing you to see your jobs build history from all repositories previously connected to travis-ci.org.

Relevant announcements from Travis:

- [The new pricing model for travis-ci.com](https://blog.travis-ci.com/2020-11-02-travis-ci-new-billing)
- [Open Source at Travis CI - An Update](https://blog.travis-ci.com/oss-announcement)
- [Migrating repositories to travis-ci.com](https://docs.travis-ci.com/user/migrate/open-source-repository-migration)

There are some issues with large open-source projects that use Travis-ci.com since the new model uses credits:

- [CI: Travis CI has been completely disabled for SciPy (no credits left) #13109 (Scipy)](https://github.com/scipy/scipy/issues/13109)
- [travis-ci no longer running. #17824 (NumPy)](https://github.com/numpy/numpy/issues/17824)

It should be fine with ViSP since there are not many builds (only for PR and on push on Master branch) on Travis.

In my opinion, it is not urgent and we can wait next year and see what happens.

s-trinh, 2021-06-03:
Some updates:

- [ORG Shutdown](https://blog.travis-ci.com/2021-05-07-orgshutdown)
- yellow message:
[image]

fspindle, 2021-06-08:
Migration should be done and ok, I hope

s-trinh, 2021-07-23:
Looks like builds are not triggered:

- https://travis-ci.com/github/lagadic/visp/builds
- https://travis-ci.com/github/lagadic/visp/pull_requests

fspindle, 2021-08-25:
To keep a trace.

Was working when changes are pushed to my account but not to lagadic organization
- https://travis-ci.com/github/fspindle/visp/builds
- https://travis-ci.com/github/fspindle/visp/pull_requests

The trick was to select the travis-ci [Free Plan](https://app.travis-ci.com/organizations/lagadic/plan) for lagadic organization (see [here](https://newbedev.com/travis-could-not-authorize-build-request)).

closed 2021-08-25 by fspindle