lagadic/visp #875: vpVirtuose class and libopenblas incompatibility on Ubuntu 20.04: Out of memory error

fspindle, 2021-01-06:
I experienced a stange behavior when running `testVirtuose` on Ubuntu 20.04 when ViSP is build with openblas support

```
$ ./testVirtuose
API version: 4.3
Try to connect to localhost port 5000
Catch an exception: Cannot open communication with haptic device using localhost#5000: Out of memory. Check ip and port values
```

This `Out of memory` error occurs when `testVirtuose` is linked to openblas library

```
$ sudo update-alternatives --config liblapack.so.3-x86_64-linux-gnu
There are 3 choices for the alternative liblapack.so.3-x86_64-linux-gnu (providing /usr/lib/x86_64-linux-gnu/liblapack.so.3).

  Selection    Path                                                       Priority   Status
------------------------------------------------------------
* 0            /usr/lib/x86_64-linux-gnu/openblas-pthread/liblapack.so.3   100       auto mode
  1            /usr/lib/x86_64-linux-gnu/atlas/liblapack.so.3              35        manual mode
  2            /usr/lib/x86_64-linux-gnu/lapack/liblapack.so.3             10        manual mode
  3            /usr/lib/x86_64-linux-gnu/openblas-pthread/liblapack.so.3   100       manual mode

$ sudo update-alternatives --config libblas.so.3-x86_64-linux-gnu
There are 3 choices for the alternative libblas.so.3-x86_64-linux-gnu (providing /usr/lib/x86_64-linux-gnu/libblas.so.3).

  Selection    Path                                                     Priority   Status
------------------------------------------------------------
* 0            /usr/lib/x86_64-linux-gnu/openblas-pthread/libblas.so.3   100       auto mode
  1            /usr/lib/x86_64-linux-gnu/atlas/libblas.so.3              35        manual mode
  2            /usr/lib/x86_64-linux-gnu/blas/libblas.so.3               10        manual mode
  3            /usr/lib/x86_64-linux-gnu/openblas-pthread/libblas.so.3   100       manual mode
``` 

When `atlas` or `netlib` is used (see [tutorial](https://visp-doc.inria.fr/doxygen/visp-daily/tutorial-basic-linear-algebra.html)), the binary works as expected.

```
$ ./testVirtuose
API version: 4.3
Try to connect to localhost port 5000
The system is operational.
The current joint values are : 0.08431963623  -0.7517606616  1.673415422  -0.01127831824  0.5821056962  -0.1727561951
```

This issue doesn't occur with openblas on Ubuntu 18.04.
All the tests are passing successfully.

For the moment, I don't know if this issue comes from ViSP, from Virtuose API or from openblas.

s-trinh, 2021-01-06:
Maybe something to try would be to remove apt OpenBLAS and test with OpenBLAS built from source. This can be done like this:

https://github.com/lagadic/visp/blob/b51344248655270c93ca789edc4635132d13e1a1/.github/workflows/ubuntu-dep-src.yml#L34-L49

fspindle, 2021-01-07:
It is working when OpenBLAS is built from source. This confirms that the issue comes from OpenBLAS.

For people having the same issue, I recap the instructions to build first OpenBLAS from source and then ViSP
- remove apt packages: `libopenblas-dev libopenblas-pthread.dev libopenblas0 libopenblas0-pthread` 
- check that openblas is no more available:
    ```
    $ sudo update-alternatives --config liblapack.so.3-x86_64-linux-gnu
    There are 2 choices for the alternative liblapack.so.3-x86_64-linux-gnu (providing /usr/lib/x86_64-linux-gnu/liblapack.so.3).

      Selection    Path                                             Priority   Status
    ------------------------------------------------------------
    * 0            /usr/lib/x86_64-linux-gnu/atlas/liblapack.so.3    35        auto mode
      1            /usr/lib/x86_64-linux-gnu/atlas/liblapack.so.3    35        manual mode
      2            /usr/lib/x86_64-linux-gnu/lapack/liblapack.so.3   10        manual mode
    ```
- download and build OpenBLAS
    ```
    $ git clone --depth 1 https://github.com/xianyi/OpenBLAS.git ${VISP_WS}/OpenBLAS
    $ cd ${VISP_WS}/OpenBLAS
    $ mkdir install
    $ make -j4
    $ make install PREFIX=${VISP_WS}/OpenBLAS/install
    ```
- Setup `OpenBLAS_HOME` environment var
    ```
    $ export OpenBLAS_HOME=${VISP_WS}/OpenBLAS/install
    ```
- Do a fresh ViSP build
    ```
    $ cd ${VISP_WS}/visp-build
    $ rm -rf ${VISP_WS}/visp-build/*
    $ cmake ../visp
    ```
- Here you can verify that OpenBLAS build from source is used
    ```
    $ grep OpenBLAS_INCLUDE_DIR CMakeCache.txt 
    OpenBLAS_INCLUDE_DIR:PATH=${VISP_WS}/OpenBLAS/install/include 
    $ grep OpenBLAS_LIB CMakeCache.txt 
    OpenBLAS_LIB:FILEPATH=${VISP_WS}/OpenBLAS/install/lib/libopenblas.so
    ```
- Complete the build
    ```
    $ make -j4
    ```

closed 2021-02-17 by fspindle