lagadic/visp #879: question on using tutorial-hand-eye-calibration

HeatherHu, 2021-01-14:
Hello,
I tried use tutorial-hand-eye-calibration to get the extrinsic camera parameters, but I didn't get correct. The object pose expressed in camera frame is obtained by using aruco_ros package. All camera poses are shown as 
[image]
This result is not correct. When running tutorial-hand-eye-calibration, I get following information:

> Rotation and translation after VVS
Distance theta between rMo(0) and mean (deg) = 2.9869
Distance theta between rMo(1) and mean (deg) = 2.7832
Distance theta between rMo(2) and mean (deg) = 2.62618
Distance theta between rMo(3) and mean (deg) = 5.25581
Distance theta between rMo(4) and mean (deg) = 4.55532
Distance theta between rMo(5) and mean (deg) = 1.97569
Distance theta between rMo(6) and mean (deg) = 2.37785
Distance theta between rMo(7) and mean (deg) = 3.07176
Distance theta between rMo(8) and mean (deg) = 5.16572
Distance theta between rMo(9) and mean (deg) = 3.11751
Distance theta between rMo(10) and mean (deg) = 4.03812

I can't figure out where I did wrong. I am very appreciate some one could tell me the reason or give some advice. Thanks a lot.

s-trinh, 2021-01-15:
Can you add the corresponding Aruco images? And maybe your calibration data.

The camera poses look strange. I don't recall exactly the Python code, but you should have camera poses all above (or below if it possible?) the tag.

5 to 10 data should be enough to have a good calibration. Try to follow this figure to have meaningful poses:

[image]

Also, remember that the Aruco marker **must not move** during the acquisition procedure.

s-trinh, 2021-01-15:
Another possibility could be that the Aruco poses suffer from the [pose ambiguity problem](https://docs.google.com/document/d/1QU9KoBtjSM2kF6ITOjQ76xqL7H0TEtXriJX5kwi9Kgc/edit#bookmark=id.ve2m6nbh7fg0).

An example:

[image]

The z-axis of the tag frame should point outward (according to the tag frame convention). It is not the case for the small tag.
This is due to imprecision in the 2D corners coordinates that leads to pose ambiguity:

[image]

If this is happening to some of your images, you have to discard the corresponding pairs of `cMo` and `rMe`.

HeatherHu, 2021-01-16:
Thanks very much for your reply. 

The following are my camera pose that is the object represented in camera frame by using aruco_ros. The marker keeps still. 
> 1,-0.00285438, -0.00257288, 0.25957, 0.910647, 0.103835, 0.229497, -0.327524
> 2,-0.0026021, -0.00302805, 0.253244, 0.932824, 0.0579156, 0.101781, -0.340771
> 3,0.00160119, 0.0127384, 0.265068, 0.90706, -0.322302, -0.00910838, -0.270703
> 4,-0.0212275, -0.0116466, 0.264455, 0.850414, -0.40129, -0.291281, -0.175832
> 5,-0.0235548, -0.0225203, 0.273582, 0.827214, -0.427956, -0.313526, -0.185128
> 6,0.0113849, -0.0130544, 0.259457, 0.894634, -0.405966, -0.132494, -0.131403
> 7,0.0014659, 0.000181318, 0.251442, 0.925364, -0.369123, 0.00611403, -0.0860976
> 8,0.0337369, 0.00464033, 0.244498, 0.93073, -0.347849, 0.108442, -0.0313558
> 9,0.00503956, -0.00184136, 0.257245, 0.902096, -0.294149, 0.228728, -0.217674
> 10,-0.00159429, -0.00557048, 0.268425, 0.921351, -0.315838, 0.114353, -0.19566

The robot pose are:

> 1,-0.4962,-0.4803,0.0658,-0.5179,-0.7883,-0.3088,0.1223
> 2,-0.5351,-0.3944,0.0973,-0.5559,-0.7977,-0.2335,0.0110
> 3,-0.5360,-0.3191,0.1164,0.8255,0.5531,0.0850,0.0737
> 4,-0.4205,-0.0989,0.0351,0.8094,0.4862,-0.1692,0.2827
> 5,-0.4104,-0.0840,0.0373,0.8154,0.4561,-0.1736,0.3114
> 6,-0.3900,-0.2517,0.0790,0.8522,0.4929,-0.1388,0.1075
> 7,-0.3899,-0.3696,0.0613,-0.8562,-0.5096,0.0529,0.0669
> 8,-0.4119,-0.4534,0.0280,-0.8461,-0.5123,0.0048,0.1472
> 9,-0.5058,-0.4994,0.0459,-0.8127,-0.5257,-0.2091,0.1395
> 10,-0.4756,-0.4318,0.0913,-0.8309,-0.5379,-0.1201,0.0759

The above attitude is converted to thetaU vector to compute hand-eye pose.

I tried several times, but every time I got quite different result. I am not sure if the smaller Distance theta between rMo(0) and mean (deg) means good result. First I tried get more poses and then look this distance. The following runs, I tried to delete the big distance until all distance are smaller than 1 degree,  aims to get good result.  Is this the correct way? 

I tried more pose about 25, the estimated position seems closer to my configuration, but distance theta is larger about 8 degrees. I am confusing how to know the result is okay or not.

closed 2021-02-17 by fspindle