lagadic/visp #885: No rule to make target '/opt/ros/melodic/lib/librealsense2.so', needed by ... tutorial-viewer

HartmannSa, 2021-02-03:
Hi,

I tried to compile "tutorial-viewer.cpp" with `catkin build mir_vision` in my catkin workspace and got this error message:

```
make[2]: *** No rule to make target '/opt/ros/melodic/lib/librealsense2.so', needed by '/home/userA/catkin_ws_mir/devel/.private/mir_vision/lib/mir_vision/tutorial-viewer'.  Stop.
make[1]: *** [CMakeFiles/tutorial-viewer.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
make: *** [all] Error 2
```
I use Ubuntu 18.04 LTS with Kernel 5.4.0-64-generic and ROS Melodic.

I installed ViSP from Source following these steps https://visp-doc.inria.fr/doxygen/visp-daily/tutorial-install-ubuntu.html .

For installing Realsense2 I followed this steps https://dev.intelrealsense.com/docs/compiling-librealsense-for-linux-ubuntu-guide .
Runnig `locate librealsense2.so` results in:
```
/home/userA/librealsense/build/librealsense2.so 
/home/userA/librealsense/build/librealsense2.so.2.40
/home/userA/librealsense/build/librealsense2.so.2.40.0
/usr/local/lib/librealsense2.so
/usr/local/lib/librealsense2.so.2.40
/usr/local/lib/librealsense2.so.2.40.0
```

I can successfully run all examples or tutorials in the ViSP folder.
I wonder what I have to change to run/build the tutorials in my catkin workspace.

I created a catkin package "mir_vision" and copied the "tutorial-viewer.cpp" into mir_vision/src.
My CMakeLists.txt:
```
cmake_minimum_required(VERSION 3.0.2)
project(mir_vision)
set( sample_dir ${CMAKE_CURRENT_SOURCE_DIR}/src/)
find_package(catkin REQUIRED COMPONENTS
  roscpp
  rospy
  std_msgs 
)
find_package(VISP REQUIRED 
  visp_core 
  visp_mbt 
  visp_io 
  visp_gui 
  visp_sensor
)
find_package(realsense2 REQUIRED)
catkin_package( CATKIN_DEPENDS  roscpp rospy std_msgs )
include_directories(
  ${catkin_INCLUDE_DIRS}
  ${VISP_INCLUDE_DIRS}
  ${realsense2_INCLUDE_DIRS}
)
add_executable(tutorial-viewer ${sample_dir}tutorial-viewer.cpp)
target_link_libraries(tutorial-viewer ${VISP_LIBRARIES} ${catkin_LIBRARIES} ${realsense2_LIBRARY} )
```
I tried to change the CMakeLists.txt and tested commands like:
```
set( REALSENSE_DIR "/home/userA/librealsense/build" )

set(REALSENSE2_LIB_SEARCH_PATH /usr/local/lib)
find_library(REALSENSE2_LIBRARIES NAMES realsense2 PATHS ${REALSENSE2_LIB_SEARCH_PATH})

link_directories("/home/userA/librealsense/build")
```

I found a somehow similar issue #620, however this is on Windows and I can not find folders where realsense2.lib or realsense2.dll are located.

Moreover I found the vision_visp ROS package ( http://wiki.ros.org/vision_visp ). 
Should I rather just install this package (from source or from prebuild packages)? 

Since I don't have much experience with c++ and Ubuntu, I lack the background knowledge to find a solution or the cause of this problem. I appreciate any help or explanation.

s-trinh, 2021-02-03:
I did some quick tests and I have no issue with the following:

- `CMakeLists.txt`

```
cmake_minimum_required(VERSION 3.0.2)
project(mir_vision)
set( sample_dir ${CMAKE_CURRENT_SOURCE_DIR}/src/)
find_package(catkin REQUIRED COMPONENTS
  roscpp
  rospy
  std_msgs 
)
find_package(VISP REQUIRED 
  visp_core 
  visp_mbt 
  visp_io 
  visp_gui 
  visp_sensor
)
catkin_package( CATKIN_DEPENDS  roscpp rospy std_msgs )
include_directories(
  ${catkin_INCLUDE_DIRS}
  ${VISP_INCLUDE_DIRS}
)
add_executable(tutorial-viewer ${sample_dir}tutorial-grabber-realsense.cpp)
target_link_libraries(tutorial-viewer ${VISP_LIBRARIES} ${catkin_LIBRARIES})
```

I have removed `find_package(realsense2 REQUIRED)` since ViSP is already built with librealsense support, but I have tested also with `find_package(realsense2 REQUIRED)` and the other things without no issue.

- `package.xml`

```
<?xml version="1.0"?>
<package format="2">
  <name>mir_vision</name>
  <version>1.0.0</version>
  <description>mir_vision</description>

  <maintainer email="first.last@email.com">First Last</maintainer>
  <license>Apache 2.0</license>

  <buildtool_depend>catkin</buildtool_depend>
  <depend>roscpp</depend>
  <depend>rospy</depend>
  <depend>std_msgs</depend>

</package>
```

I have also cloned `realsense-ros` in the `catkin_ws` and everything build (using `catkin_make`) and run normally (`roslaunch realsense2_camera rs_camera.launch` works): `rosrun mir_vision tutorial-viewer` works correctly (this is actually the [tutorial-grabber-realsense.cpp](https://github.com/lagadic/visp/blob/e0c58342865601d7416a4700220c4ad1cbf5752b/tutorial/grabber/tutorial-grabber-realsense.cpp) code that is running to check that frame grabbing works).

---

On my computer `echo $VISP_DIR` points to `/.../ViSP/ViSP-build` folder and `echo $realsense2_DIR` points to `/.../librealsense/build/install/lib/cmake/realsense2`.
And `ldd devel/lib/mir_vision/tutorial-viewer | grep realsense` gives `librealsense2.so.2.41 => /.../librealsense/build/install/lib/librealsense2.so.2.41`.

You should check that ViSP was built with librealsense support (copy/paste the `ViSP-third-party.txt` file in the ViSP build folder).
You can run also the `tutorial-grabber-realsense.cpp` to confirm there is no issue with ViSP and RealSense support.

Finally, I am still on Ubuntu 16.04 and ROS Kinetic but this should not be related.

HartmannSa, 2021-02-04:
Hi, thanks for the fast reply!

### Image-viewer.cpp works
I modified my CMakeLists.txt (like yours) and deleted the line `find_package(realsense2 REQUIRED)` and the `${realsense2_LIBRARY}`. I deleted the build and devel folder and run `catkin_make` again. 
It builds the executables without errors and I can run `./devel/lib/mir_vision/image-viewer`.

### tutorial-mb-generic-tracker-rgbd-realsense.cpp doesnt work
Running `./devel/lib/mir_vision/rgbd-tracker` results in 
`Install librealsense2 3rd party, configure and build ViSP again to use this example`.

If I add `#define VISP_HAVE_REALSENSE2 true` as the first line in "tutorial-mb-generic-tracker-rgbd-realsense.cpp", `catkin_make`gives the error:
```
Scanning dependencies of target realsense2_camera
[ 95%] Building CXX object Mir200_Sim_withD435/mir_dwb_critics/CMakeFiles/mir_dwb_critics.dir/src/path_progress.cpp.o
[ 96%] Building CXX object realsense-ros-2.2.20/realsense2_camera/CMakeFiles/realsense2_camera.dir/src/realsense_node_factory.cpp.o
/home/rosmatch/visp-ws/src/Mir200_Sim_withD435/mir_vision/src/tutorial-mb-generic-tracker-rgbd-realsense.cpp: In function ‘int main(int, char**)’:
/home/rosmatch/visp-ws/src/Mir200_Sim_withD435/mir_vision/src/tutorial-mb-generic-tracker-rgbd-realsense.cpp:430:47: error: ‘class vpMbGenericTracker’ has no member named ‘getNbFeaturesEdge’; did you mean ‘getFeaturesKlt’?
           ss << "Features: edges " << tracker.getNbFeaturesEdge()
                                               ^~~~~~~~~~~~~~~~~
                                               getFeaturesKlt
/home/rosmatch/visp-ws/src/Mir200_Sim_withD435/mir_vision/src/tutorial-mb-generic-tracker-rgbd-realsense.cpp:431:37: error: ‘class vpMbGenericTracker’ has no member named ‘getNbFeaturesKlt’; did you mean ‘getFeaturesKlt’?
              << ", klt " << tracker.getNbFeaturesKlt()
                                     ^~~~~~~~~~~~~~~~
                                     getFeaturesKlt
/home/rosmatch/visp-ws/src/Mir200_Sim_withD435/mir_vision/src/tutorial-mb-generic-tracker-rgbd-realsense.cpp:432:39: error: ‘class vpMbGenericTracker’ has no member named ‘getNbFeaturesDepthDense’; did you mean ‘getFeaturesCircle’?
              << ", depth " << tracker.getNbFeaturesDepthDense();
                                       ^~~~~~~~~~~~~~~~~~~~~~~
                                       getFeaturesCircle
Mir200_Sim_withD435/mir_vision/CMakeFiles/rgbd_tracker.dir/build.make:62: recipe for target 'Mir200_Sim_withD435/mir_vision/CMakeFiles/rgbd_tracker.dir/src/tutorial-mb-generic-tracker-rgbd-realsense.cpp.o' failed
make[2]: *** [Mir200_Sim_withD435/mir_vision/CMakeFiles/rgbd_tracker.dir/src/tutorial-mb-generic-tracker-rgbd-realsense.cpp.o] Error 1
CMakeFiles/Makefile2:8122: recipe for target 'Mir200_Sim_withD435/mir_vision/CMakeFiles/rgbd_tracker.dir/all' failed
```
 
### tutorial-grabber-realsense.cpp doesnt compile
I tried to build "tutorial-grabber-realsense.cpp". However, `catkin_make` again throws an error:
```
[  6%] Generating Javascript code from mir_msgs/Encoders.msg
/home/rosmatch/visp-ws/src/Mir200_Sim_withD435/mir_vision/src/tutorial-grabber-realsense.cpp:8:10: fatal error: visp3/io/vpImageStorageWorker.h: Datei oder Verzeichnis nicht gefunden
 #include <visp3/io/vpImageStorageWorker.h>
          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
Mir200_Sim_withD435/mir_vision/CMakeFiles/tutorial-viewer.dir/build.make:62: recipe for target 'Mir200_Sim_withD435/mir_vision/CMakeFiles/tutorial-viewer.dir/src/tutorial-grabber-realsense.cpp.o' failed
make[2]: *** [Mir200_Sim_withD435/mir_vision/CMakeFiles/tutorial-viewer.dir/src/tutorial-grabber-realsense.cpp.o] Error 1
CMakeFiles/Makefile2:8280: recipe for target 'Mir200_Sim_withD435/mir_vision/CMakeFiles/tutorial-viewer.dir/all' failed
make[1]: *** [Mir200_Sim_withD435/mir_vision/CMakeFiles/tutorial-viewer.dir/all] Error 2
make[1]: *** Auf noch nicht beendete Prozesse wird gewartet …
[  6%] Generating C++ code from mir_msgs/MissionCtrlState.msg
```
### Env variables
The commands you mentioned result in:

```
echo $VISP_DIR
-> /home/rosmatch/visp-ws/visp-build
echo $realsense2_DIR
-> 
```
So I dont have $realsense2_DIR as an env variable set. I will do this in my ~/.bashrc and try it again.

### ViSP-third-party.txt
Here is my ViSP-third-party.txt and the build was with Realsense2:
```
==========================================================
General configuration information for ViSP 3.3.1

  Version control:               3.2.0-1344-g940cc1b39

  Platform:
    Timestamp:                   2021-01-17T18:05:52Z
    Host:                        Linux 5.4.0-62-generic x86_64
[... 118 lines cut ...]
  Misc: 
    Use Clipper (built-in):      yes (ver 6.4.2)
    Use pugixml (built-in):      yes (ver 1.9.0)
    Use libxml2:                 yes (ver 2.9.4)

  Optimization: 
    Use OpenMP:                  yes
    Use pthread:                 yes
    Use pthread (built-in):      no
    Use cxx standard:            11

  Documentation: 
    Use doxygen:                 yes

  Tests and samples:
    Use catch2 (built-in):       yes (ver 2.9.2)
    Tests:                       yes
    Demos:                       yes
    Examples:                    yes
    Tutorials:                   yes

  Install path:                  /usr/local

==========================================================
```

HartmannSa, 2021-02-04:
I edited my issue with some steps to reproduce my first error message. 
DId you ( @s-trinh ) try this? Or should the rgbd-realsense-tracker work, when the grabber-realsense.cpp works?

Besides, my package.xml fits yours (so I do have the run and build depends vor catkin, etc.).

s-trinh, 2021-02-04:
> Use Realsense2:              yes (ver 2.40.0)

ViSP seems to be built correctly with librealsense support. You can run `./tutorial-grabber-realsense` in the `visp_build/tutorial/grabber` folder to check that image acquisition with the D435 works correctly.

---

> ```
> /home/rosmatch/visp-ws/src/Mir200_Sim_withD435/mir_vision/src/tutorial-mb-generic-tracker-rgbd-realsense.cpp:430:47: error: ‘class vpMbGenericTracker’ has no member named ‘getNbFeaturesEdge’; [...]

> ```
> /home/rosmatch/visp-ws/src/Mir200_Sim_withD435/mir_vision/src/tutorial-grabber-realsense.cpp:8:10: fatal error: visp3/io/vpImageStorageWorker.h: Datei oder Verzeichnis nicht gefunden
>  #include [...]

Somehow you have an issue with ViSP version. You have to check that you are using the correct ViSP version for the `tutorial-mb-generic-tracker-rgbd-realsense.cpp` and `tutorial-grabber-realsense.cpp`.

---

You have to check also that you have only one version of ViSP on your computer. Easiest thing is to not do `make install`. This way you can have multiple ViSP versions on your computer.

Also, check that you don't have ViSP lib coming from ROS. Try to do a locate. Try to do `ldd` on `image-viewer` to check on which ViSP `.so` lib it is built against.

---

> Or should the rgbd-realsense-tracker work, when the grabber-realsense.cpp works?

It is simple. ViSP repo is built on the [CI](https://github.com/lagadic/visp/actions) (continuous integration) regularly. So every tutorial code, example code and test code are regularly tested for correct build.

fspindle, 2021-02-05:
As mentioned by @s-trinh  I'm pretty sure that you have multiple ViSP versions installed. To localize them run:
```
$ locate libvisp-core.so
```
I suspect that you have the `ros-<distro>-visp` package that is installed and that doesn't have the support for `librealsense`. If this is the case, remove this package and do a fresh `catkin` build with `catkin_make -DVISP_DIR=/home/rosmatch/visp-ws/visp-build`

HartmannSa, 2021-02-05:
First of all, thanks for your answers!

So I just made a clean new installation. 

While removing/uninstalling old versions, I realised that the server 
"deb http://realsense-hw-public.s3.amazonaws.com/Debian/apt-repo bionic main" 
(which I added via `sudo apt add-repository` during the installation of librealsense as a pre-build package) 
was still in my/etc/apt/sources.list. It caused an error during sudo apt-get update. 
I simply removed the line from the source.list. Just in case someone else messes up the (un)installation like I did. 
 
After reinstalling visp from source, I can run ./tutorial-grabber-realsense in the visp_build/tutorial/grabber without error.

@fspindle  `$ locate libvisp-core.so` doesn't return anything now. (Maybe because I didnt run `make install''?) Should it return something?

I have one more question concerning your reply @s-trinh: 
> On my computer `echo $VISP_DIR` points to `/.../ViSP/ViSP-build` folder and `echo $realsense2_DIR` points to `/.../librealsense/build/install/lib/cmake/realsense2`.

For what do I need to set $realsense2_DIR? 

By the way, I needed to downgrade librealsense2 to 2.40.0 for other issues with my D435, so that: 
```
:~/visp-ws$ ldd devel/lib/mir_vision/tutorial-viewer | grep realsense
	librealsense2.so.2.40 => /usr/local/lib/librealsense2.so.2.40 (0x00007f47cc1dc000)
```
 
I think the error was propably due to the fact that I installed ViSP from source and as a pre-build package and didnt uninstalled it correctly.

s-trinh, 2021-02-06:
Sorry, `realsense2_DIR` was for another library, I don't remember which one. My comment was in case ViSP was not built with librealsense support, but it is since you checked the `ViSP-third-party.txt` file and you were able to run `tutorial-grabber-realsense` correctly.
ViSP uses these environment variables to detect librealsense:

https://github.com/lagadic/visp/blob/e0c58342865601d7416a4700220c4ad1cbf5752b/cmake/FindRealSense2.cmake#L64-L67

---

It is `locate libvisp_core.so` with **_** instead of **-**. You can also do `ldd tutorial-grabber-realsense | grep visp` to know which libraries it is built against.

---

Are you able to run `tutorial-grabber-realsense` in your `/catkin_ws` now?

HartmannSa, 2021-02-06:
Okay, thanks!
Yes, it works now! :)

```
:~$ locate libvisp_core.so
/home/rosmatch/visp-ws/visp-build/lib/libvisp_core.so
/home/rosmatch/visp-ws/visp-build/lib/libvisp_core.so.3.3
/home/rosmatch/visp-ws/visp-build/lib/libvisp_core.so.3.3.1

:~$ ldd tutorial-viewer | grep visp
	libvisp_gui.so.3.3 => /home/rosmatch/visp-ws/visp-build/lib/libvisp_gui.so.3.3 (0x00007f79ee4ba000)
	libvisp_sensor.so.3.3 => /home/rosmatch/visp-ws/visp-build/lib/libvisp_sensor.so.3.3 (0x00007f79ee27e000)
	libvisp_io.so.3.3 => /home/rosmatch/visp-ws/visp-build/lib/libvisp_io.so.3.3 (0x00007f79ee05e000)
	libvisp_core.so.3.3 => /home/rosmatch/visp-ws/visp-build/lib/libvisp_core.so.3.3 (0x00007f79edae4000)
```

closed 2021-02-06 by HartmannSa