lagadic/visp #890: Ubuntu 20.04 installation

juliamp22, 2021-02-11:
Hello, 
I am trying to install visp3 with different methodes:

1) I have seen that the official package for ubuntu 20.04 does not exist (sudo apt-get install libvisp-dev).
2) I have tried to install visp using the package **visp-3.3.0.tar.gz** and following the page https://visp-doc.inria.fr/doxygen/visp-3.3.0/tutorial-install-ubuntu.html -> Get ViSP source code.

In both cases, the installation breaks.

Some of the errors are the following ones:

/usr/bin/ld: CMakeFiles/cmTC_fac74.dir/src.c.o: in function `main':
src.c:(.text+0x46): undefined reference to `pthread_create'
/usr/bin/ld: src.c:(.text+0x52): undefined reference to `pthread_detac
/usr/bin/ld: src.c:(.text+0x63): undefined reference to `pthread_join'

usr/bin/cmake -E cmake_link_script CMakeFiles/cmTC_5ebfa.dir/link.txt --verbose=1
usr/bin/cc  -DCHECK_FUNCTION_EXISTS=pthread_create    -rdynamic CMakeFiles/cmTC_5ebfa.dir/CheckFunctionExists.c.o  -o cmTC_5ebfa  -lpthreads /usr/lib/x86_64-linux-gnu/libdc1394.so 
usr/bin/ld: cannot find -lpthreads

error: ‘isnan’ was not declared in this scope; did you mean ‘std::isnan’?
    5 |     (void)isnan(1.0)
      |           ^~~~~
      |           std::isnan

/usr/include/c++/9/cmath:632:5: note: ‘std::isnan’ declared here
  632 |     isnan(_Tp __x)
      |     ^~~~~

--> ‘isinf’ was not declared in this scope; did you mean ‘std::isinf’?
--> _finite’ was not declared in this scope

Could anyone tell me if there's something that I am doing wrong or if I should proceed with other type of installation?

Thanks for all,

fspindle, 2021-02-11:
You should get the source from github

```
$ git clone https://github.com/lagadic/visp
```

juliamp22, 2021-02-11:
Worked!! Thanks for all!!

closed 2021-02-11 by fspindle