lagadic/visp #894: Java wrappings for vpImageConvert

failiz, 2021-02-18:
Hi,
I am planning to use Visp for Apriltag detection in java and I saw that there are no wrappings for vpImageConvert. My intention is to use this class to convert a frame obtained through opencv (Mat) to VpImageUChar.

Is there an easy way to make them or documentation about how to use the generator? I added this class to the file list and added some types in the gen_dict.jason, but every time that I tried to define the Mat type I got this error: 

```
Severity	Code	Description	Project	File	Line	Suppression State
Error	MSB8066	Custom build for 'C:\visp-ws\visp-build-vc16\CMakeFiles\cad67428fd38b45df881ac291d5c0db4\visp_java_jar.rule;C:\visp-ws\visp-build-vc16\CMakeFiles\316d1088996a8f8c33372d6144645757\visp_java_jar.rule' exited with code 1.	visp_java_jar	C:\Program Files (x86)\Microsoft Visual Studio\2019\Community\MSBuild\Microsoft\VC\v160\Microsoft.CppCommon.targets	238	
```

And without this type, the generated jar states:
```
    //
    // C++: static void convert(Mat src, vpImage_vpRGBa dest, bool flip = false)
    //

    // Unknown type 'Mat' (I), skipping the function
```

Or is there another way to convert from Mat to VpImageUChar?
Andres

fspindle, 2021-02-18:
I didn't try but from you message I suggest to use `cv::Mat` instead of `Mat` if not already done

failiz, 2021-02-18:
I already tried that and then the message does not appear, but the jar still states "Unknown type 'Mat' (I), skipping the function"

fspindle, 2021-02-18:
From what I see, Mat to vpImage_vpRGBa should be introduced in `generator/src/java/org/visp/utils/Converters.java` which should ideally import `org.opencv.core.Mat`. But a quick test shows that `org.opencv.core.Mat` package is not found.

```
import org.visp.core.VpColVector;
import org.visp.core.VpHomogeneousMatrix;
import org.visp.core.VpImagePoint;
import org.visp.core.VpImageUChar;
import org.visp.core.VpImageRGBa;
import org.visp.core.VpCameraParameters;

import org.opencv.core.Mat;

public class Converters {

  public static VpImageRGBa Mat_to_VpImageRGBa(org.opencv.core.Mat mat) {
     org.visp.core.VpImageRGBa I;
     return I;
  }
  ...
```
The only solution I see for now is to implement the converter outside ViSP, typically in your app.

Maybe @s-trinh  could give some hints.

s-trinh, 2021-02-18:
Unfortunately, I have not used the ViSP Java interface since a long time.

After a quick look, I would try something like this:
- use this constructor: `public VpImageUChar(byte[] array, int height, int width, boolean copyData)`
- convert from OpenCV `Mat` to `byte[]` with something like this:
```
byte[] imgData = new byte[(int) (img.total() * img.channels())];
img.get(0, 0, imgData);
```

I don't remember how the Java binding works, so the easiest solution would be to do the conversion in Java I think.

ViokingTung, 2024-04-28:
hi,I have the same issue,did you solve it?

failiz, 2024-04-29:
I think I did not manage to fix it. I replace the filelist and the gen_dict.json files for the ones attached her to get a few extra functions (I forgot which ones), but not that one: [visp.zip](https://github.com/lagadic/visp/files/15146131/visp.zip)

I implemented the conversion following @s-trinh suggestion (most likely not a very efficient conversion):

```
    public static VpImageUChar Mat2VpImageUChar(Mat matrix) {
        int length = (int) (matrix.total() * matrix.elemSize());
        //System.out.println("Length mat1: " + length + " Size mat1: " + matrix.size());
        // Create the black and white image

        Mat mat1;
        if (matrix.type() != CvType.CV_8UC1) {
            mat1 = new Mat(matrix.size(), CvType.CV_8UC1);
            Imgproc.cvtColor(matrix, mat1, Imgproc.COLOR_RGB2GRAY);
        } else {
            mat1 = matrix;
        }

        // Creating buffer from the matrix
        length = (int) (mat1.total() * mat1.elemSize());
        //System.out.println("Length mat1: " + length + " size: " + mat1.size());
        byte buffer[] = new byte[length];
        mat1.get(0, 0, buffer);

        BufferedImage bImage = new BufferedImage(mat1.width(), mat1.height(), BufferedImage.TYPE_BYTE_GRAY);
        bImage.getRaster().setDataElements(0, 0, mat1.width(), mat1.height(), buffer);

        // Create VpImageUChar
        return new VpImageUChar(buffer, mat1.height(), mat1.width(), true);
    }
```

ViokingTung, 2024-04-29:
> I think I did not manage to fix it. I replace the filelist and the gen_dict.json files for the ones attached her to get a few extra functions (I forgot which ones), but not that one: [...]

Thank you for your valuable advice，I rewrote the [code](https://github.com/lagadic/visp/pull/417#issuecomment-2081300501) with the same idea as yours, compare yours ,I know something where is wrong of my code，thank you so much，have a nice day!

closed 2025-11-03 by fspindle