lagadic/visp #897: Tutorial: Installation from prebuilt packages for Linux Ubuntu/Debian

Muligheder, 2021-02-24:
When i download and follow this guide, i don't receive the packages as stated. I'm missing some header files. I wanted to use the camera calibration for extrinsic parameters, where i wanted to test your tutorial out. When i'm building the repository (make -j4) as stated in ('Tutorial: How to create and build a project that uses ViSP and CMake on Unix or Windows') I'm missing some header files. I've try'd to research my computer the location of such files, and they seem to be gone i.e not downloaded. the files i'm currently talking about are 
#include <visp3/vision/vpHandEyeCalibration.h>
#include <visp3/robot/vpRobotFranka.h>

In my directory /usr/include/x86_64-linux-gnu/visp3(vision/robot) non of the files are there.

fspindle, 2021-02-24:
I you want to use the Franka robot, you need to install ViSP from source code.

Muligheder, 2021-02-24:
Thanks for answer. Actually i wanted to use your library to test some data i have made with another setup. So i have some images and robot poses and want to test it through your library. The interesting header file for me is the HandEyeCalibration. Do i have to get it through your git repository to use it?

fspindle, 2021-02-24:
Just clone github repo !

Muligheder, 2021-02-24:
Okay i will try that, thanks for the fast response

closed 2021-02-24 by Muligheder