lagadic/visp #901: How does one do extrinsic calibration for gantry/cartesian robot?

capeieboi, 2021-04-22:
I have a basic 3 axis gantry robot with No roll, pitch and yaw, only x,y,z movement. When I try to calibrate with the calibration tutorial(https://visp-doc.inria.fr/doxygen/visp-3.3.0/tutorial-calibration-extrinsic.html), I get the following errors. 
[image]

fspindle, 2021-04-23:
The methods implemented in ViSP necessitate rotational motions. In your case, you could estimate the rotational part of eMc by solving the sytem:
eRc ^{c_j}t_{c_i} = ^{e_j}t_{e_i}
but it will be impossible for you to estimate the translational part of eMc if you do not introduce any supplementary knowledge, such as the knowledge of the pose rMo between the reference frame of the robot and the calibration rig/object frame.

capeieboi, 2021-04-23:
Thank you very much for the prompt reply!  
I do have the pose of the robot (x,y,z) for every image of the chessboard, would that suffice?

neherh, 2021-05-06:
To clarify the above equation would this be true? We must solve eRc * t_c = t_e.  Where t_c is a transformation from c_j to c_i and where t_e is a transformation from e_j to e_i. Is this a correct assumption?

fspindle, 2021-05-07:
Yes

neherh, 2021-05-12:
I am a bit new. How do I get the e_j to e_i transformation? And is this method assuming we are using Tsai's method for eye-in-hand calibration?

fspindle, 2021-05-17:
If wMe is the end-effector pose in the robot base frame, knowing 2 end-effector poses in the robot base frame wMe_i and wMe_j, you can compute the homogenous transformation between the corresponding end-effector positions: e_iMe_j.