lagadic/visp #902: Camera Attachment to Panda Robot

jk-ethz, 2021-05-09:
Hi,

I just went over the tutorial at https://visp-doc.inria.fr/doxygen/visp-daily/tutorial-franka-pbvs.html
It says to attach the Realsense camera to the robot end-effector.
My question: How did you do it? Where exactly is it mounted to the robot? I find it hard to find any good points of attachment on the robot. Did you use some custom 3D printed mount?

Here is the corresponding video
https://www.youtube.com/watch?v=7A5cqUEKXHg

fspindle, 2021-05-17:
It is here a mechanical part in aluminium, but you can also 3D print your own interface.

closed 2021-05-17 by fspindle

ndahn, 2021-09-02:
So.... would you provide said model somewhere? :)

fspindle, 2021-09-21:
We released the CAD model of a 3D printed part that allows to attach an Intel RealSense D435 camera to the Panda flange.
The STL file is available here: https://github.com/lagadic/visp_ros/tree/master/tutorial/franka/real-robot

ndahn, 2021-09-23:
Thank you! But that's for a different camera than in the video :/
I found a generic camera mount in franka world. Not quite perfect for the RealSense VF0810, but with some adjustments and/or some glue it will be fine.