lagadic/visp #922: issue with computePlanarObjectPoseFromRGBD with obstacle

longhathuc, 2021-09-02:
Dear ViSP developpers,

In our project at team Gepetto - LAAS, we are using  the function computePlanarObjectPoseFromRGBD to compute the pose of an object by multiple tags in a planar plane. Everything is perfect, our pose estimation improved with the help of the depth info. You could see the pose of the object at the base with 9 tags on the plane 

[image]

But a "big" problem has risen when we do some manipulation on that plane, which has the obstacle like the robot's arm between the camera and the plane. As I saw on the code (sorry if I am wrong), the plane estimation is computed by taking account of the depth value in side the polygon containing the computed 2d points of the tags. Then if by chance, the obstacle is inside the polygon, the depth values of the obstacle are taken account, resulting in the problem of estimation of plane equation. You could see that the pose's frame has gone.

[image]

Currently, I temporarily fixed by modifying your function in our code to export the recomputed 3d points for each tags when it is visible (the area to calculate the plane equation restricted inside the tags)   and feed them to the vpPose class to compute the cMo matrix. It works in our case but not good as original when having no obstacle.

[image]

I would like to know if you have any suggestion in our case?

 Thank you,
Ha Thuc Long

s-trinh, 2021-09-03:
First of all, it is with great pleasure to see that this feature can be useful :+1: 

---

If I have understood correctly, you are feeding to the `computePlanarObjectPoseFromRGBD()` function for all the detected tags all the 2D tags corners and the corresponding 3D points.
So in the 2nd image, a large part of the polygon bounding box is cluttered by the obstacle, which would degrade largely the estimation.

I would try the following:
- get a first estimation of the pose using only the visible tags and without depth information
- use this pose to get a 2D polygon of the whole object
- minority of the 2D polygon should be cluttered by the obstacle
- estimate the final pose with the visible tags + depth
- Tukey robust M-estimator should take care of the obstacle
- otherwise try to experiment with RANSAC?

s-trinh, 2021-09-03:
Try to display also the depth map with the computed inliers/outliers to understand what is going on.

fspindle, 2021-09-03:
There is also the `confidence_index` out parameter that could be interesting to checked.

A quick and easy fix could be to modify `computePlanarObjectPoseFromRGBD()` passing a vector of image points vectors where each vector of image points corresponds to one AprilTag corners.

```
bool vpPose::computePlanarObjectPoseFromRGBD (  const vpImage< float > & depthMap,
  const std::vector< std::vector< vpImagePoint > > & vec_corners,
  const vpCameraParameters & colorIntrinsics,
  const std::vector< vpPoint > & point3d,
  vpHomogeneousMatrix & cMo,
  double * confidence_index = NULL
)  
```
and then inside the function create a `std::vector<vpPolygon> vec_polygon(vec_corners.size())` where `vec_polygon[i]` is filled as an AprilTag bounding box. Then you should fill `points_3d` with the points that belong to `vec_polygon[i]`.

Don't hésitate to submit a pull request if it does the trick ;-)

longhathuc, 2021-09-06:
Dear ViSP developers,

Thank you for the suggestion, the plane is now more stable. It indeed better than my previous solution. I have submitted a pull request. 

By the way, there is a case that works but I don't understand.  I have a driller that use 2 april tags to estimate the base_link of it. Each tag has the relative position to the base link defined in urdf and our system will use the combine poses calculated from 3d points to estimate the object pose. 

Previously, we did not use the depth and it is understandable. But with the depth, those two tags considered in a plane. But it still works, even somehow better. 

[image]

fspindle, 2021-09-07:
It is really strange that your results are better taking into account the depth as implemented in PR #923 for the drilling application. See my additional comments to  PR #923.

closed 2021-11-30 by fspindle