lagadic/visp #925: Dont run: Tutorial: Deep learning object detection

RodrigoCodeBernardo, 2021-09-14:
I am trying to run the example, only it doesn't run or give any errors. Can anyone tell me what I'm missing?

```
euroage@euroage:~/visp-ws/dnn/build$ cmake .. -DCMAKE_BUILD_TYPE=Release
-- The C compiler identification is GNU 7.5.0
-- The CXX compiler identification is GNU 7.5.0
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Configuring done
-- Generating done
-- Build files have been written to: /home/euroage/visp-ws/dnn/build
euroage@euroage:~/visp-ws/dnn/build$ make
Scanning dependencies of target tutorial-dnn-object-detection-live
[ 50%] Building CXX object CMakeFiles/tutorial-dnn-object-detection-live.dir/tutorial-dnn-object-detection-live.cpp.o
[100%] Linking CXX executable tutorial-dnn-object-detection-live
[100%] Built target tutorial-dnn-object-detection-live
euroage@euroage:~/visp-ws/dnn/build$ ./tutorial-dnn-object-detection-live 
```

fspindle, 2021-11-30:
@RodrigoCodeBernardo Any update ?

RodrigoCodeBernardo, 2021-12-30:
Hi @fspindle 

I already checked and the library libopencv_dnn.so is in my directory usr/local/lib

In the vpConfig.h file where the VISP_HAVE_OPENCV_DNN library was defined it was in comment. 

I defined the library (`#define VISP_HAVE_OPENCV_DNN` ) and run `cmake ..` on the folder unix-install

Then I compiled the visp-build folder again, but the library in the vpConfig.h file comes back in comment and doesn't work.

What could I be doing wrong?

Thanks for help.

jnader, 2022-01-21:
Hi @RodrigoCodeBernardo,

If ViSP finds OpenCV DNN module, the line `#define VISP_HAVE_OPENCV_DNN` in `vpConfig.h` should not be commented.
Uncommenting and building this line will not make any changes.

Try the following steps to see if the issue can be solved:

- Make sure to have OpenCV installed by running:
```
$ sudo apt-get install libopencv-dev
```
- Go back to your `ViSP` directory:
```
$ cd visp-ws/dnn
```
- Enter `build` directory (if you still have it), and clean the created files/folders from the last build:
```
$ cd build
$ make clean
```
- Run `cmake` again
```
$ cmake .. -DCMAKE_BUILD_TYPE=Release
```
- At this stage, you can check again `vpConfig.h` to see if OpenCV DNN module is detected:
```
$ grep VISP_HAVE_OPENCV_DNN include/visp3/core/vpConfig.h
```
- If it's found you should have:
```
#define VISP_HAVE_OPENCV_DNN
```
- If not, it may be that your system is not searching in `/usr/local/lib` to find `opencv` libraries. Try then adding `/usr/local/lib` to `LD_LIBRARY_PATH`:
```
$ export LD_LIBRARY_PATH=${LD_LIBRARY_PATH}:/usr/local/lib
```
- Run `cmake` again
- Build ViSP:
```
$ make -j4
$ sudo make install
```
- Run the tutorial again

RodrigoCodeBernardo, 2022-01-21:
Thanks @jnader for the very detailed answer.

I perform the steps you said, when I run the command:

`$ grep VISP_HAVE_OPENCV_DNN include/visp3/core/vpConfig.h`

I found:

euroage@euroage:~/visp-ws/visp/build$  grep VISP_HAVE_OPENCV_DNN include/visp3/core/vpConfig.h
#define VISP_HAVE_OPENCV_DNN

I run tutorial but... Don run...

fspindle, 2022-01-21:
It could be due to multiple ViSP installations and that you are linking with one that is not build with OpenCV DNN enabled

To check:
```
$ sudo updatedb
$ locate libvisp_core
```

RodrigoCodeBernardo, 2022-01-21:
I have:

```
/home/euroage/visp-ws/visp/build/lib/libvisp_core.so
/home/euroage/visp-ws/visp/build/lib/libvisp_core.so.3.4
/home/euroage/visp-ws/visp/build/lib/libvisp_core.so.3.4.1
/opt/ros/melodic/lib/libvisp_core.so
/opt/ros/melodic/lib/libvisp_core.so.3.4
/opt/ros/melodic/lib/libvisp_core.so.3.4.0
/opt/ros/melodic/lib/libvisp_core.so.3.4.1
/usr/lib/x86_64-linux-gnu/libvisp_core.so
/usr/lib/x86_64-linux-gnu/libvisp_core.so.3.1
/usr/lib/x86_64-linux-gnu/libvisp_core.so.3.1.0
/usr/local/lib/libvisp_core.so
/usr/local/lib/libvisp_core.so.3.4
/usr/local/lib/libvisp_core.so.3.4.
```
Could it be conflicting with the visp installation for ROS?

fspindle, 2022-01-21:
Wouah

You have a very old installation 3.1.0 in `/usr/lib/`, an other coming with ros in `/opt/ros/melodic` (maybe now overwritten with your installation from `visp-build-ros`), and an other installation 3.4.1 maybe in `/usr/local/lib` that comes probably from a `make install` launched from `/home/euroage/visp-ws/visp/build`.

I suspect that you are linking with the one in `/usr/lib/` that doesn't have the DNN support.

1. You need first to remove the installed version in `/usr` that was maybe installed using `apt`. The easiest is to install `synaptic` and remove packages like `libvisp-dev`, `libvisp_core`, `libvisp_io`...

    Run `sudo updatedb` and `locate` again to see if `/usr/lib/x86_64-linux-gnu/` is empty with visp libraries

2. Remove than your installation in `/usr/local/lib`

    ```
    $ cd /home/euroage/visp-ws/visp/build
    $ sudo make unsinstall
    ```
    again run `sudo updatedb` and `locate` to see if `/usr/local/lib/` is empty with visp libraries

3. What you did creating `visp-build-ros` and installing in `/opt/ros/$ROS_DISTRO` is the good solution to overwrite `ros-melodic-visp`package. It should work at this step

RodrigoCodeBernardo, 2022-01-21:
I cleaned up all the visp installations, and installed it clean:

```
euroage@euroage:~$ locate libvisp_core
/home/euroage/visp-ws/visp-build-ros/lib/libvisp_core.so
/home/euroage/visp-ws/visp-build-ros/lib/libvisp_core.so.3.4
/home/euroage/visp-ws/visp-build-ros/lib/libvisp_core.so.3.4.1
/opt/ros/melodic/lib/libvisp_core.so
/opt/ros/melodic/lib/libvisp_core.so.3.4
/opt/ros/melodic/lib/libvisp_core.so.3.4.1
```

OpenCV DNN module is detected:

```
euroage@euroage:~/visp-ws/visp-build-ros$ grep VISP_HAVE_OPENCV_DNN include/visp3/core/vpConfig.h
#define VISP_HAVE_OPENCV_DNN
```

Yet it still doesn't run....

`euroage@euroage:~/visp-ws/visp-build-ros/tutorial/detection/dnn$ ./tutorial-dnn-object-detection-live `

The same installation on ubuntu 20.04 works without any problems.

jnader, 2022-01-21:
In fact,

The problem comes most probably from the version of `opencv` installed. On `ubuntu 18.04` the version is `3.2` when installing from package. And the module `DNN` is not packaged on ubuntu 18.04.

To begin with, you may check which libraries the tutorial is using with :
```
$ ldd ./tutorial-dnn-object-detection-live | grep opencv_
```

---

But still, did you ever build `opencv` from source before? Because the fact that you can find `libopencv_dnn.so` in `/usr/local/lib` it means you have the module.

Can you execute the following command, to see to where is this library linked:
```
$ ll /usr/local/lib/libopencv_dnn.so
```

---

You should maybe consider building opencv from source (version `> 3.4.3`) to have DNN module available.

To do so, clone the latest version of `opencv` from [github project](https://github.com/opencv/opencv), and build it. Then, rebuild `visp` with the latest version of `opencv` installed.

RodrigoCodeBernardo, 2022-01-21:
I have updated my OpenCV library, my DNN version went from libopencv_dnn.so.3.3 to libopencv_dnn.so.405.

I can't see where the library is linked with the command:

`euroage@euroage:~/visp-ws/visp-build-ros/tutorial/detection/dnn$ ldd ./tutorial-dnn-object-detection-live | grep opencv_
`
It still doesn't work even with the updated DNN.

jnader, 2022-01-22:
Have you tried to build `ViSP` again with the newer version of opencv?

RodrigoCodeBernardo, 2022-01-25:
Thank you all very much, the problem was with the version of opencv that was installed.

Best Regards

closed 2022-01-25 by RodrigoCodeBernardo