lagadic/visp #927: Problem with vpRealSense2 Class Reference

lucamarchionna, 2021-09-20:
Hi everyone, I was trying to implement an eye-in-hand configuration through a markerless generic model-based tracking. For doing this, I am going to leverage a RealSense d435i and benefit from the vpRealSense2 Class. However, when I include the header 
<p style="font: 400 14px / 22px Roboto, sans-serif; color: rgb(0, 0, 0); letter-spacing: normal; orphans: 2; text-align: start; text-indent: 0px; text-transform: none; white-space: normal; widows: 2; word-spacing: 0px; -webkit-text-stroke-width: 0px; background-color: rgb(255, 255, 255); text-decoration-thickness: initial; text-decoration-style: initial; text-decoration-color: initial;"><code>#include &lt;<a class="el" href="https://visp-doc.inria.fr/doxygen/visp-daily/vpRealSense2_8h_source.html" style="color: rgb(61, 87, 140); font-weight: bold; text-decoration: none;">visp3/sensor/vpRealSense2.h</a>&gt;</code></p>

an error appear in the catkin_make, mentioning the following:

**undefined reference to vpRealSense2::vpRealSense2()
undefined reference to vpRealSense2::open(rs2::config const&)**

I have successfully installed the 3rd parties according to VISP tutorial. Have you an idea on how to fix this error? Thank you in advance.

fspindle, 2021-09-20:
Investigate the following points:
- is ViSP build with librealsense support ? Check `vpConfig.h` file to see if `VISP_HAVE_REALSENSE2` is defined
- it is also possible that you have multiple ViSP installation, some with, some without librealsense support
  ```
  $ sudo updatedb
  $ locate libvisp_core.so
  ```

fspindle, 2021-11-30:
@lucamarchionna Any update ?

lucamarchionna, 2021-11-30:
Thanks for your support. I solved the problem following what you have suggested in another issue. For other users, the steps are:
-  Identify local installations with 
 `
 sudo updated `
 `locate libvisp_core.so 
`
- Remove duplicates

> 1. Install Synaptic `sudo apt-get install synaptic`
> 2. Run Synaptic `sudo synaptic`
> 3. Delete any installation from packages that contains visp-ros, vision-visp

- Build visp in a separate catkin_ws (e.g. visp-ws) following [https://visp-doc.inria.fr/doxygen/visp-daily/tutorial-install-ubuntu.html](url)
- Clone visp_ros package into your catkin and build it
- Use visp dependencies to create your own code

closed 2021-11-30 by fspindle