lagadic/visp #930: Make error: ‘class vpCalibration’ has no member named ‘setAspectRatio’

RyugaLaser, 2021-09-21:
Trying to install ViSP from source following the [quick installation](https://visp-doc.inria.fr/doxygen/visp-daily/tutorial-install-ubuntu.html) on platform Ubuntu 18.04, I'm having trouble with this make error when I use `make -j4` in the visp-build folder. 

The cmake does not give any error.
I already tried to remove everything and cloning again the visp repository, with no effect

```
[ 58%] Built target servoSimuSphere
Scanning dependencies of target calibrate-hand-eye
[ 58%] Building CXX object demo/wireframe-simulator/CMakeFiles/servoSimuCylinder.dir/servoSimuCylinder.cpp.o
[ 58%] Building CXX object example/calibration/CMakeFiles/calibrate-hand-eye.dir/calibrate-hand-eye.cpp.o
[ 58%] Built target testGenericTrackerCAOParsing
Scanning dependencies of target calibrate-camera
[ 58%] Building CXX object example/calibration/CMakeFiles/calibrate-camera.dir/calibrate-camera.cpp.o
[ 58%] Linking CXX executable calibrate-hand-eye
/home/giulio/visp-ws/visp/example/calibration/calibrate-camera.cpp: In function ‘int main(int, const char**)’:
/home/giulio/visp-ws/visp/example/calibration/calibrate-camera.cpp:231:15: error: ‘class vpCalibration’ has no member named ‘setAspectRatio’
         calib.setAspectRatio(opt_aspect_ratio);
               ^~~~~~~~~~~~~~
example/calibration/CMakeFiles/calibrate-camera.dir/build.make:62: recipe for target 'example/calibration/CMakeFiles/calibrate-camera.dir/calibrate-camera.cpp.o' failed
make[2]: *** [example/calibration/CMakeFiles/calibrate-camera.dir/calibrate-camera.cpp.o] Error 1
CMakeFiles/Makefile2:11664: recipe for target 'example/calibration/CMakeFiles/calibrate-camera.dir/all' failed
make[1]: *** [example/calibration/CMakeFiles/calibrate-camera.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
[ 58%] Built target calibrate-hand-eye
[ 58%] Linking CXX executable servoSimuCylinder
[ 58%] Built target servoSimuCylinder
[ 58%] Linking CXX executable testGenericTrackerDeterminist
[ 58%] Built target testGenericTrackerDeterminist
Makefile:140: recipe for target 'all' failed
make: *** [all] Error 2

```

RyugaLaser, 2021-09-21:
Thank you for the answer.

I have also a libvisp_core version 3.1.0 in my x86_64-linux-gnu folder.
However I tried to purge the libvisp-dev package installation that I previously made (which for Ubuntu 18.04 is in fact 3.1.0) but libvisp_core 3.1.0 is still present in the system.

How can I completely remove the package installation and just use the source installation?

fspindle, 2021-09-21:
3.4.0 is not the last github version.

1. You should remove any existing ViSP build and potential installation (have in mind that `make install` for ViSP is useless since ViSP can be used as a third-party from the build folder)

2. Get ViSP github version

    ```
    $ cd ~/visp-ws
    $ git clone https://github.com/lagadic/visp
    $ mkdir visp-build
    $ cmake ../visp
    $ make -j4
    ```

3. Build your example 
 
    ```
    $ cmake <source folder> -DVISP_DIR=~/visp-ws/visp-build 
    ```

RyugaLaser, 2021-09-28:
Solved following:
#766 

Still have to reinstall ROS packages and verify that they reference the main directory though

closed 2021-11-30 by fspindle