lagadic/visp #931: Framerate issue on V4L2 camera

Myrmecoman, 2021-09-27:
Hello,

I've been working with VISP for a few days, and eventually realized I needed a larger resolution to have better detection and pose estimation. My camera is HD capable in both MJPG and yuyv422 as can be seen doing this command :

```
v4l2-ctl --list-formats-ex

ioctl: VIDIOC_ENUM_FMT
	Index       : 0
	Type        : Video Capture
	Pixel Format: 'MJPG' (compressed)
	Name        : Motion-JPEG
		Size: Discrete 1920x1080
			Interval: Discrete 0.033s (30.000 fps)
		Size: Discrete 1280x720
			Interval: Discrete 0.017s (60.000 fps)
		Size: Discrete 1024x768
			Interval: Discrete 0.033s (30.000 fps)
		Size: Discrete 640x480
			Interval: Discrete 0.008s (120.101 fps)
		Size: Discrete 800x600
			Interval: Discrete 0.017s (60.000 fps)
		Size: Discrete 1280x1024
			Interval: Discrete 0.033s (30.000 fps)
		Size: Discrete 320x240
			Interval: Discrete 0.008s (120.101 fps)

	Index       : 1
	Type        : Video Capture
	Pixel Format: 'YUYV'
	Name        : YUYV 4:2:2
		Size: Discrete 1920x1080
			Interval: Discrete 0.167s (6.000 fps)
		Size: Discrete 1280x720
			Interval: Discrete 0.111s (9.000 fps)
		Size: Discrete 1024x768
			Interval: Discrete 0.167s (6.000 fps)
		Size: Discrete 640x480
			Interval: Discrete 0.033s (30.000 fps)
		Size: Discrete 800x600
			Interval: Discrete 0.050s (20.000 fps)
		Size: Discrete 1280x1024
			Interval: Discrete 0.167s (6.000 fps)
		Size: Discrete 320x240
			Interval: Discrete 0.033s (30.000 fps)
```

However  yuyv422 is limited to 6fps, when MJPG can reach 30fps in HD. When I run cheese it also chooses yuyv422 and cannot be changed.

In the code doing this still uses yuyv and still has 6fps which is normal since it uses yuyv422 :

```
    vpV4l2Grabber g;
    g.setVerboseMode(true);
    g.setDevice(device.str());
    g.setScale(1);
    g.setWidth(1920);
    g.setHeight(1080);
    g.open(I);
```

I tried using other parameters using :

```
    g.setPixelFormat((vpV4l2Grabber::vpV4l2PixelFormatType) pixelFormat);
    g.setInput(inputType);
```

But nothing worked I always had 6fps in the end no matter the parameters. It seems like I just can't use the full capabilities of my camera. I ran it on windows and had out of the box 30fps however. Is it because I cannot load a compressed format, or am i missing something else here ?

fspindle, 2021-09-27:
Did you try with OpenCV `cv:VideoCapture` ?
There is an example in `tutorial-grabber-opencv.cpp` that can be modified introducing required framerate and resolution.

Myrmecoman, 2021-09-27:
I did not and I will, I'll keep you updated on the attempt.

Thanks for the quick answer.

Myrmecoman, 2021-09-27:
Ok so I tried this code :

```
    cv::VideoCapture g(opt_device); // Open the default camera
    g.set(CV_CAP_PROP_FRAME_WIDTH, 1920);
    g.set(CV_CAP_PROP_FRAME_HEIGHT, 1080);
    g.set(6, CV_FOURCC('M', 'J', 'P', 'G')); //CV_CAP_PROP_FOURCC
    if (!g.isOpened()) {            // Check if we succeeded
      std::cout << "Failed to open the camera" << std::endl;
      return -1;
    }
    cv::Mat frame;
    g >> frame; // get a new frame from camera
    vpImageConvert::convert(frame, I);
```

Inspired from the tutorial-grabber-opencv-bis.cpp file. I still have the same framerate even though I ask clearly for MJPG format.

fspindle, 2021-09-27:
It sounds that MJPEG is not well supported.
Googling with mjpg and opencv you can found a lot of post that may help to investigate.

Myrmecoman, 2021-09-27:
I'm not so sure, I think it is possible to use MJPEG format.

Indeed this command works in shell for displaying the camera view, and it indeed has 30fps (nearly).

`gst-launch-1.0 v4l2src device=/dev/video0 io-mode=2 ! image/jpeg, width=1920, height=1080 !  nvjpegdec ! video/x-raw ! xvimagesink`

The use of the command in the code should be :

`cv::VideoCapture g('v4l2src device=/dev/video0 io-mode=2 ! image/jpeg, width=(int)1920, height=(int)1080 !  nvjpegdec ! video/x-raw ! xvimagesink', cv::CAP_GSTREAMER);`

Running this code outputs :

```
Use OpenCV grabber on device 0
[ WARN:0] global /home/nvidia/host/build_opencv/nv_opencv/modules/videoio/src/cap_gstreamer.cpp (1757) handleMessage OpenCV | GStreamer warning: Embedded video playback halted; module v4l2src0 reported: Cannot identify device '/dev/video1936289387'.
[ WARN:0] global /home/nvidia/host/build_opencv/nv_opencv/modules/videoio/src/cap_gstreamer.cpp (886) open OpenCV | GStreamer warning: unable to start pipeline
[ WARN:0] global /home/nvidia/host/build_opencv/nv_opencv/modules/videoio/src/cap_gstreamer.cpp (480) isPipelinePlaying OpenCV | GStreamer warning: GStreamer: pipeline have not been created
Failed to open the camera
```
First I do not know why it tries to read in /dev/video1936289387, and second I think I might not have a proper version of opencv since I work on a Jetson nano and it is not enabled by default as said in this thread : https://forums.developer.nvidia.com/t/gstreamer-can-open-ip-camera-successfully-but-uisng-opencv-cap-gstreamer-cant-open-ip-camera/68668/2

I think I'm getting closer to the solution, and probably a few visp users using this kind of hardware might like to have a few more fps too. :+1:

Myrmecoman, 2021-09-28:
So I was wrong, this code works for decoding  MJPG :

```
    g.set(CV_CAP_PROP_FRAME_WIDTH, 1920);
    g.set(CV_CAP_PROP_FRAME_HEIGHT, 1080);
    g.set(CV_CAP_PROP_FOURCC, CV_FOURCC('M', 'J', 'P', 'G'));
```

I was limited by the CPU not by my camera. I should take a look at GPU accelerated decoding for my Jetson nano. This is getting out of the topic of this thread so I'm closing it. Thanks for your time.

closed 2021-09-28 by Myrmecoman