lagadic/visp #986: AprilTag Memory Leak ViSP 3.4.0

erikstauber, 2022-01-14:
I've seen a few older bugs related to memory leaks with the AprilTag detector; it still is leaking on my system. (Win 10 x64)

I'm running Visp3.4.0, and the method **bool vpDetectorAprilTag::detect** leaks about 20MB/min when detecting 5 tags at 60hz.

s-trinh, 2022-01-14:
Could you add a minimal code to reproduce the issue?

erikstauber, 2022-01-26:
Yes, I'll prepare that..

erikstauber, 2022-02-03:
I've done a more comprehensive test using the tutorial code unmodified, and it appears after an hour or so the memory usage flattens out.  Here's my test setup:

- VISP version = 3.4.0
- Test Application = tutorial-apriltag-detector-live (using OpenCV)
- opencv version = 4.5.3
- architecture = x64
- OS = Windows 64
- cpu = AMD 3700x
- Camera = Logitech USB 910 webcam
- VISP build options = C++17, AVX2
- Build type = release

Test 	#Threads	# of Tags detected	Detection Time(ms)	0min	10min	20min	30min	60min	120min
1		1		5			12				67mb	84mb	101mb	113mb	108mb		
2		4		5			6				72mb	88mb	104mb	114mb	147mb	151mb

So after a while, it seems like the memory usage stabilizes.  The issue seems magnified with larger images and more threads.  So this behavior isn't as bad as I feared, but it still seems odd IMO, but is ultimately something I can live with.

erikstauber, 2022-02-03:
Test update:

After leaving the tutorial application running for 4 hours, memory usage is up to 282mb.

Erik

fspindle, 2022-02-10:
@erikstauber Can you check again if this issue is now fixed on your side using the master branch

closed 2022-07-21 by fspindle