lagadic/visp #1039: Planar pose computation and pose ambiguity

s-trinh, 2022-03-14:
This thread is to document the pose ambiguity issue that can arise with planar pose computation.

Visually:
- from [Iterative Pose Estimation Using Coplanar Feature Points, Denis Oberkampf, Daniel F.DeMenthon, Larry S.Davis, 1995](https://citeseerx.ist.psu.edu/viewdoc/download?doi=10.1.1.441.251&rep=rep1&type=pdf):
[image]

- from [Sensor Fusion for Fiducial Tags: Highly Robust Pose Estimation from Single Frame RGBD, Pengju Jin, Pyry Matikainen, Siddhartha S. Srinivasa, 2017](https://personalrobotics.cs.washington.edu/publications/jin2017rgbdtags.pdf):
[image]

- from the [ArUco FAQ](https://docs.google.com/document/d/1QU9KoBtjSM2kF6ITOjQ76xqL7H0TEtXriJX5kwi9Kgc/edit#heading=h.sfvvnk7hklcd):
[image]

---

https://github.com/lagadic/visp/pull/671 implements pose from RGBD to try to overcome this issue, but you need an RGBD sensor and the quality of the depth map, the range of the depth sensor, the extrinsic calibration between color and depth sensors will matter.

---

@fspindle @chaumett

Is it possible to detect these problematic situations? Is something similar to the Lowe ratio test (if the reprojection error for the two poses are too close, maybe we can say that the computed pose can be ambiguous?) possible to detect this situation?

Or the only easy solution is to use two non-planar tags (similar to http://www.particlerobots.com/luismateos/latching/apriltags3d.html)?

---

A video to see the pose flipping issue (**warning: in the video I was using some arbitrary focal length values (it was to illustrate marker detection and pose computation), that was probably accentuating the problem**):

https://user-images.githubusercontent.com/8035162/158278092-11eb64f1-bfe5-478d-89e8-4b6b1ed728a8.mp4

chaumett, 2022-03-15:
Dementhon's method for planar object tracks the two ambiguous poses (with a 2-branch tree) and selects the one with the lowest residual. The residuals for the  two ambiguous poses are very low and near when the object is far away/small in the image, which is not a recommended situation. We could imagine to use the Lowe ratio test but the issue would be to tune the threshold...
In practice, for an image sequence, by using as initial pose for VVS/LM the pose obtained for the previous image, the flipping should not occur since it is very unlikely that VVS/LM converges to the other branch. So the problem may be if the wrong branch has been selected for the very first image... 
If the pose is flipping duriing an image sequence, it probably means you have initialized VVS/LM by Dementhon/Lagrange/other for each image, which is really not recommended. 
Hope this helps...

s-trinh, 2022-03-15:
Thanks for your answer @chaumett 

I took the opportunity to ask you some long-standing questions.

---

Some work result advert.
We were involved in the H2020 PULSAR project (it ended in 04/2021).

Nothing new in the following video but the result is still cool :smiley: (thanks to ViSP of course).

https://user-images.githubusercontent.com/8035162/158483521-f3f86319-d39d-4637-81fc-c58f1e128b1c.mp4